Moddable-OpenSource/moddable #1206: Couldn't connect to Wifi on moddable_three

sebastien-burel, 2023-09-06:
**Build environment:** macOS Ventura on M1 
**Moddable SDK version:** 4.0.0-32-g3d52e78b9
**Target device:**  Moddable Three

**Description**
I try to build the commodetto/epaper-mini-travel-time on a moddable/three and i get Wi-Fi disconnected message inside xsbug log. 
**Steps to Reproduce**
- fresh install the latest moddble sdk
- fresh install esp8266 tools
- getting the API key from google, fixing the cert files to "$(MODULES)/crypt/data/ca233"

**Expected behavior**
Connected to the Wifi networks

**Other information**
The same project works fine on simulator, 
The same project works fine on moddable_two, 
The same project connect to the wifi on moddable_one but get a memory full which is something normal i guess.
_Wi-Fi connected to "La Fourbe"
IP address 192.168.50.16
\# Chunk allocation: failed for 988 bytes
XS abort: memory full
/Users/sb/Projects/moddable/xs/platforms/esp/xsPlatform.c (274) # Break: C: xsDebugger!_

Other wifi examples doesn't connect to Wifi on moddable_three.
webConfigWifi example doesn't show myAccessPoint but hs the same log than moddable_one
No Wi-Fi SSID
configAP
configServer
advertiseServer mywifi
probe for mywifi
probe 1
probe 2
probe 3
probe claimed mywifi

It looks like the wifi module works but no transmission.

Many thanks
Sébastien

phoddie, 2023-09-07:
Sébastien, thank you for the report.

There are a few different things going on here!

- It appears that you have received a defective Moddable Three. We will send you a replacement. We have also rechecked our inventory to confirm Wi-Fi transmit and receive are working, to avoid this problem for anyone else in the future. Apologies.
- Indeed, Google changed the certificate needed to establish the TLS connection. We'll update the example.
- The JSON directions  can get quite big. In one test I ran, it was over 600 KB. There is a limit to the size of the response the ESP8266 can handle. To allow more cases to work, I experimented with receiving the JSON to flash instead of RAM. This roughly doubles the size of the JSON it can accept. (Of course, the ESP32 has more memory and the simulator has almost infinite memory, so these can accept a larger JSON response.)
- The `HOME` and `WORK` addresses should be URI encoded or the Google API rejects addresses with spaces (or other special characters) as malformed requests. I've made that change as well.

The changes need some testing and review, so they are not available yet. They should be available in a day or so.

sebastien-burel, 2023-09-08:
Peter, 

Thanks you for your respoonse.
I have three moddable three, all are defffective ? 
it seems really a bad luke.

Sebastien

phoddie, 2023-09-12:
Crazy bad luck, if that's the case.

Moddable One and Moddable Three use the same ESP8266 module. Since Moddable One works for you, it doesn't seem like a software problem.

Let's try one more test to see if any of your Moddable Three units are transmitting. Instead of connecting to a Wi-Fi access point, let's try setting up Moddable Three as an access point. Maybe this is redundant with your `webConfigWifi` above, but just to be sure....

```
cd $MODDABLE/examples/network/wifi/wifiaccesspoint
mcconfig -d -m -p esp/moddable_three
```

When that start up, it should trace something like `http server ready at 192.168.4.1`. That means the Wi-Fi access point is ready.

On your computer or phone, check to see if the access point is visible. It Is called "South Village". If you can see it, Wi-Fi transmit is working. Otherwise, no.

If it is visible, you can connect (password is `12345678`). From your computer ping and curl should work:

```
> ping 192.168.4.1
PING 192.168.4.1 (192.168.4.1): 56 data bytes
64 bytes from 192.168.4.1: icmp_seq=0 ttl=128 time=2.664 ms
^C
> curl 192.168.4.1/123
hello, client at path /123
```

sebastien-burel, 2023-09-18:
I did the wifiaccesspoint test with my 3 moddable_three, i can't see any access point. 
In xsbug i got (all the messages are twice)

> No Wi-Fi SSID
> No Wi-Fi SSID
> 
> http server ready at 192.168.4.1
> http server ready at 192.168.4.1
> 

I did the same with moddable_one i can connect to the access point, ping the device and curl the server.

The main difference between moddable_one and moddable_three, is the way i plug to my usb-c macbook.
- moddable_one connection is direct while moddable_two is connected throw the moddable programmer rev.1.
- I try to plug a newly flash moddable_three directly to my usb-c macbook, no network is showing
- I try use a second programmer rev.1, no network is showing.
- i try different USB-c port, intern, external monitor, no network is showing

I can send you back my 3 moddable_three if you want.

Sébastien

The moddbale_three test:
[image]

The moddable_one test:
[image]

phoddie, 2023-09-18:
Sébastien, thank you for running the additional tests. It definitely looks like the Wi-Fi transmit is broken on your Moddable Three units. All the W-Fi functionality (including the connection to the antenna) is on the ESP8266 module we purchased for use in Moddable One and Moddable Three. Perhaps there was a small batch of defective modules. Or, perhaps the module was damaged when being attached to the board or in transit (but both of those seem unlikely). We have seen this problem only once before on a Moddable Three and never on Moddable One. As noted above, we rechecked our inventory and didn't find any other bad units.

Anyway... we will ship you replacement units that work. If it isn't too much trouble, it would be helpful to get those bad units back at some point so we can try to determine what went wrong.

sebastien-burel, 2023-09-20:
Peter, i will send you backs my moddable three units. Can you precise you mailing adress.

phoddie, 2023-09-20:
Sébastien, here's the Moddable office mail address:

> Moddable
> 408 Florence Street
> Palo Alto, CA 94301

Thank you!

sebastien-burel, 2023-09-29:
Peter
I just receive the new units, all of them have a working wifi.
Many thanks.

phoddie, 2023-09-29:
Sébastien, glad that the new units are working for you.

We received your three units earlier this week. We were able to reproduce your results -- Wi-Fi doesn't transmit. We checked the soldering of the ESP8266 module to the main PCB under a microscope and didn't see any bad connections. We re-soldered the connections on one of the boards, but that didn't help. We then replaced the ESP8266 module, and then Wi-Fi worked as expected. So, it looks like the problem is that we received a few bad ESP8266 modules from our supplier. It is bizarre that you received three of those though. Sorry again for the trouble.

closed 2023-09-29 by phoddie