Moddable-OpenSource/moddable #1207: Unable to write to nRF52(Windows, xiao)

NextWorldLab, 2023-09-10:
Hello,

I cannot write to nRF52 (xiao) on Windows.
xiao writes and recognizes the bootloader.
[image]

When I write Examples/helloworld, the Volume name seems different and I have to wait.
[image]

I think "waiting for volume MODDABLE4" is incorrect and should be "MODDABLEnRF".

Thanks,

mkellner, 2023-09-10:
Thank you for the bug report. You are right. The nmake.nrf52.mk file does not handle the UF2_VOLUME_NAME variants. I will fix that.

mkellner, 2023-09-10:
It will be fixed in the next repository update.

You can make the following changes until then:

```
diff --git a/tools/mcconfig/nmake.nrf52.mk b/tools/mcconfig/nmake.nrf52.mk
index 744cd06c6..29ddb5584 100644
--- a/tools/mcconfig/nmake.nrf52.mk
+++ b/tools/mcconfig/nmake.nrf52.mk
@@ -94,7 +94,9 @@ OBJCOPY = $(TOOLS_BIN)\arm-none-eabi-objcopy
 SIZE = $(TOOLS_BIN)\arm-none-eabi-size

 PLATFORM_DIR = $(MODDABLE)\build\devices\nrf52
+!IF "$(UF2_VOLUME_NAME)"==""
 UF2_VOLUME_NAME = MODDABLE4
+!ENDIF
 WAIT_FOR_M4 = $(PLATFORM_DIR)\config\waitForVolumeWindows.bat $(UF2_VOLUME_NAME) $(TMP_DIR)\_drive.tmp $(TMP_DIR)\_port.tmp $(M4_VID) $(M4_PID)
 DO_COPY = -for /F "tokens=1" %%i in ( $(TMP_DIR)\_drive.tmp ) do @copy $(BIN_DIR)\xs_nrf52.uf2 %%i
 ECHO_GIT_AND_SIZE = $(PLATFORM_DIR)\config\echoGitTagAndSizeWindows.bat $(TMP_DIR)\_size.tmp $(MODDABLE) $(NRF52_HEAP_SIZE)
```

NextWorldLab, 2023-09-10:
@mkellner san
thank you for your reply.
After I fixed it, it worked.
thank you.

closed 2023-09-11 by NextWorldLab