Moddable-OpenSource/moddable #1208: Link error occurs in Node-RED simple flow

NextWorldLab, 2023-09-11:
Hello,

When I try to check with the simple flow of Node-RED MCU, a link error occurs.
I got the same error both in Plugin and in CLI.

Sorry if this is not the repository.

environment
Windows
Node-RED 3.1
device nrf52/xiao

[image]

flow
[image]
```
[{"id":"1f82f515b29c9cb8","type":"tab","label":"フロー 4","disabled":false,"info":"","env":[],"_mcu":{"mcu":true}},{"id":"e3e1fa57279a0fad","type":"inject","z":"1f82f515b29c9cb8","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","_mcu":{"mcu":true},"x":160,"y":120,"wires":[["935eb5642e874fdc"]]},{"id":"935eb5642e874fdc","type":"debug","z":"1f82f515b29c9cb8","name":"debug 6","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","_mcu":{"mcu":true},"x":460,"y":140,"wires":[]}]
```

command
```
mcconfig -d -m -p nrf52/xiao
```

Output
```
C:\pjt\node-red-mcu>mcconfig -d -m -p nrf52/xiao
# application: debugger.o
# application: debugger_usbd.o
# application: ftdi_trace.o
# application: main.o
# application: systemclock.o
# application: xsmain.o
# application: app_usbd_vendor.o
# nodered2mcu flows
# xsc flows.xsb
# xsc preference.xsb
# xsc time.xsb
# xsc timer.xsb
# xsc Resource.xsb
# xsc deepEqual.xsb
# xsc structuredClone.xsb
# xsc modules.xsb
# xsc hex.xsb
# xsc nodered.xsb
# xsc main.xsb
# xsc pins\digital.xsb
# xsc pins\digital\monitor.xsb
# xsc button.xsb
# xsc led.xsb
# xsc setup\target.xsb
# xsc commodetto\Bitmap.xsb
# xsc ~.embeddedioanalog.xsb
# xsc ~.embeddediodigital.xsb
# xsc ~.embeddediodigitalbank.xsb
# xsc ~.embeddedioi2c.xsb
# xsc ~.embeddediosmbus.xsb
# xsc ~.embeddediopwm.xsb
# xsc ~.embeddediopulsecount.xsb
# xsc ~.embeddedioserial.xsb
# xsc ~.embeddediospi.xsb
# xsc system.xsb
# xsc ~.embeddedproviderbuiltin.xsb
# xsc mc\config.xsb
# xsid modPreference.c.xsi
# xsid modGPIO.c.xsi
# xsid modMonitor.c.xsi
# xsid modTime.c.xsi
# xsid modTimer.c.xsi
[... 57 lines cut ...]
# cc commodettoBitmap.c.o
# cc _analog.c.o
# cc _pwm.c.o
# cc _pulsecount.c.o
# cc serial.c.o
# cc system.c.o
# buildinfo
# project xs: xsHost.o
# project xs: xsHosts.o
# project xs: xsPlatform.o
# project xs: xsDebug.o
creating xs_nrf52.out
link to .out file
c:/users/tauchi/nrf5/arm-gnu-toolchain-12.2.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\_pwm.c.o: in function `xs_pwm_destructor_':
c:\pjt\moddable\modules\io\pwm\nrf52/_pwm.c:319: undefined reference to `nrfx_pwm_uninit'
c:/users/tauchi/nrf5/arm-gnu-toolchain-12.2.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\_pwm.c.o: in function `xs_pwm_constructor_':
c:\pjt\moddable\modules\io\pwm\nrf52/_pwm.c:269: undefined reference to `nrfx_pwm_init'
c:/users/tauchi/nrf5/arm-gnu-toolchain-12.2.rel1-mingw-w64-i686-arm-none-eabi/bin/../lib/gcc/arm-none-eabi/12.2.1/../../../../arm-none-eabi/bin/ld.exe: c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\_pwm.c.o: in function `xs_pwm_write_':
c:\pjt\moddable\modules\io\pwm\nrf52/_pwm.c:364: undefined reference to `nrfx_pwm_simple_playback'
collect2.exe: error: ld returned 1 exit status
NMAKE : fatal error U1077: 'C:\Users\tauchi\nrf5\arm-gnu-toolchain-12.2.rel1-mingw-w64-i686-arm-none-eabi\bin\arm-none-eabi-gcc -mabi=aapcs  -mcpu=cortex-m4  -mfloat-abi=hard  -mfpu=fpv4-sp-d16  -mthumb  --specs=nano.specs  -LC:\Users\tauchi\nrf5\nRF5_SDK_17.0.2_d674dde\modules\nrfx\mdk  -Tc:\pjt\moddable\build\devices\nrf52\config\xsproj.ld  -Wl,--gc-sections  -Wl,--no-warn-rwx-segments  -Xlinker -no-enum-size-warning  -Xlinker -Map=c:\pjt\moddable\build\bin\nrf52\xiao\debug\node-red-mcu\xs_lib.map c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\buildinfo.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\xsHost.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\xsHosts.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\xsPlatform.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\xsDebug.o   c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\modPreference.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\modGPIO.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\modMonitor.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\modTime.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\modTimer.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\timer.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\Resource.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\modInstrumentation.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\deepEqual.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\structuredClone.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\modules.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\modHex.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\modSPI.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\builtinCommon.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\digitalbank.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\_i2c.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\spi.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\nodered.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\digital.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\commodettoBitmap.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\_analog.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\_pwm.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\_pulsecount.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\serial.c.o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\system.c.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\boards.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\bsp.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\bsp_btn_ble.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\croutine.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\event_groups.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\heap_4.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\list.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\port.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\port_cmsis.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\port_cmsis_systick.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\queue.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\stream_buffer.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\tasks.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\timers.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\auth_status_tracker.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\ble_advdata.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\ble_advertising.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\ble_conn_params.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\ble_conn_state.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\ble_radio_notification.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\ble_srv_common.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\gatt_cache_manager.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\gatts_cache_manager.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\id_manager.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\nrf_ble_gatt.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\nrf_ble_lesc.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\nrf_ble_qwr.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\nrf_ble_scan.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\peer_data_storage.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\peer_database.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\peer_id.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\peer_manager.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\peer_manager_handler.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\pm_buffer.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\security_dispatcher.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\security_manager.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\nrf_crypto_aead.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\nrf_crypto_aes.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\nrf_crypto_aes_shared.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\nrf_crypto_ecc.o  c:\pjt\moddable\build\t [... cut ...] xiao\debug\lib\xsTree.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\xsType.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\xsdtoa.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\xsmc.o  c:\pjt\moddable\build\tmp\nrf52\xiao\debug\lib\xsre.o -lc  -lnosys  -lm  C:\Users\tauchi\nrf5\nRF5_SDK_17.0.2_d674dde\external\nrf_cc310\lib\cortex-m4\hard-float\no-interrupts\libnrf_cc310_0.9.13.a -o c:\pjt\moddable\build\tmp\nrf52\xiao\debug\node-red-mcu\xs_nrf52.out' : リターン コード '0x1'
Stop.
```

Thanks,

phoddie, 2023-09-11:
Thank you for the report. On macOS, there is no link error building for `nrf52/xiao`. It looks like the `nrfx_pwm_*` functions aren't present in your build. We'll check the Windows build.

mkellner, 2023-09-11:
We'll get that fixed in the next repository update.

But for now, in `%MODDABLE%\tools\mcconfig\nmake.nrf52.mk`, at about line 320, add `$(LIB_DIR)\nrfx_pwm.o`.

```
diff --git a/tools/mcconfig/nmake.nrf52.mk b/tools/mcconfig/nmake.nrf52.mk
index 29ddb5584..01dd8bcad 100644
--- a/tools/mcconfig/nmake.nrf52.mk
+++ b/tools/mcconfig/nmake.nrf52.mk
@@ -317,6 +317,7 @@ NRF_DRIVERS_OBJ = \
        $(LIB_DIR)\nrfx_power.o \
        $(LIB_DIR)\nrfx_ppi.o \
        $(LIB_DIR)\nrfx_prs.o \
+       $(LIB_DIR)\nrfx_pwm.o \
        $(LIB_DIR)\nrfx_qdec.o \
        $(LIB_DIR)\nrfx_saadc.o \
        $(LIB_DIR)\nrfx_spim.o \
```

NextWorldLab, 2023-09-12:
After the modification, compilation became OK.
But I can't connect to xsbug.
Looks like some more fixes are needed.
I'll take a look and report.

thank you

phoddie, 2023-09-12:
We did some tests to see if we can reproduce the problem.

We used the following flow which logs the following on Moddable Four:
- Heartbeat every 10 seconds
- Every turn of the jog wheel
- Every press/release of the jog wheel (GPIO 11)
- Every press/release user button (GPIO 13)

<details>
<summary>Moddable Four IO</summary>

```json
[
    {
        "id": "05081066d0c46e09",
        "type": "tab",
        "label": "Moddable Four IO",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": {
            "mcu": false
        }
    },
    {
        "id": "7fa65f038655f533",
        "type": "inject",
        "z": "05081066d0c46e09",
        "name": "",
        "props": [
            {
                "p": "payload"
            }
        ],
        "repeat": "10",
        "crontab": "",
        "once": true,
        "onceDelay": 0.1,
        "topic": "",
        "payload": "",
        "payloadType": "date",
        "_mcu": {
            "mcu": false
        },
        "x": 150,
        "y": 100,
        "wires": [
            [
                "c4d1bfad60bc379f"
            ]
        ]
    },
    {
        "id": "c4d1bfad60bc379f",
        "type": "debug",
        "z": "05081066d0c46e09",
        "name": "debug 89",
        "active": true,
        "tosidebar": true,
        "console": false,
        "tostatus": false,
        "complete": "payload",
        "targetType": "msg",
        "statusVal": "",
        "statusType": "auto",
        "_mcu": {
            "mcu": false
        },
        "x": 380,
        "y": 180,
        "wires": []
    },
    {
        "id": "01b5d11aae73cbdb",
        "type": "mcu_pulse_count",
        "z": "05081066d0c46e09",
        "name": "",
        "signal": "6",
        "control": "8",
        "moddable_manifest": {
            "include": "$(NODEREDMCU)/nodes/mcu/pulsecount/manifest.json"
        },
        "_mcu": {
            "mcu": false
        },
        "x": 150,
        "y": 160,
        "wires": [
            [
                "c4d1bfad60bc379f"
            ]
        ]
    },
    {
        "id": "59a2ace6476bc986",
        "type": "mcu_digital_in",
        "z": "05081066d0c46e09",
        "name": "",
        "pin": "11",
        "mode": "InputPullUp",
        "edge": "3",
        "debounce": "25",
        "initial": true,
        "invert": true,
        "moddable_manifest": {
            "include": "$(NODEREDMCU)/nodes/mcu/digital/manifest.json"
        },
        "_mcu": {
            "mcu": false
        },
        "x": 150,
        "y": 220,
        "wires": [
            [
                "c4d1bfad60bc379f"
            ]
        ]
    },
    {
        "id": "e5cc80ca823dc4e1",
        "type": "mcu_digital_in",
        "z": "05081066d0c46e09",
        "name": "",
        "pin": "13",
        "mode": "InputPullUp",
        "edge": "3",
        "debounce": "25",
        "initial": true,
        "invert": true,
        "moddable_manifest": {
            "include": "$(NODEREDMCU)/nodes/mcu/digital/manifest.json"
        },
        "_mcu": {
            "mcu": false
        },
        "x": 150,
        "y": 280,
        "wires": [
            [
                "c4d1bfad60bc379f"
            ]
        ]
    }
]
```
</details>

The flow builds and runs on both macOS and Windows. The output is logged to xsbug as expected. The test was done from the command line using `mcconfig -d -m -p nrf52/moddable_four` to build.

I'm not sure if this flow will run on your hardware, since the pins connections will be different (you can delete or modify the nodes as needed). But, the test confirms that fundamentals are working.  Does something like that work for you?

NextWorldLab, 2023-09-12:
@phoddie san
thank you for your reply.
When using Node-RED MCU Plugin, Node-RED Editor and Xsbug are not connecting properly.

I have confirmed that xsbug and device (xiao) are successfully connected in moddable's example/helloworld.

I changed the injector to work when I started it, and it showed up in Debug in the Node-Red Editor.
However, there is no response when I press the Injector button.
So that's it
MCU --> Node-RED Editor OK
Node-RED Editor --> MCU NG

I haven't checked yet, but Mac and Linux makefiles are different than Windows makefiles, so I think this is probably the problem.

make.nrf52.mk
```
ifeq ($(DEBUG),1)
		ifeq ($(XSBUG_LOG),1)
			DO_XSBUG =
			CONNECT_XSBUG=@echo "Connect to xsbug." ; cd $(MODDABLE)/tools/xsbug-log && XSBUG_PORT=$(XSBUG_PORT) XSBUG_HOST=$(XSBUG_HOST) node xsbug-log serial2xsbug $(M4_VID):$(M4_PID) 921600 8N1
			NORESTART=-norestart
			WAIT_FOR_COPY_COMPLETE =
		else
			DO_XSBUG = open -a $(MODDABLE_TOOLS_DIR)/xsbug.app -g
			CONNECT_XSBUG=@echo "Connect to xsbug." ; XSBUG_PORT=$(XSBUG_PORT) XSBUG_HOST=$(XSBUG_HOST) serial2xsbug $(M4_VID):$(M4_PID) 921600 8N1
			NORESTART=-norestart
			WAIT_FOR_COPY_COMPLETE =
		endif
	else
		DO_XSBUG =
		CONNECT_XSBUG =
		NORESTART =
		WAIT_FOR_COPY_COMPLETE = $(PLATFORM_DIR)/config/waitForVolume -x $(UF2_VOLUME_PATH)
	endif
```

nmake.nrf52.mk
```
!IF "$(DEBUG)"=="1"
DO_XSBUG = tasklist /nh /fi "imagename eq xsbug.exe" | find /i "xsbug.exe" > nul || (start $(MODDABLE_TOOLS_DIR)\xsbug.exe)
KILL_SERIAL_2_XSBUG =-tasklist /nh /fi "imagename eq serial2xsbug.exe" | (find /i "serial2xsbug.exe" > nul) && taskkill /f /t /im "serial2xsbug.exe" >nul 2>&1
WAIT_FOR_NEW_SERIAL = $(PLATFORM_DIR)\config\waitForNewSerialWindows.bat 1 $(UF2_VOLUME_NAME) $(TMP_DIR)\_port.tmp $(M4_VID) $(M4_PID)
SERIAL_2_XSBUG = echo Starting serial2xsbug. Type Ctrl-C twice after debugging app. && $(MODDABLE_TOOLS_DIR)\serial2xsbug $(M4_VID):$(M4_PID) $(DEBUGGER_SPEED) 8N1 -dtr
NORESTART = 
!ELSE
DO_XSBUG =
KILL_SERIAL_2_XSBUG =
WAIT_FOR_NEW_SERIAL = $(PLATFORM_DIR)\config\waitForNewSerialWindows.bat 0 $(UF2_VOLUME_NAME) $(TMP_DIR)\_port.tmp $(M4_VID) $(M4_PID)
SERIAL_2_XSBUG = 
NORESTART =
!ENDIF
````

Thank you,

phoddie, 2023-09-12:
Interesting. It looks like the change to remove "kill serial2xsbug" didn't make it to nRF52 Windows. What happens if you change

```
KILL_SERIAL_2_XSBUG =-tasklist /nh /fi "imagename eq serial2xsbug.exe" | (find /i "serial2xsbug.exe" > nul) && taskkill /f /t /im "serial2xsbug.exe" >nul 2>&1
```

...to...

```
KILL_SERIAL_2_XSBUG =
```

...so that serial2xsbug is left as-is as intended?

NextWorldLab, 2023-09-14:
https://github.com/Moddable-OpenSource/moddable/pull/1211

I did a PR. Thank you.

phoddie, 2023-09-14:
Thank you for the PR. It is really great you worked through that. We'll have the merged in a day or so.

Regarding my comment above about "kill serial2xsbug" -- that was mistaken. There were changes to not kill `xsbug`. That was unrelated to `serial2xsbug`.

mkellner, 2023-09-14:
@NextWorldLab, which node-red version are you using?

NextWorldLab, 2023-09-14:
@mkellner san

I confirmed the operation with node-red 3.0.2 and 3.1.
3.1 is not yet supported by NodeRED Plugin, so I am making a pull request.

It can be used in the following ways
```
cd %USERPROFILE%\.node-red
npm install https://github.com/NextWorldLab/node-red-mcu-plugin
```

thank you,

mkellner, 2023-09-14:
Excellent. Thank you for the 3.1 patch.
I verified the fix. It will be in the next code drop.

closed 2023-09-15 by NextWorldLab