Moddable-OpenSource/moddable #1210: Can't compile in linux

hiperiondev, 2023-09-13:
Following: http://www.moddable.com/documentation/Moddable%20SDK%20-%20Getting%20Started#linux

after:
cd $MODDABLE/build/makefiles/lin
make

error:

# xsc piu/All.xsb
make[1]: xsc: No existe el fichero o el directorio
make[1]: *** [/opt/Desarrollos/Moddable-Projects/moddable/build/tmp/lin/release/xsbug/makefile:401: /opt/Desarrollos/Moddable-Projects/moddable/build/tmp/lin/release/xsbug/modules/piu/All.xsb] Error 127
make[1]: se sale del directorio '/home/opt/Desarrollos/Moddable-Projects/moddable/tools/xsbug'
make: *** [makefile:51: release] Error 2

mkellner, 2023-09-13:
Hello @hiperiondev, I do not have problems building for x86 Ubuntu 22.04.1 or 20.04.1, or Arm Linux (rPi 4).

Please make sure your repository is up to date, remove the bin and tmp directories and try again. If you have a problem, please send the full build output.

```
cd $MODDABLE
git pull --rebase
rm -rf $MODDABLE/build/bin $MODDABLE/build/tmp
cd $MODDABLE/build/makefiles/lin
make
```

phoddie, 2023-10-25:
Closing as no additional information has been provided. If the problem remains, please re-open.

closed 2023-10-25 by phoddie