Moddable-OpenSource/moddable #1213: Cannot build: fx_Math_idivmod’ undeclared here

tve, 2023-09-21:
**Build environment:** Arch Linux
**Moddable SDK version:** 4.1 
**Target device:** N/A

**Description**
```
# cc mc.xs.c
/home/sw/moddable/mod2/build/tmp/lin/release/xsbug/mc.xs.c:18102:132: error: ‘fx_Math_idivmod’ undeclared here (not in a function); did you mean ‘fx_Math_imod’?
18102 | .flag = 0x8b, .kind = XS_CALLBACK_X_KIND}, .value = { .callback = { fx_Math_idivmod, NULL } }},
      |                                                                     ^~~~~~~~~~~~~~~
      |                                                                     fx_Math_imod
```

**Steps to Reproduce**
1. git clone of repo
2. git checkout 4.1
3. cd build/makefiles/lin
4. make

phoddie, 2023-09-21:
@tve – nice to hear from you.

I think you just need to do a clean build of tools. The `Math.idivmod()` function was removed a little while back as part of an update to the integer math proposal. If you do a search of this repo, you will see it doesn't appear in any of the source code.

tve, 2023-09-21:
.. I got sucked into a 3D printing rabbit hole...

I see, you're right. I'm trying to build the latest version in a separate dir and I had the bin dir of the old version still in the path. Thanks for catching this!

closed 2023-09-21 by tve