Moddable-OpenSource/moddable #1216: Error when compiling QR code example on mac - file missing

kairos0ne, 2023-09-22:
**Build environment:** macOS
**Moddable SDK version:** 4.1-32-gc5845e51c
**Target device:** mac

**Description**
[image]

phoddie, 2023-09-23:
Thanks for the report. I'm able to reproduce the error. It looks like fallout from the recent set of manifest improvements. I'll post a fix in the next day or so.

phoddie, 2023-09-23:
The application manifest at `$MODDABLE/examples/data/qrcode/manifest.json` should also include `$(MODDABLE)/examples/manifest_commodetto.json`.With that in place, it builds and runs as expected. I'll commit that and it should be available in this repo in the coming days.

FWIW – `manifest_commodetto.json` is a required include for projects using Commodetto and Poco directly. For projects using Piu (which uses Commodetto and Poco), it is not necessary to include `manifest_commodetto.json` as the necessary pieces are included by `manifest_piu,json`, which is required for projects using Piu.

phoddie, 2023-09-29:
Fix committed. Closing.

closed 2023-09-29 by phoddie