Moddable-OpenSource/moddable #1223: XS fails to cache `next` for array-destructuring

gibson042, 2023-10-02:
**Environment:** XS 14.2.0 32 4

**Description**
Array-destructuring does not cache an iterator's `next` method as required by the ECMAScript specification.

**Steps to Reproduce**
```js
Object.defineProperty(Array.prototype, Symbol.iterator, {
  get() {
    print("get Symbol.iterator", this);
    return function makeIterator() {
      print("@@iterator", this, ...arguments);
      let i = 0;
      const src = this;
      return {
        [Symbol.iterator]: () => iterator,
        get next() {
          print("get", "next");
          return () => {
            if (i >= src.length) return { done: true, value: undefined };
            const value = src[i];
            print("yield", i, value);
            i += 1;
            return { done: false, value };
          };
        },
      };
    };
  },
});
const [x, y] = ["a", "b", "c"];
```

**Actual behavior**
```
get Symbol.iterator a,b,c
@@iterator a,b,c
get next
yield 0 a
get next
yield 1 b
```

**Expected behavior**
```
get Symbol.iterator a,b,c
@@iterator a,b,c
get next
yield 0 a
yield 1 b
```

[DestructuringAssignmentEvaluation](https://tc39.es/ecma262/multipage/ecmascript-language-expressions.html#sec-runtime-semantics-destructuringassignmentevaluation) on an |ArrayAssignmentPattern| with argument _value_ (the iterable responsible for providing values) is specified to always start with GetIterator(_value_, <tt>SYNC</tt>), which gets the Symbol.iterator method of _value_ and uses [GetIteratorFromMethod](https://tc39.es/ecma262/multipage/abstract-operations.html#sec-getiteratorfrommethod) to invoke it and return a corresponding Iterator Record capturing the resulting iterator **and** its `next` method for use by [IteratorDestructuringAssignmentEvaluation](https://tc39.es/ecma262/multipage/ecmascript-language-expressions.html#sec-runtime-semantics-iteratordestructuringassignmentevaluation). But XS incorrectly gets the `next` method before each step (and fails to get it at all for `const [] = …`).

bakkot, 2023-10-03:
Note that the current behavior (at least for this test) was correct prior to https://github.com/tc39/ecma262/pull/1021.

phoddie, 2023-10-03:
Thanks for the test case. It is a little odd that this isn't covered by test262, but that's a different discussion.

We have a fix for this that will be part of Modable SDK 4.2.1, likely later this week.

phoddie, 2023-10-10:
Closing, as the fix for this is live.