Moddable-OpenSource/moddable #1226: Crash on IO Serial constructor

meganetaaan, 2023-10-03:
**Build environment:** Linux (22.04)
**Moddable SDK version:** v4.2.0
**Target device:** M5Stack Basic

**Description**
The application crashes when instantiating the Serial class of TC53 IO. As a result, examples/io/serial/echo keeps rebooting.

**Steps to Reproduce**
1. `cd $MODDABLE/examples/io/serial/echo`
2. `mcconfig -m -i -p esp32/m5stack`
3. See error

**Expected behavior**
`Serial` correctly instantiates and work.

**Other information**
Below is the core dump printed via instrumental build.

<details>

```
Executing action: monitor
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Running idf_monitor in directory /home/sskw/.local/share/moddable/build/tmp/esp32/m5stack/instrument/echo/xsProj-esp32
Executing "/home/sskw/.espressif/python_env/idf5.1_py3.10_env/bin/python /home/sskw/.local/share/esp32/esp-idf-v5.1.1/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --decode-coredumps info --target esp32 --revision 0 /home/sskw/.local/share/moddable/build/tmp/esp32/m5stack/instrument/echo/xsProj-esp32/build/xs_esp32.elf -m '/home/sskw/.espressif/python_env/idf5.1_py3.10_env/bin/python' '/home/sskw/.local/share/esp32/esp-idf-v5.1.1/tools/idf.py' '-n'"...
--- esp-idf-monitor 1.3.0 on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:344
load:0x40078000,len:14500
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3284
entry 0x40080584
I (36) cpu_start: Multicore app
I (36) cpu_start: Pro cpu up.
I (36) cpu_start: Starting app cpu, entry point is 0x40080e7c
0x40080e7c: call_start_cpu1 at /home/sskw/.local/share/esp32/esp-idf-v5.1.1/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (50) cpu_start: Pro cpu start user code
I (50) cpu_start: cpu freq: 240000000 Hz
I (50) cpu_start: Application information:
I (52) cpu_start: Project name:     xs_esp32
I (56) cpu_start: App version:      4.2.0-dirty
I (60) cpu_start: Compile time:     Oct  3 2023 12:09:52
I (65) cpu_start: ELF file SHA256:  8eb129f0dd5f84f7...
I (70) cpu_start: ESP-IDF:          v5.1.1
I (74) cpu_start: Min chip rev:     v0.0
I (77) cpu_start: Max chip rev:     v3.99 
I (81) cpu_start: Chip rev:         v3.0
I (85) heap_init: Initializing. RAM available for dynamic allocation:
I (91) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (96) heap_init: At 3FFB9DB8 len 00026248 (152 KiB): DRAM
I (101) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (107) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (112) heap_init: At 400914DC len 0000EB24 (58 KiB): IRAM
I (118) spi_flash: detected chip: generic
I (121) spi_flash: flash io: qio
W (124) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
[... 223 lines cut ...]
Entering gdb stub now.
$T0b#e6GNU gdb (esp-gdb) 12.1_20221002
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=xtensa-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/sskw/.local/share/moddable/build/tmp/esp32/m5stack/instrument/echo/xsProj-esp32/build/xs_esp32.elf...
Remote debugging using /dev/ttyUSB0
warning: multi-threaded target stopped without sending a thread-id, using first non-exited thread
0x4010d659 in uart_enable_intr_mask (
    uart_num=<optimized out>, enable_mask=257)
    at /home/sskw/.local/share/esp32/esp-idf-v5.1.1/components/driver/uart/uart.c:356
356	    p_uart_obj[uart_num]->rx_int_usr_mask |= enable_mask;
(gdb) 
```

meganetaaan, 2023-10-03:
```
================== CURRENT THREAD REGISTERS ===================
exccause       0x1c (LoadProhibitedCause)
excvaddr       0xb0
```
```
#0  0x4010d659 in uart_enable_intr_mask (uart_num=<optimized out>, enable_mask=257) at /home/sskw/.local/share/esp32/esp-idf-v5.1.1/components/driver/uart/uart.c:356
```

I think this is a kind of null pointer exception.
`p_uart_obj[uart_num]` is not initialized below.

```
esp_err_t uart_enable_intr_mask(uart_port_t uart_num, uint32_t enable_mask)
{
    ESP_RETURN_ON_FALSE((uart_num < UART_NUM_MAX), ESP_FAIL, UART_TAG, "uart_num error");
    UART_ENTER_CRITICAL(&(uart_context[uart_num].spinlock));
    /* Keep track of the interrupt toggling. In fact, without such variable,
     * once the RX buffer is full and the RX interrupts disabled, it is
     * impossible what was the previous state (enabled/disabled) of these
     * interrupt masks. Thus, this will be very particularly handy when
     * emptying a filled RX buffer. */
    p_uart_obj[uart_num]->rx_int_usr_mask |= enable_mask; // <- Here
    uart_hal_clr_intsts_mask(&(uart_context[uart_num].hal), enable_mask);
    uart_hal_ena_intr_mask(&(uart_context[uart_num].hal), enable_mask);
    UART_EXIT_CRITICAL(&(uart_context[uart_num].spinlock));
    return ESP_OK;
}
```

The current serial.c has [stopped using the inline copy of uart functions](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/io/serial/esp32/serial.c#L442), despite the discussion suggesting its use in [Issue #931](https://github.com/Moddable-OpenSource/moddable/issues/931). It appears that the same problem is still occurring. Is that correct?

phoddie, 2023-10-03:
Thank you for the analysis. I'm not sure how that change was disabled. After re-enabling, some small changes were needed to get it building. But then it works. We'll get that merged.

phoddie, 2023-11-22:
@ralphwetzel, yes it is working. But, what your are trying won't work as-is. Let me explain.

The default serial port on Moddable is on pins 1 and 3. You can confirm that in xsbug:

[image]

Those are the same two pins are used for by the debugging communication in a debug build. When the serial instance is created, it overrides the debugging connection and so xsbug stops responding. (We should change that so it throws an exception, so that you would have a better idea what is happening)

You have a couple options.

- Use an instrumented build (`mcconfig -d -m -p esp32/moddable_two`). Since there is no debugging, there is no conflict. The ESP32 serial monitor lets you enter text so you see that it is working.
- Change the pins for echo's serial port to something else. Of course, then to test you will need to connect to those pins with a serial monitor. That would look like this:

```
	let serial = new device.io.Serial({
		...device.Serial.default,
		receive: YOUR_RECEIVE_PIN,
		transmit: YOUR_TRANSMIT_PIN,
		baud: 115200,
		...
```

phoddie, 2023-11-22:
I added code to reserve the serial pins when debugging. That causes an exception to be thrown when when the serial constructor tries to use the debug serial port. That seems better. I'll get that committed to help minimize confusion in the future.

[image]

ralphwetzel, 2023-11-22:
Hi Peter!
Thank you very much for this explanation.
Makes perfectly sense & I wasn't aware of it at all...

phoddie, 2023-11-22:
I added code to reserve the serial pins when debugging. That causes an exception to be thrown when when trying to reuse the serial port. That seems better. I'll get that committed to help minimize confusion in the future.

[image]

phoddie, 2023-11-22:
(Closing as this has been fixed)

closed 2023-11-22 by phoddie