Moddable-OpenSource/moddable #1227: M5Stack Core2 panics on startup

meganetaaan, 2023-10-03:
**Build environment:** Linux(22.04)
**Moddable SDK version:** v4.2.0
**Target device:** M5Stack Core2

**Description**

M5Stack Core2 panics on startup.
According to the instrument log, it seems to be a kind of deprecation error for i2s driver.

```
W (613) ADC: legacy driver is deprecated, please migrate to `esp_adc/adc_oneshot.h`
E (620) i2s(legacy): CONFLICT! The new i2s driver can't work along with the legacy i2s driver

abort() was called at PC 0x400fca4c on core 0
0x400fca4c: check_i2s_driver_conflict at /home/sskw/.local/share/esp32/esp-idf-v5.1.1/components/driver/deprecated/i2s_legacy.c:1939 (discriminator 3)
```

**Steps to Reproduce**
1. `cd $MODDABLE/examples/helloworld`
2. `mcconfig -m -i -p esp32/m5stack_core2`
3. See error

**Other information**

<details>
<summary>Full Instrumentation log</summary>

```
Executing action: monitor
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Running idf_monitor in directory /home/sskw/.local/share/moddable/build/tmp/esp32/m5stack_core2/instrument/helloworld/xsProj-esp32
Executing "/home/sskw/.espressif/python_env/idf5.1_py3.10_env/bin/python /home/sskw/.local/share/esp32/esp-idf-v5.1.1/tools/idf_monitor.py -p /dev/ttyUSB0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --decode-coredumps info --target esp32 --revision 300 /home/sskw/.local/share/moddable/build/tmp/esp32/m5stack_core2/instrument/helloworld/xsProj-esp32/build/xs_esp32.elf -m '/home/sskw/.espressif/python_env/idf5.1_py3.10_env/bin/python' '/home/sskw/.local/share/esp32/esp-idf-v5.1.1/tools/idf.py' '-n'"...
--- esp-idf-monitor 1.3.0 on /dev/ttyUSB0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
I (13) cpu_start: Multicore app
I (13) quad_psram: This chip is ESP32-D0WD
I (13) esp_psram: Found 8MB PSRAM device
I (13) esp_psram: Speed: 80MHz
I (15) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
W (21) esp_psram: Virtual address not enough for PSRAM, map as much as we can. 4MB is mapped
I (29) cpu_start: Pro cpu up.
I (32) cpu_start: Starting app cpu, entry point is 0x40080eec
0x40080eec: call_start_cpu1 at /home/sskw/.local/share/esp32/esp-idf-v5.1.1/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (525) esp_psram: SPI SRAM memory test OK
I (533) cpu_start: Pro cpu start user code
I (533) cpu_start: cpu freq: 240000000 Hz
I (533) cpu_start: Application information:
I (533) cpu_start: Project name:     xs_esp32
I (537) cpu_start: App version:      4.2.0-dirty
I (542) cpu_start: Compile time:     Oct  3 2023 12:27:08
I (547) cpu_start: ELF file SHA256:  9cfbe59df6d7b9ca...
I (552) cpu_start: ESP-IDF:          v5.1.1
I (556) cpu_start: Min chip rev:     v3.0
I (559) cpu_start: Max chip rev:     v3.99 
I (563) cpu_start: Chip rev:         v3.0
I (567) heap_init: Initializing. RAM available for dynamic allocation:
I (573) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (578) heap_init: At 3FFB4AC0 len 0002B540 (173 KiB): DRAM
I (584) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (589) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (595) heap_init: At 4008E644 len 000119BC (70 KiB): IRAM
I (600) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
I (607) spi_flash: detected chip: generic
I (610) spi_flash: flash io: qio
[... 118 lines cut ...]
Entering gdb stub now.
$T0b#e6GNU gdb (esp-gdb) 12.1_20221002
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-linux-gnu --target=xtensa-esp-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/sskw/.local/share/moddable/build/tmp/esp32/m5stack_core2/instrument/helloworld/xsProj-esp32/build/xs_esp32.elf...
Remote debugging using /dev/ttyUSB0
warning: multi-threaded target stopped without sending a thread-id, using first non-exited thread
0x400815f9 in panic_abort (details=0x3ffe3b80 "abort() was called at PC 0x400fca4c on core 0") at /home/sskw/.local/share/esp32/esp-idf-v5.1.1/components/esp_system/panic.c:452
452	    *((volatile int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
(gdb) 
```

</details>

mkellner, 2023-10-03:
Could you please remove build output and try again with a verbose build?
```
rm -rf $MODDABLE/build/bin/esp32 $MODDABLE/build/tmp/esp32
VERBOSE=1 mcconfig -d -m -p esp32/m5stack_core2 > $MODDABLE/build/tmp/debug.output 2>&1
```

Then send `$MODDABLE/build/tmp/debug.output` to me?

Thank you. Are you able to build for other targets?

phoddie, 2023-10-03:
I think the key thing is here: `check_i2s_driver_conflict`.

meganetaaan, 2023-10-03:
> Could you please remove build output and try again with a verbose build?
> 
> ```
> rm -rf $MODDABLE/build/bin/esp32 $MODDABLE/build/tmp/esp32
> VERBOSE=1 mcconfig -d -m -p esp32/m5stack_core2 > $MODDABLE/build/tmp/debug.output 2>&1
> ```
> 
> Then send `$MODDABLE/build/tmp/debug.output` to me?

Here it is. I put instrumental build version (debug_instrument.output) aside just in case.
[debug.zip](https://github.com/Moddable-OpenSource/moddable/files/12788592/debug.zip)

>Are you able to build for other targets?

`esp32/m5stack` and `esp32/m5stack_fire` build and run. AudioOut can play sound.

`esp32/m5stack_cores3` builds and runs but no sound plays. I will take a look if it's related to this issue.

meganetaaan, 2023-10-03:
I've delved into the implementation of audioout.c. As the error message suggests, I checked for any remaining legacy APIs for I2S. It appears that the i2s_write_expand function does not exist in esp-idf v5, which seems to be leading to a runtime error.

https://github.com/Moddable-OpenSource/moddable/blame/public/modules/pins/i2s/audioout.c#L1602

To verify this, I modified esp32/m5stack_core2/manifest.json and changed MODDEF_AUDIOOUT_I2S_BITSPERSAMPLE to 16 to bypass the use of i2s_write_expand. Doing so eliminates the relevant error, and audio playback proceeds as expected.

phoddie, 2023-10-03:
It looks like the M5Stack Core2 is the only public device that uses 32-bit output. That's why the use of the deprecated `i2s_write_expand` API was missed. The expansion has been added to the new channel API as part of initialization. Unfortunately, I don't have an M5Stack Core2 to try. Would you test the following changes in audioOut.c?

In `audioOutLoop()` set the 32-bit case to expand from 16-bit source to 32-bit out:

```c
#if MODDEF_AUDIOOUT_I2S_BITSPERSAMPLE == 32
	i2s_config.slot_cfg.data_bit_width = I2S_DATA_BIT_WIDTH_16BIT;
	i2s_config.slot_cfg.ws_width = I2S_DATA_BIT_WIDTH_32BIT;
```

In the same function, fold together the 16 and 32 bit cases, so both use `i2s_channel_write`:

```c
#elif (16 == MODDEF_AUDIOOUT_I2S_BITSPERSAMPLE) || (32 == MODDEF_AUDIOOUT_I2S_BITSPERSAMPLE)
		i2s_channel_write(out->tx_handle, (const char *)out->buffer, sizeof(out->buffer), &bytes_written, portMAX_DELAY);
#else
	#error invalid MODDEF_AUDIOOUT_I2S_BITSPERSAMPLE
#endif
```

It would be good to have this case working for other devices that may need it. Still, it may be just fine on M5Stack Core2 to use the 16 bit case.

Thank you!

meganetaaan, 2023-10-06:
I applied the change above and the sound is back! Thank you for your quick fix.

https://github.com/Moddable-OpenSource/moddable/assets/1918755/6f375541-191c-4b26-9e84-f24c7e75e0aaf

closed 2023-11-24 by phoddie