Moddable-OpenSource/moddable #1230: mcrun don't detect UPLOAD_PORT

stc1988, 2023-10-08:
**Build environment:** macOS
**Moddable SDK version:** https://github.com/Moddable-OpenSource/moddable/commit/d0a90fa1b4740ce7516aae5d60c6c356a877b69d
**Target device:** M5stack

**Steps to Reproduce**
1. Build and install  host app `examples/piu/outline/figures` using this build command: `mcconfig -d -m  -p esp32/m5stack`
2. Build and install  mod app `examples/piu/outline/figures/CanvasPath`  using this build command: `mcrun -d -m  -p esp32/m5stack`
3. mcrun don't detect UPLOAD_PORT, so installing mod don't finish.

```
mcrun -d -m -p esp32/m5stack
/Users/satoshi/.local/share/esp32/esp-idf/tools/idf.py
## XSBUG_PORT=5002 XSBUG_HOST=localhost serial2xsbug /dev/cu.usbserial-0001 460800 8N1 -install /Users/Satoshi/Projects/moddable/build/bin/esp32/debug/CanvasPath/CanvasPath.xsa
```

My device is recognized as `/dev/cu.usbserial-01CD8DA8`.

phoddie, 2023-10-09:
Thanks for the report. We can see the problem and are exploring solutions.

mkellner, 2023-10-11:
We have made some changes to the method that `mcrun` uses to find a serial port on macOS and Linux.

The `esptool.py` tool from Espressif is used to probe for a serial port. This is new and will resolve the problem described here.

If the environment variable `UPLOAD_PORT` is set, `mcrun` will use it.

If the device connects over USB, `mcrun` will continue to use Vendor ID and Product ID to identify the device.

Please give it a try and let us know if you have any problems.

stc1988, 2023-10-12:
I confirmed this issue fixed in Moddable SDK 4.2.1.

closed 2023-10-12 by stc1988