Moddable-OpenSource/moddable #1232: import failed in mod

stc1988, 2023-10-10:
**Build environment:** macOS
**Moddable SDK version:** https://github.com/Moddable-OpenSource/moddable/commit/d0a90fa1b4740ce7516aae5d60c6c356a877b69d
**Target device:** M5stack

**Steps to Reproduce**
1. Build and install  host app `contributed/ble/hid-peripheral/host` using this build command: `mcconfig -d -m  -p mac`
2. Build and install  mod app `contributed/ble/hid-peripheral/mod-media`  using this build command: `mcrun -d -m  -p mac`
3. I got below error in xsbug

```
/Users/satoshi/Projects/moddable/contributed/ble/hid-peripheral/host/main.js (29) # Break: importNow: import VolumeDownSkin not found!
```

I got similar errors in other mods `mod-keyboard` and `mod-combo`
`/Users/satoshi/Projects/moddable/contributed/ble/hid-peripheral/host/main.js (29) # Break: importNow: import BottomRowTexture not found!
`

phoddie, 2023-10-10:
Interesting. I get the same error. The assets module exports appears to be correct:

[image]

If I remove the import and use of `VolumeDownSkin`, the mod launches normally.

We'll investigate Thank you.

phoddie, 2023-10-11:
We tracked down the problem to a long-standing bug in the linker that only triggers when building mods. It was difficult to trigger, however. This particular issue likely only appeared because of another recent fix (#1209) to the linker. The fix will be included in Moddable SDK 4.2.1. It will be necessary to do a clean rebuild of the mod.

stc1988, 2023-10-12:
I confirmed this issue fixed in Moddable SDK 4.2.1.

closed 2023-10-12 by stc1988