Moddable-OpenSource/moddable #1236: No AudioOut sounds with M5Stack CoreS3

meganetaaan, 2023-10-23:
**Build environment:** Linux
**Moddable SDK version:** v4.2.0, v4.2.1
**Target device:** M5Stack CoreS3

**Description**
With M5Stack CoreS3 and the ModdableSDK v4.2.1, no sound can be played through the AudioOut class, leaving a slight noise at the beginning.

**Steps to Reproduce**
1. Build and install `piu/sounds` example app using `mcconfig -m -p esp32/m5stack`.
2. The device cannot play any sound including startup sound(bflatmajor.maud) .

**Other information**
- The sound works fine on M5Stack Basic and Core2 with Moddable v4.2.1
- The issue seems similar to what is described in https://github.com/Moddable-OpenSource/moddable/issues/881. I am currently investigating whether this is a degradation on the esp-idf side or if there is something problematic in the CoreS3 initialization code that I wrote.

meganetaaan, 2023-10-30:
I've been progressing with debugging, but I still haven't seen a solution. As I continue to read through the esp-idf code, I'm starting to think that the cause might be the incorrect calculation of clock division, which was mentioned in a previous discussion.
https://github.com/Moddable-OpenSource/moddable/discussions/1134#discussioncomment-6165089

Indeed, regarding this calculation process, there have been changes from esp-idf v4 to v5, and it seems that even if the same sample rate is given, the results are different.
v5.1: https://github.com/espressif/esp-idf/blob/release/v5.1/components/hal/i2s_hal.c#L36
v4.4: https://github.com/espressif/esp-idf/blob/release/v4.4/components/hal/i2s_hal.c#L28

However, I'm not sure how to identify the cause from here or how to fix it, and I'm stuck. I would appreciate assistance from anyone with knowledge of esp-idf and who has the CoreS3.

phoddie, 2023-10-31:
Thank you for working on this issue. I hoped that IDF v5 would bring an end to these kinds of audio problems.

There isn't much to configure for the I2S audio output:
- The I2S pins must be correct (the hardware didn't change and it worked before!)
- It must still be 16-bit audio (right?)
- The M5Stack Core S3 is the **only** device that's manifest uses the `mck` pin (set to 0, which is a little suspicious). Is that really required? Could it actually be causing the problem somehow?
- The manifest sets `MODDEF_AUDIOOUT_I2S_NUM` to 1. That corresponds to the ESP-IDF "I2S Port id". I think `1` is OK, but I do remember that some ESP family silicon has restrictions on certain I2S ports. Maybe try `0`?

Sorry, I don't have a better idea right now. I would also suggest trying I2S audio output on a different ESP32-S3 board, to confirm that it works correctly (it should, I know we tested this!). Maybe @mkellner can help with that.

stc1988, 2024-01-24:
@meganetaaan     
I think this issue fixed by #1297.  Could you confirm this?

meganetaaan, 2024-01-30:
@stc1988
I confirmed that the sound has been restored!
Thanks to this, I can now keep Stack-chan (the product I'm developing) up-to-date. Thank you so much!

closed 2024-01-30 by phoddie