Moddable-OpenSource/moddable #1237: Program Hangs When Drawing Outline on M5Stack CoreS3

meganetaaan, 2023-10-23:
**Build Environment:** Linux  
**Moddable SDK Version:** v4.2.1  
**Target Device:** M5Stack CoreS3  

**Description**

When calling `poco.end()` after drawing an outline using `poco.blendOutline()`, the program hangs and no debug logs are output.

**Steps to Reproduce**
1. Build and install the sample app located at `examples/commodetto/outline/random-ellipses` using this build command: `mcconfig -m -d -p esp32/m5stack_cores3`
2. Program hangs up

**Other Information**  
- With M5Stack Basic/Core2 outline works as expected. This only happens with M5Stack CoreS3

phoddie, 2023-10-23:
Thank you for the report. This one is strange. Here's what I found:

- The problem happens on ESP32-S3 (I tried a different board than you used)
- It is not a result of somewhat recent garbage-collection changes in the outline code. Removing those makes no difference.
- It is not a native stack overflow -- doubling the stack size didn't help
- The random-rectangles example works, but anything more complex than that seems to fail.
- The shape being drawn is not very complex, so it doesn't seem resource related.
- It appears to hang in the FreeType renderer -- running an instrumented build the watch-dog timer eventually fires
- Perhaps it is a code generation bug in the tools for ESP32-S3? I'm not quite sure how to look into that.

meganetaaan, 2023-10-24:
After digging deeper using printf debugging, I found that the hang occurs at the following line:

https://github.com/Moddable-OpenSource/moddable/blob/cdcab496ffa3108cbb4fe48bc284dd630984c9f6/modules/commodetto/outline/freetype/src/ftgrays.c#L1703

The actual implementation of this MEM_ZERO macro is the regular memset function.

https://github.com/Moddable-OpenSource/moddable/blob/cdcab496ffa3108cbb4fe48bc284dd630984c9f6/modules/commodetto/outline/freetype/src/ftgrays.c#L171

I suspected that, in the case of the S3, the address for memory allocation (dest) or the size of allocation (count) might have incorrect values, so I tried the following:

```c
      printf("mem zero begin: %p, %d\n", ras.ycells, height);
      FT_MEM_ZERO(ras.ycells, height * sizeof(PCell));
      printf("mem zero end\n");
```

I've attached the log results for both the successful case (M5Stack Basic) and the case where it hangs (M5Stack CoreS3). Hope this will give us some insight.

<details>
<summary>M5Stack Basic</summary>

```
Executing action: monitor
Serial port /dev/ttyACM0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Running idf_monitor in directory /home/ishikawa/.local/share/moddable/build/tmp/esp32/m5stack/instrument/random-ellipses/xsProj-esp32
Executing "/home/ishikawa/.espressif/python_env/idf5.1_py3.10_env/bin/python /home/ishikawa/.local/share/esp32/esp-idf/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32-elf- --decode-coredumps info --target esp32 --revision 0 /home/ishikawa/.local/share/moddable/build/tmp/esp32/m5stack/instrument/random-ellipses/xsProj-esp32/build/xs_esp32.elf -m '/home/ishikawa/.espressif/python_env/idf5.1_py3.10_env/bin/python' '/home/ishikawa/.local/share/esp32/esp-idf/tools/idf.py' '-n'"...
--- esp-idf-monitor 1.3.1 on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:344
load:0x40078000,len:14500
load:0x40080400,len:4
0x40080400: _init at ??:?

load:0x40080404,len:3284
entry 0x40080584
I (35) cpu_start: Multicore app
I (35) cpu_start: Pro cpu up.
I (35) cpu_start: Starting app cpu, entry point is 0x40080df4
0x40080df4: call_start_cpu1 at /home/ishikawa/.local/share/esp32/esp-idf/components/esp_system/port/cpu_start.c:154

I (0) cpu_start: App cpu up.
I (49) cpu_start: Pro cpu start user code
I (49) cpu_start: cpu freq: 240000000 Hz
I (49) cpu_start: Application information:
I (51) cpu_start: Project name:     xs_esp32
I (55) cpu_start: App version:      4.2.1-dirty
I (59) cpu_start: Compile time:     Oct 24 2023 14:02:10
I (64) cpu_start: ELF file SHA256:  dd2cd358473d25a1...
I (69) cpu_start: ESP-IDF:          v5.1.1
I (73) cpu_start: Min chip rev:     v0.0
I (76) cpu_start: Max chip rev:     v3.99 
I (80) cpu_start: Chip rev:         v3.0
I (84) heap_init: Initializing. RAM available for dynamic allocation:
I (90) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (95) heap_init: At 3FFB49E8 len 0002B618 (173 KiB): DRAM
I (100) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (106) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (111) heap_init: At 4008F570 len 00010A90 (66 KiB): IRAM
I (117) spi_flash: detected chip: generic
I (120) spi_flash: flash io: qio
[... 75 lines cut ...]
I (2007) heap_init: At 600FE014 len 00001FD4 (7 KiB): RTCRAM
I (2013) esp_psram: Adding pool of 8192K of PSRAM memory to heap allocator
I (2020) spi_flash: detected chip: generic
I (2023) spi_flash: flash io: dio
I (2026) sleep: Configure to isolate all GPIO pins in sleep state
I (2032) sleep: Enable automatic switching of GPIO sleep configuration
I (2038) app_start: Starting scheduler on CPU0
I (2042) app_start: Starting scheduler on CPU1
I (2042) main_task: Started on CPU0
I (2050) main_task: Calling app_main()
instruments key: Pixels drawn,Frames drawn,Network bytes read,Network bytes written,Network sockets,Timers,Files,Poco display list used,Piu command List used,SPI flash erases,Turns,System bytes free,CPU 0,CPU 1,Chunk used,Chunk available,Slot used,Slot available,Stack used,Stack available,Garbage collections,Keys used,Modules loaded,Parser used,Floating Point,Promises settled
No Wi-Fi SSID
I (2096) main_task: Returned from app_main()
I (2127) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
I (2128) gpio: GPIO[35]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 
instruments: 0,0,0,0,0,0,0,0,0,0,0,8526820,54,17,1200,78848,2608,78832,1552,8192,0,0,0,0,4,0
E (3086) i2s_common: i2s_channel_disable(1030): the channel has not been enabled yet
instruments: 0,0,0,0,0,1,0,0,0,0,0,8532924,0,0,1200,78848,2640,78832,608,8192,0,0,0,0,0,0
mem zero begin: 0x4037bffc, 10
0x4037bffc: espRead32 at ??:?
```

</details>

Next I would like to verify if the '10 bytes from 0x4037bffc' is a valid memory address for my M5Stack CoreS3

phoddie, 2023-10-24:
Nice debugging work! The problem seems to be trigged by this line:

https://github.com/Moddable-OpenSource/moddable/blob/cdcab496ffa3108cbb4fe48bc284dd630984c9f6/modules/commodetto/outline/freetype/src/ftgrays.c#L1648

That puts the render buffer in IRAM. That works on ESP32 but not ESP32-S3 Apparently something changed in the memory architecture. The problem goes away by simply deleting `ICACHE_RAM_ATTR`. I confirmed this with several of the outline examples. Please confirm the results yourself. 

The advantages of using ICACHE_RAM_ATTR are that it is always fast memory (on chip, never PSRAM) and it is often underused in projects -- so it frees up a little heap space. But, maybe it is too fragile here so we should not try. I'm not sure.

phoddie, 2023-10-25:
ESP32 and ESP32-S3 manage IRAM differently.
- ESP32 has a fixed size area of memory for IRAM ([docs](https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32/api-guides/memory-types.html)).
- ESP32-S3 adjusts the IRAM allocation based on the firmware configuration; any remaining space is used for DRAM ([docs](https://docs.espressif.com/projects/esp-idf/en/release-v5.1/esp32s3/api-guides/memory-types.html)).

One reason for using IRAM on ESP32 is that it is often not entirely filled, so it is kind of "free memory". Because ESP32-S3 can adjust the amount of iRAM and DRAM, there is no such advantage. Statically allocated data goes into internal (fast) memory by default, so performance should be the same.

So... it seems like this memory optimization is helpful on ESP32, but is otherwise unnecessary. Let's conditionalize it to only be used on the original ESP32 and not other members of the ESP32 family:

```c
#if ESP32 && kCPUESP32
    static ICACHE_RAM_ATTR TCell    buffer[FT_MAX_GRAY_POOL];		//@@ not thread safe, but avoids having to allocate a big stack
```

Note: I'm still unsure why this failed on ESP32-S3. Maybe someone can explain.

meganetaaan, 2023-10-26:
Your solution worked!

```c
#if ESP32 && kCPUESP32
    static ICACHE_RAM_ATTR TCell    buffer[FT_MAX_GRAY_POOL];		//@@ not thread safe, but avoids having to allocate a big stack
```

The Stack-chan face is back to my M5Stack CoreS3 :+1: 

[image]

I'm also not sure about the cause of the memory failure but I'm glad we found a reasonable solution.

phoddie, 2023-10-26:
Thank you for the good news. And the nice photo. I hoped Stack-chan would smile about our success, but I understand that is not their personality. ;)

(We'll get this fix live in a the next day or three. Thank you for your excellent debugging work to help isolate the problem.)

phoddie, 2023-11-24:
Closing as this is now working.

closed 2023-11-24 by phoddie