Moddable-OpenSource/moddable #1238: An issue about parsing URL with regular expression

Jinyn01, 2023-10-23:
**Description**:

When I parse URLs with a complex regular expression, the result of XS is Killed,  which looks like an issue.

**Environment**：

```
Ubuntu 18.04 x86_64
```

**JS Information**:

```
Moddable SDK 3.7.0
```

**Testcase**:

```public class Test {
const regex = /^(?:(?![^:@]+:[^:@\/]*@)([^:\/?#.]+):)?(?:\/\/\/?)?((?:(([^:@]*):?([^:@]*))?@)?([^:\/?#]*)(?::(\d*))?)(((\/(?:[^?#](?![^?#\/]*\.[^?#\/.]+(?:[?#]|$)))*\/?)?([^?#\/]*))(?:\?([^#]*))?(?:#(.*))?)/
const str = 'https://www.example.com/path/to/resource?param1=value1&param2=value2#section';

const result = regex.exec(regex);
```

**Result**:

```
Killed
```

phoddie, 2023-10-23:
Thank you for the report. The problem reproduces here on macOS as well. It looks like the process is killed because it eventually allocates too much memory. 

Note that the title of this issue is incorrect. The test case does not parse a URL. The URL is in the variable `str` but that is unused when calling `exec`. Is this a test case from a fuzzing run?

Note also that this issue is reported against Moddable SDK 3.7.0. The [current release](https://github.com/Moddable-OpenSource/moddable/releases/) is 4.2.1.

FWIW - JavaScriptCore and V8 evaluate the regex quickly and without an error.

We will investigate.

phoddie, 2023-10-24:
Thank you for the clarification. The problem happens using either `str` or `regex` as the argument. We are testing a fix now. That should be available later in the week.

phoddie, 2023-10-24:
The fix for this is live now. Please give it a try.

phoddie, 2023-10-24:
Just FWIW -- the Moddable SDK includes an [implementation of the web's `URL` class](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/data/data.md#url). That might be a little more straightforward than a regular expression. But, do whatever works best for you.

Jinyn01, 2023-10-25:
I have tried the solution you provided and the problem has been successfully resolved. Thank you very much.

closed 2023-10-25 by Jinyn01

phoddie, 2023-10-25:
Excellent. Thanks again for reporting the issue. We're always glad to fix issues like this.