Moddable-OpenSource/moddable #1239: Build ssd1306 fails

stc1988, 2023-10-25:
**Build environment:** macOS
**Moddable SDK version:** c493992eab09b3c3e29f03ad28a575a63a261738
**Target device:**　esp32/heltec_lora_32,

**Steps to Reproduce**
Build   `examples/drivers/ssd1306` using this build command: `mcconfig -d -m -p esp32/heltec_lora_32 -t build`

```
# cc spi.c.o
/Users/satoshi/Projects/moddable/modules/drivers/ssd1306/modSsd1306.c: In function 'xs_SSD1306_destructor':
/Users/satoshi/Projects/moddable/modules/drivers/ssd1306/modSsd1306.c:190:43: error: 'configSPIandI2C' has no member named 'rst'
  190 |                 modGPIOUninit(&ssd->config.rst);
      |                                           ^
# cc digitalbank.c.o
# cc _i2c.c.o
make: *** [/Users/Satoshi/Projects/moddable/build/tmp/esp32/heltec_lora_32/debug/ssd1306/modSsd1306.c.o] Error 1
make: *** Waiting for unfinished jobs....
```

I get a similar error when build with `modules/drivers/ssd1306/manifest_spi.json`

```
# cc digitalbank.c.o
/Users/satoshi/Projects/moddable/modules/drivers/ssd1306/modSsd1306.c: In function 'xs_SSD1306_destructor':
/Users/satoshi/Projects/moddable/modules/drivers/ssd1306/modSsd1306.c:183:43: error: 'configSPIandI2C' has no member named 'cs'
  183 |                 modGPIOUninit(&ssd->config.cs);
      |                                           ^
/Users/satoshi/Projects/moddable/modules/drivers/ssd1306/modSsd1306.c:184:43: error: 'configSPIandI2C' has no member named 'dc'
  184 |                 modGPIOUninit(&ssd->config.dc);
      |                                           ^
/Users/satoshi/Projects/moddable/modules/drivers/ssd1306/modSsd1306.c:190:43: error: 'configSPIandI2C' has no member named 'rst'
  190 |                 modGPIOUninit(&ssd->config.rst);
      |                                           ^
make: *** [/Users/Satoshi/Projects/moddable/build/tmp/esp32/heltec_lora_32/debug/ssd1306/modSsd1306.c.o] Error 1
make: *** Waiting for unfinished jobs....
```

phoddie, 2023-10-25:
The destructor should be as follows:

```c
void xs_SSD1306_destructor(void *data)
{
	if (data) {
		ssd1606 ssd = (ssd1606)data;
#if MODDEF_SSD1306_SPI
		modGPIOUninit(&ssd->cs);
		modGPIOUninit(&ssd->dc);
		modSPIUninit(&ssd->config.spi);
#elif MODDEF_SSD1306_I2C
		modI2CUninit(&ssd->config.i2c);
#endif
#ifdef MODDEF_SSD1306_RST_PIN
		modGPIOUninit(&ssd->rst);
#endif
		c_free(data);
	}
}
```

We'll get that into the next update. Thank you for reporting.

PLCMercenary, 2023-11-04:
Ran into this using the Heltec Wifi 32 devkit this week also. I got things working by commenting the if block for the RST_PIN. Thanks for posting the correct destructor. Yours is likely a better solution

phoddie, 2023-11-04:
Glad it helped.

It is pretty rare for the display destructor to be called, since most embedded projects just keep the driver open forever. But, it should work correctly and it absolutely should compile. ;)

FYI – the patch is live (see commit above) and will be part of our November release in the coming days.

closed 2023-11-04 by phoddie