Moddable-OpenSource/moddable #1246: File not found error when creating a new file while running /examples/network/http/httpserverputfile

kallistisoft, 2023-11-15:
**Build environment:** Linux
**Moddable SDK version:** 4.2.0
**Target device:** ESP32-S3

**Description**
When uploading a file via curl the example program halts at line 24 with a file not found error.
`this.file = new File(path, true);`

**Steps to Reproduce**
1. Build and install the app using `mcconfig -d -m -p eps32/esp32s3`
2. Upload file using curl ex: `curl --data-binary "~/test.txt" http://192.168.1.199/test.txt -v`

**Expected behavior**
The file to be uploaded to the device

**Resolution**
Including `mc/config` and prefixing the path with `config.file.root` fixes the problem.
```
import config from "mc/config";

...

let path = config.file.root + value;
```

phoddie, 2023-11-15:
Thank you for the report and the proposed fix. Your change makes sense. I want to check it on ESP8266, just to be safe, before applying change (my guess is that ESP8266 may works without the change, which may be why we didn't catch this sooner).

phoddie, 2023-11-15:
Yes, this works correctly unchanged on ESP8266. This example was written originally on ESP8266. And, conveniently the ESP8266 uses "/" for `config.file.root`. Just concatenating them gives a double slash. The http path has a leading slash. So, for this to work everywhere (ESP8266, ESP32, Pico, macOS, etc)  the path should be calculated as follows:

```js
const path = config.file.root + value.slice(1);
```

And.... in testing, I found the `curl` line is incorrect. Here's a better line:

```
curl http://10.0.0.31/contributing.md --upload-file $MODDABLE/contributing.md -v
```

We'll get an update posted.

Thanks for bringing this up. It is often valuable to revisit examples.

kallistisoft, 2023-11-15:
> Thanks for bringing this up. It is often valuable to revisit examples.

You're welcome. Big fan the platform, happy to contribute :)

closed 2023-11-15 by kallistisoft