Moddable-OpenSource/moddable #1257: modMessageService triggers watchdog on ESP32

beckerzito, 2023-11-21:
> Note: This template is intended to help organize your report. Feel free to modify it as you need. The most important thing is to file a clear, detailed report.

> You may use this template to file a report of a security vulnerability. If you prefer to report the issue privately, please send the report by email to info@moddable.com.

**Build environment:** macOS, Windows, or Linux
**Moddable SDK version:** 4.1
**Target device:** NodeMCU ESP32

**Description**
[This commit ](https://github.com/Moddable-OpenSource/moddable/commit/dccc1bbd56d2d5b55c44d99f8743b2b8a57458d3) has changed the execution loop of the modMessageService. Before the change, the loop used to iterate only through the number of messages present on the queue at the beginning of the execution, retrieved by: `unsigned portBASE_TYPE count = uxQueueMessagesWaiting(the->msgQueue);`.

After the change, the loop happens based on the `while (xQueueReceive(the->msgQueue, &msg, maxDelayMS))` condition. Having said that, even though the `maxDelayMS` is set to 0, if the message callback execution is slow (performing flash operation for example), more messages are added to the queue (by a socket connection for example), the loop will never stop the execution and never yield to the main loop to give the opportunity to the timers to execute and the watchdog to be reset. 

**Steps to Reproduce**
1. Create an app to download data from a web page and write to the flash, for example, an OTA operation.

**Expected behavior**
The operation shall end with success in writing all the downloaded data to the flash. But instead, the application will reset due watchdog reset.

phoddie, 2023-11-21:
> Steps to Reproduce 
>
> Create an app to download data from a web page and write to the flash, for example, an OTA operation.

We have products that perform OTA on ESP32 without triggering the watchdog. Therefore, this one step appears to be insufficiently precise to allow your scenario to be reproduced. Would you please provide additional details? Even better, a working example that shows the problem would be most helpful. Thank you.

FWIW – the way `maxDelayMS` is used is imperfect, in that the loop can run longer than that. This behavior, however, does match what the ESP-IDF does with the same parameter in many situations. Perhaps this case could be changed to better respect the input value. But that exploration is best done with a testbed app that reproduces the behavior you see.

phoddie, 2023-11-24:
A little more on this. The callbacks should take priority over messages. Callbacks have a deadline, and should fire as close to that time as possible. Messages, by contrast, are analogous to the JavaScript job queue used for promises and so they run when nothing else is active.

The behavior you describe is the priorities being inverted -- messages are starving callbacks. That should be fixed and, I think, respecting `maxDelayMS` will help, it may not be enough by itself. Correcting that means that callbacks have some potential to starve messages, of course, since they take priority.

To avoid stressing the overall system, many modules in the Moddable SDK minimize messages sent. I'm not certain what's going on in your situation but it sounds like there may be a lot of messages in flight. If that can be reduced somehow, that's probably for the best since the ESP32 is fast, but not infinitely so.

phoddie, 2023-12-18:
@beckerzito - Thanks for the scenario. For that case, I see why the previous behavior would be preferred. Still, once the message queue is full, the worker machine is going to be blocked for a significant part of its execution time.

I'll try to get back to a behavior where the main machine can't spin forever in a scenario like this. Still, the system is unbalanced so all the SDK can do is make it less bad. Restoring balance is up to the developer (you).

phoddie, 2023-12-19:
I modified the base/worker example to implement the scenario you describe above. That spins forever handling messages, has expected. I had previously started changes to address this. Finishing that up gives a proof-of-concept that eliminates spinning forever. The implementation only processes the messages pending on entry and also handles the maximum delay correctly.

I need to construct some more test cases to make sure the behavior is reasonable in all scenarios, not just this one. The code is different for instrumented and debug builds (to allow debug builds to timeout, as previously discussed) and a dependency on the watchdog timer -- so a few variations to test.

phoddie, 2023-12-20:
Some more progress here.

- Confirmed that the behavior is also correct when the roles are reserved (swapped code in worker and main)
- Works for both debug & non-debug (instrumented, release) builds
- Behaves correctly when posting a message times out (debug builds only, by default)
- Debugger interactions continue to work correctly
- Tasted with much shorter values timing and random timing in main & worker - looking to stress the system
- Refined handling of max delay to better respect value passed in. (A callback can still overshoot this; if one does, timers are gives a chance to run before resuming message servicing)
- Moved the reset of the watchdog to immediately before invoking the callback to give callback a consistent amount of time for execution before triggering watchdog. This isn't strictly necessary but seems reasonable.

Also, reconfirmed that the timer code only services those timers eligible when the function is entered. That prevents the timer service from spinning forever. That's the same idea as limiting the messages serviced to the pending message count on entry. So, neither should ever completely starve the other.

The code is probably about ready to commit.

phoddie, 2023-12-23:
Code committed. Please give it a try.

beckerzito, 2024-01-15:
Sorry for the long delay @phoddie I'm gonna test that and will let you know! 
Thanks!

phoddie, 2024-01-15:
Great, thank you.

tve, 2024-01-16:
FYI, I believe I'm running into this same issue. What I see is that the machine deadlocks on mxMessagePostToMachine, coming from fxPromiseThen > fxQueueJob > fxQueuePromiseJobs. Are those the symptoms?

phoddie, 2024-02-07:
> I'm gonna test that and will let you know!

@beckerzito – any update?

beckerzito, 2024-03-28:
Hi @phoddie sorry for the long delay. Thanks for the support, it fixed the issue!

phoddie, 2024-03-28:
That's great to hear. Thanks for confirming.

closed 2024-03-28 by phoddie