Moddable-OpenSource/moddable #1258: contributed/giphy app failed to launch

stc1988, 2023-11-23:
**Build environment:** macOS
**Moddable SDK version:** [1918979](https://github.com/Moddable-OpenSource/moddable/commit/1918979ced6d36f3a4c380ade146c083beaeb084)
**Target device:** M5stack, M5stack core2

**Steps to Reproduce**
1.  modify config section in [manifest.json](https://github.com/Moddable-OpenSource/moddable/blob/public/contributed/giphy/manifest.json)
2. build `mcconfig -d -m -p esp32/m5stack`
3. see xsbug and nothing happened.

**Other information**
release instrumented build `mcconfig -i -m -p esp32/m5stack` shows error.

```
I (46) cpu_start: App cpu up.
I (548) esp_psram: SPI SRAM memory test OK
I (555) cpu_start: Pro cpu start user code
I (555) cpu_start: cpu freq: 240000000 Hz
I (555) cpu_start: Application information:
I (556) cpu_start: Project name:     xs_esp32
I (560) cpu_start: App version:      4.3.0-26-g27d226006-dirty
I (565) cpu_start: Compile time:     Nov 23 2023 11:05:20
I (570) cpu_start: ELF file SHA256:  a3b623e29643ae80...
I (575) cpu_start: ESP-IDF:          v5.1.1
I (579) cpu_start: Min chip rev:     v3.0
I (583) cpu_start: Max chip rev:     v3.99
I (587) cpu_start: Chip rev:         v3.0
I (591) heap_init: Initializing. RAM available for dynamic allocation:
I (597) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (602) heap_init: At 3FFB9E60 len 000261A0 (152 KiB): DRAM
I (607) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (613) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (618) heap_init: At 40091B18 len 0000E4E8 (57 KiB): IRAM
I (624) esp_psram: Adding pool of 4096K of PSRAM memory to heap allocator
I (630) spi_flash: detected chip: generic
I (634) spi_flash: flash io: qio
I (637) esp_core_dump_uart: Init core dump to UART
I (641) app_start: Starting scheduler on CPU0
I (645) app_start: Starting scheduler on CPU1
I (645) main_task: Started on CPU0
I (653) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (660) main_task: Calling app_main()
# Slot allocation: failed for 78848 bytes
XS abort: memory full
```

phoddie, 2023-11-24:
Your step 1 above doesn't say how to modify the manifest. I did this to eliminate the "memory full" error:

```json
	"creation":{
		"static": 69152,
		"chunk": {
			"initial": 8192,
			"incremental": 1024
		},
		"heap": {
			"initial": 512,
			"incremental": 64
		},
		"stack": 360
	}
```

This is necessary because of esp32 manifest changes to support PSRAM.

There is one other problem. ESP-IDF v5 does't allow calling `esp_partition_mmap` more than once for a given region. In IDF v4 this was allowed. The solution is to only instantiate the flash partition once in JavaScript. With the app works as usual. (We'll push this change shortly)

phoddie, 2023-11-24:
[Update](https://github.com/Moddable-OpenSource/moddable/commit/92ef810eae5d90373f3d38a499f232e2189c2aaf) now available.

stc1988, 2023-11-26:
> Your step 1 above doesn't say how to modify the manifest. I did this to eliminate the "memory full" error:

Sorry. Step1 is modification in [config](https://github.com/Moddable-OpenSource/moddable/blob/public/contributed/giphy/manifest.json#L2-L6) section.

I updated my branch to [e95cde0](https://github.com/Moddable-OpenSource/moddable/commit/e95cde090ea079e79779246eea8611b0f8628cd4) and successed to launch giphy app, but got an error message  at `downloadGIF` function.

```
/Users/satoshi/Projects/moddable/modules/files/flash/esp32/flash.c (53) # Break: Flash: can't find partition!
```

phoddie, 2023-11-26:
You didn't indicate the build target that you used. You indicate two possibilities above. I believe `esp32/m5stack` builds and launches. I'm guessing that you used `esp32/m5stack_core2`. I don't have either device with me, so I cannot test myself.

As a test, please try modifying `$MODDABLE/build/devices/esp32/targets/m5stack_core2/sdkconfig/partitions.csv` to include an "XS" partition as shown:

```
nvs,      data, nvs,     0x9000,  0x006000,
phy_init, data, phy,     0xf000,  0x001000,
factory,  app,  factory, 0x10000, 0x800000,
xs,       0x40, 1,       0x810000,0x400000,
```

Does that help?

stc1988, 2023-11-26:
I  modified `$MODDABLE/build/devices/esp32/targets/m5stack_core2/sdkconfig/partitions.csv` and got an another error.

```
/Users/satoshi/Projects/moddable/modules/files/flash/esp32/flash.c (127) # Break: map: map failed!
```

FYI, This app worked with `esp32/m5stack_core2`  two years ago.
https://x.com/stc1988/status/1401134611569078273?s=20