Moddable-OpenSource/moddable #1268: Pico display can't run examples with display code

taktran, 2023-12-13:
**Build environment:** macOS (Apple M1 Max)
**Moddable SDK version:** 4.3.4
**Target device:** `pico/pico_display`

**Description**

I'm trying to run the examples on the pico display, and the rgb led (using `pins/pwm` in `pins/tricolor-led` example) and the buttons (using `pins/digital` in `pins/button` example) work, but when I try any of the `commodetto` or `piu` examples, it builds, but doesn't show on the pico display. It does work on the `simulator/pico_display` though.

**Steps to Reproduce**

1. Go to one of the display examples eg, `cd $MODDABLE/examples/piu/balls` (same for `piu/images`, `piu/backlight`, `piu/list`, `piu/qrcode`, and `commodetto/text`)
1. Build and install the app using this build command: `mcconfig -d -m -p pico/pico_display`
    No error displays in the build
    It loads something on the pico:

    ```
    # Setting programming mode
    Waiting for /Volumes/RPI-RP2...
    # Programming: /Users/.../moddable/build/bin/pico/pico_display/debug/text/xs_pico.uf2 /Volumes/RPI-RP2
    Connect to xsbug @ 2e8a:000a.
    ```

**Expected behavior**

Display is shown on the pico display

**Images**

After building, nothing is shown on the screen:
[image]

The RGB LED does show some light (even though the example doesn't tell it to), but nothing on the display (not even a back light)

Works on the simulator

[image]

**Other information**

I've tested the pico display using the standard micropython examples and it works as expected, but I'm hoping I can use moddable instead since I'm much more familiar with JavaScript 🤞🏼 

Also, I couldn't get `arm-none-eabi-gcc` compiling on my M1 machine, so had to follow the following steps to install:

```
brew uninstall arm-none-eabi-gcc
brew autoremove
brew install --cask gcc-arm-embedded
```

From https://github.com/raspberrypi/pico-feedback/issues/355#issuecomment-1754808749. Since the RGB and buttons work, I'm assuming the build tools are ok.

mkellner, 2023-12-23:
I can reproduce the problem.

It is strange.

If I build for a pico/lilygo_t_display it runs correctly. That board has the same display.

The pico/pico_display_2 is a larger display, but also displays correctly.

I took the RPi Pico board off of the pico display and was able to program it and see that it connects to xsbug and draws regularly:

[image]

I then took the programmed board and reconnected it to the pico display and the screen animated as expected.

[image]

I was thereafter able to program other graphics apps to the pico display.

Can you please try to program the board when it is removed from the pico display and see if it connects to xsbug (like I did above)?

If it does connect, can you reassemble the pico display board and see if it displays correctly?

taktran, 2024-03-14:
I tried it without the pico display with the [commodetto/fireworks](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/commodetto/fireworks) example, and as suggested ran it without the pico display first, then put the pico display on. The onboard led, buttons and log inside xsbug seems to work but the screen still does not.

[image]

Any other ideas?