Moddable-OpenSource/moddable #1272: Resizing panels in xsbug doesn't work when run from WSL2 (Gtk resource error)

kallistisoft, 2023-12-15:
**Build environment:** Windows 10 Pro 22H2 (19045.3693) + WSL2 Ubuntu 20.04.6 LTS (5.15.133.1-microsoft-standard-WSL2)
**Moddable SDK version:** 4.2
**Target device:** N/A

**Description**
The panels in xsbug can't be resized due to Gtk being unable to find the cursor icons
```
Gdk-Message: 06:04:22.021: Unable to load sb_v_double_arrow from the cursor theme
Gdk-Message: 06:04:23.000: Unable to load sb_h_double_arrow from the cursor theme
```

**Steps to Reproduce**
1. Have a taste for absurdity :)
2. Install the moddable sdk tools in WSL2
3. Run xsbug and attempt to resize the panels

**Other information**
Other than the inability to resize the panels it seems to run just fine... obviously a low priority bug as it's an usual use case! 8P

phoddie, 2023-12-15:
If I understand correctly, this is an issue in WSL2 / Gtk, not xsbug. If there's a known work-around, we could try to apply that.

s1tnk, 2023-12-16:
Try this.
`sudo apt install adwaita-icon-theme-full`

s1tnk, 2023-12-16:
These issues are related.

https://github.com/microsoft/wslg/issues/757
https://github.com/microsoft/WSL/issues/7750#issuecomment-1657263369

kallistisoft, 2023-12-16:
> Try this. `sudo apt install adwaita-icon-theme-full`

Can confirm that installing the icon pack fixes the issue

kallistisoft, 2023-12-16:
I suggest adding adwaita-icon-theme-full to the list of perquisite packages in the Getting Started documentation for linux.

closed 2023-12-16 by kallistisoft

phoddie, 2023-12-17:
Seems like a reasonable suggestion. Is it enough to add `sudo apt install adwaita-icon-theme-full` (from your note above) to the instructions?

kallistisoft, 2023-12-17:
I would be inclined to add to the instructions for installing libgtk as they're related.

```
--- "Moddable SDK - Getting Started.md" 2023-10-26 22:01:19.617440147 -0500
+++ "Moddable SDK - Getting Started.md" 2023-12-17 07:57:41.629217399 -0600
@@ -319,10 +319,10 @@
    sudo apt-get install gcc git wget make libncurses-dev flex bison gperf
    ```
 
-2. Install the development version of the GTK+ 3 library:
+2. Install Gtk icons and development version of the GTK+ 3 library:
 
    ```text
-   sudo apt-get install libgtk-3-dev
+   sudo apt-get install libgtk-3-dev adwaita-icon-theme-full
    ```
 
 3. Create a `Projects` directory in your home directory at `~/Projects` for the Moddable SDK repository.
```

phoddie, 2023-12-18:
Thanks for the proposed text. @mkellner also did some work on updating the install instructions and that should be available soon.