Moddable-OpenSource/moddable #1275: No such file or directory  driver/dac_continuous.h:

NW-Lab, 2023-12-21:
Hello,

When building with the Windows Node-RED MCU Plugin, an error occurs during the build.
````
c:\pjt\moddable\modules\pins\i2s\audioout.c:75:26: fatal error: driver/dac_continuous.h: No such file or directory
    75 | #include "driver/dac_continuous.h"
       | ^~~~~~~~~~~~~~~~~~~~~~~~~
````
Platform is esp32/m5stack

Flow is Inject(Timestamp) and Debug Node.
[image]

Thanks,

mkellner, 2023-12-21:
Can you try to add `-I$(IDF_PATH)\components\driver\dac\include` to the tools/mcconfig/nmake.esp32.mk file?

```
diff --git a/tools/mcconfig/nmake.esp32.mk b/tools/mcconfig/nmake.esp32.mk
index 37c2dfc89..4b17fe352 100644
--- a/tools/mcconfig/nmake.esp32.mk
+++ b/tools/mcconfig/nmake.esp32.mk
@@ -193,6 +193,7 @@ INC_DIRS = \
        -I$(IDF_PATH)\components\bt\include\$(ESP32_BT_SUBCLASS)\include \
        -I$(IDF_PATH)\components\bt\host\bluedroid\api\include \
        -I$(IDF_PATH)\components\bt\host\bluedroid\api\include\api \
+    -I$(IDF_PATH)\components\driver\dac\include \
        -I$(IDF_PATH)\components\driver\gpio\include \
        -I$(IDF_PATH)\components\driver\gptimer\include \
        -I$(IDF_PATH)\components\driver\i2c\include \
```

NW-Lab, 2023-12-21:
mkellner san

Thank you for your prompt response.
The problem has been resolved.

Thank you.

NW-Lab, 2024-01-04:
mkellner san

hello
Not reflected in new releases.

thanks,

mkellner, 2024-01-04:
My apologies.

phoddie, 2024-01-10:
The commit is live now. Closing. If the problem is still present, please re-open with details. Thank you.

closed 2024-01-10 by phoddie