Moddable-OpenSource/moddable #1285: ReferenceError: module "url" not found

NW-Lab, 2024-01-04:
Hello

I get an error when I try to run the old sample. (But this is my first time doing this on Windows)

# Version
```
Welcome to Node-RED
===================

4 Jan 17:00:40 - [info] Node-RED version: v3.1.0
4 Jan 17:00:40 - [info] Node.js  version: v18.13.0
4 Jan 17:00:40 - [info] Windows_NT 10.0.19045 x64 LE
4 Jan 17:00:41 - [info] Loading palette nodes
4 Jan 17:00:43 - [info] Node-RED MCU Edition Runtime Version: #548dc37
4 Jan 17:00:43 - [info] Node-RED MCU Edition Plugin  Version: v1.4.1-beta.1
4 Jan 17:00:47 - [info] Moddable SDK Version: v4.3.8-0-gddde79a (x86)
4 Jan 17:00:48 - [info] *** node-red-mcu-plugin:
4 Jan 17:00:48 - [info] It looks as if a new platform option has been added.
4 Jan 17:00:48 - [info] Please raise an issue @ our GitHub repository, stating the following information:
4 Jan 17:00:48 - [info] > New platform: esp32/m5stamp_s3
4 Jan 17:00:48 - [info] *** node-red-mcu-plugin:
4 Jan 17:00:48 - [info] There seems to be an additional simulator option available.
4 Jan 17:00:48 - [info] Please raise an issue @ our GitHub repository, stating the following information:
4 Jan 17:00:48 - [info] > New simulator: sim/pico_ws_round
4 Jan 17:00:51 - [info] Dashboard version 3.4.0 started at /ui
4 Jan 17:00:52 - [warn] ------------------------------------------------------
4 Jan 17:00:52 - [warn] [@NW-Lab/node-red-mcu-m5/m5_button] 'mcu_digital_in' already registered by module @moddable-node-red/mcu
4 Jan 17:00:52 - [warn] ------------------------------------------------------
4 Jan 17:00:52 - [info] Settings file  : C:\Users\tauchi\.node-red\settings.js
4 Jan 17:00:52 - [info] Context store  : 'default' [module=memory]
4 Jan 17:00:52 - [info] User directory : \Users\tauchi\.node-red
4 Jan 17:00:52 - [warn] Projects disabled : editorTheme.projects.enabled=false
4 Jan 17:00:52 - [warn] Flows file name not set. Generating name using hostname.
4 Jan 17:00:52 - [info] Flows file     : \Users\tauchi\.node-red\flows_NotePC-SZ6.json
4 Jan 17:00:52 - [warn]
```

# Error
```
# xsc main.xsb
NMAKE : warning U4004: �^�[�Q�b�g 'c:\pjt\moddable\build\tmp\esp32\m5dial\debug\ikp87y60qoh\ftimage.h.xsi' �ɑ΂��ă��[�����������܂��B
# xsl modules
ReferenceError: module "url" not found
### ReferenceError: module "url" not found!
```

Thanks,

phoddie, 2024-01-04:
I'm not sure how to reproduce this issue. Would you share a simple flow that demonstrates the problem? Thank you.

phoddie, 2024-01-04:
Also, you encountered this error on Windows. Does it work on another platform (macOS or Linux) for you?

NW-Lab, 2024-01-05:
Thank you for the immediate reply.
The flow will be this one by @kitazaki .
```
[{"id":"5ed813308402a4b0","type":"tab","label":"フロー 1","disabled":false,"info":"","env":[],"_mcu":{"mcu":true}},{"id":"3c6c04d315fddf06","type":"ui_button","z":"5ed813308402a4b0","name":"A","group":"88ee1734c254e1cb","order":0,"width":0,"height":0,"passthru":false,"label":"A","tooltip":"","color":"","bgcolor":"","className":"","icon":"","payload":"A","payloadType":"str","topic":"topic","topicType":"msg","_mcu":{"mcu":true},"x":70,"y":40,"wires":[["def69a7b250c2736","172e2f277f989576"]]},{"id":"def69a7b250c2736","type":"ui_text","z":"5ed813308402a4b0","group":"88ee1734c254e1cb","order":1,"width":0,"height":0,"name":"","label":"Song","format":"{{msg.payload}}","layout":"row-spread","className":"","_mcu":{"mcu":true},"x":270,"y":80,"wires":[]},{"id":"2c72b81b70d481d9","type":"ui_button","z":"5ed813308402a4b0","name":"B","group":"88ee1734c254e1cb","order":0,"width":0,"height":0,"passthru":false,"label":"B","tooltip":"","color":"","bgcolor":"","className":"","icon":"","payload":"B","payloadType":"str","topic":"topic","topicType":"msg","_mcu":{"mcu":true},"x":70,"y":120,"wires":[["def69a7b250c2736","b1bc49e3731414ac"]]},{"id":"172e2f277f989576","type":"change","z":"5ed813308402a4b0","name":"","rules":[{"t":"set","p":"wave","pt":"msg","to":"http://host01.norahack.tk/bell.wav","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"_mcu":{"mcu":true},"x":300,"y":220,"wires":[["93893f1b5c7f304e"]]},{"id":"b1bc49e3731414ac","type":"change","z":"5ed813308402a4b0","name":"","rules":[{"t":"set","p":"wave","pt":"msg","to":"http://test.moddable.com/audio/ChristmasMusic/god_rest.wav","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"_mcu":{"mcu":true},"x":300,"y":280,"wires":[["93893f1b5c7f304e"]]},{"id":"93893f1b5c7f304e","type":"audioout","z":"5ed813308402a4b0","name":"","volume":"1.5","_mcu":{"mcu":true},"x":520,"y":220,"wires":[]},{"id":"88ee1734c254e1cb","type":"ui_group","name":"Happy X'mas!","tab":"5a15d716f5894281","order":1,"disp":true,"width":"6","collapse":false,"className":"","_mcu":{"mcu":false}},{"id":"5a15d716f5894281","type":"ui_tab","name":"home","icon":"dashboard","disabled":false,"hidden":false,"_mcu":{"mcu":false}}]
```

phoddie, 2024-01-05:
It looks like the "audioout" Node depends on URL. But. it doesn't include that module in its manifest. Could you try changing it? In [this file](https://github.com/phoddie/node-red-mcu/blob/548dc371181bf0ff7f8548ff3ceeaadec1478dfa/nodes/audioout/manifest.json#L2-L5), change the "includes" section to:

```json
	"include": [
		"$(MODDABLE)/examples/io/tcp/httpclient/manifest_httpclient.json",
		"$(MODDABLE)/modules/data/wavreader//manifest.json",
		"$(MODDABLE)/modules/data/url/manifest.json"
	],
```

NW-Lab, 2024-01-05:
I got an error on Mac as well.
```
Welcome to Node-RED
===================

5 Jan 09:44:46 - [info] Node-RED バージョン: v3.1.3
5 Jan 09:44:46 - [info] Node.js  バージョン: v18.17.1
5 Jan 09:44:46 - [info] Darwin 23.2.0 arm64 LE
5 Jan 09:44:46 - [info] パレットノードのロード
5 Jan 09:44:47 - [info] Node-RED MCU Edition Runtime Version: #548dc37
5 Jan 09:44:47 - [info] Node-RED MCU Edition Plugin  Version: v1.4.1-beta.1
5 Jan 09:44:47 - [info] Moddable SDK Version: v4.0.0-0-gc692d33
5 Jan 09:44:47 - [info] *** node-red-mcu-plugin:
5 Jan 09:44:47 - [info] It looks as if a new platform option has been added.
5 Jan 09:44:47 - [info] Please raise an issue @ our GitHub repository, stating the following information:
5 Jan 09:44:47 - [info] > New platform: esp32/s3_tft_feather
5 Jan 09:44:47 - [info] *** node-red-mcu-plugin:
5 Jan 09:44:47 - [info] It looks as if a platform option has been removed.
5 Jan 09:44:47 - [info] Please raise an issue @ our GitHub repository, stating the following information:
5 Jan 09:44:47 - [info] > Verify platform: esp32/esp32c6
5 Jan 09:44:47 - [info] > Verify platform: esp32/esp32c6_gc9a01
5 Jan 09:44:47 - [info] > Verify platform: esp32/esp32h2
5 Jan 09:44:47 - [info] > Verify platform: esp32/esp32h2_ili9341
5 Jan 09:44:47 - [info] > Verify platform: esp32/feather_s3_tft
5 Jan 09:44:47 - [info] > Verify platform: esp32/moddable_two_io_v5
5 Jan 09:44:47 - [info] > Verify platform: esp32/qtpyc3_ili9341
5 Jan 09:44:47 - [info] > Verify platform: esp32/qtpys2_ili9341
5 Jan 09:44:47 - [info] > Verify platform: esp32/qtpys3_cdc
5 Jan 09:44:47 - [info] > Verify platform: esp32/qtpys3_ili9341
5 Jan 09:44:47 - [info] > Verify platform: esp32/xiao_esp32c3_ili9341
5 Jan 09:44:47 - [info] > Verify platform: esp32/xiao_esp32s3_ili9341
5 Jan 09:44:47 - [info] > Verify platform: pico/ili9341_i2s
5 Jan 09:44:47 - [info] > Verify platform: pico/qtpy_ili9341
5 Jan 09:44:47 - [info] 設定ファイル: /Users/tauchi/.node-red/settings.js
5 Jan 09:44:47 - [info] コンテキストストア : 'default' [module=memory]
5 Jan 09:44:47 - [info] ユーザディレクトリ : /Users/tauchi/.node-red
5 Jan 09:44:47 - [warn] プロジェクトは無効化されています : editorTheme.projects.enabled=false
5 Jan 09:44:47 - [info] フローファイル     : /Users/tauchi/.node-red/flows.json
5 Jan 09:44:47 - [warn]
```
```
Starting build process...
Host system check: Darwin Kernel Version 23.2.0: Wed Nov 15 21:53:34 PST 2023; root:xnu-10002.61.3~2/RELEASE_ARM64_T8103
MCU Build system check: p1.4.1-beta.1 + #548dc37 @ m4.0.0-0-gc692d33
HOME directory check: /Users/tauchi
Creating build environment for platform esp32/m5stack_core2.
Working directory: /Users/tauchi/.node-red/mcu-plugin-cache/eabuujc5u5j
$IDF_PATH is defined: /Users/tauchi/esp32/esp-idf
$IDF_TOOLS_PATH identified: /Users/tauchi/.espressif
> cd /Users/tauchi/.node-red/mcu-plugin-cache/eabuujc5u5j
UPLOAD_PORT = undefined
IDF_TOOLS_EXPORT_CMD = undefined
IDF_PYTHON_ENV_PATH = undefined
OPENOCD_SCRIPTS = undefined
ESP_IDF_VERSION = undefined
Creating build script file...
> /bin/bash ./build.sh
>> source /Users/tauchi/esp32/esp-idf/export.sh
Detecting the Python interpreter
Checking "python" ...
/Users/tauchi/esp32/esp-idf/tools/detect_python.sh: line 16: python: command not found
Checking "python3" ...
Python 3.9.6
"python3" has been detected
Adding ESP-IDF tools to PATH...
Using Python interpreter in /Users/tauchi/.espressif/python_env/idf4.4_py3.9_env/bin/python
Checking if Python packages are up to date...
/Users/tauchi/esp32/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Python requirements from /Users/tauchi/esp32/esp-idf/requirements.txt are satisfied.
[... 29 lines cut ...]
/Users/tauchi/esp32/esp-idf/tools/check_python_dependencies.py:12: DeprecationWarning: pkg_resources is deprecated as an API. See https://setuptools.pypa.io/en/latest/pkg_resources.html
  import pkg_resources
Loading defaults file /Users/tauchi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/eabuujc5u5j/xsProj-esp32/sdkconfig.mc...
/tmp/confgen_tmply6ojdnd:62 line was updated to CONFIG_ESPTOOLPY_FLASHSIZE_4MB=n
CMake Deprecation Warning at /Users/tauchi/esp32/esp-idf/components/mbedtls/mbedtls/CMakeLists.txt:23 (cmake_minimum_required):
  Compatibility with CMake < 3.5 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done (2.1s)
-- Generating done (0.2s)
-- Build files have been written to: /Users/tauchi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/eabuujc5u5j/xsProj-esp32/build
Executing action: reconfigure
Running cmake in directory /Users/tauchi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/eabuujc5u5j/xsProj-esp32/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DSDKCONFIG_DEFAULTS=/Users/tauchi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/eabuujc5u5j/xsProj-esp32/sdkconfig.mc -DSDKCONFIG_HEADER=/Users/tauchi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/eabuujc5u5j/xsProj-esp32/build/config/sdkconfig.h -DCMAKE_MESSAGE_LOG_LEVEL=ERROR -DDEBUGGER_SPEED=1500000 -DIDF_TARGET=esp32 -DESP32_SUBCLASS=esp32 -DCCACHE_ENABLE=0 /Users/tauchi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/eabuujc5u5j/xsProj-esp32"...
# mcrez resources
# xsl modules
Total resource size: 115777 bytes
# cc mc.resources.c (slots in flash)
ReferenceError: module "url" not found
### ReferenceError: module "url" not found!
make: *** [/Users/tauchi/Projects/moddable/build/tmp/esp32/m5stack_core2/debug/eabuujc5u5j/mc.xs.c] Error 22
```

NW-Lab, 2024-01-05:
Adding it to manifest.json fixed it.
thank you.

phoddie, 2024-01-05:
Cool. Thanks for the quick confirmation. I've [committed](https://github.com/phoddie/node-red-mcu/commit/d445b50ba8251ed43d600c0843b712fc8528a5b2) the change.

FWIW – I think this got broken when the build was changed to only include the modules actually used by a project. This makes the binary size smaller and build faster. Another node must have included the "url" module , so the "audioout" node didn't have to include it.

NW-Lab, 2024-01-05:
sorry.
I have only checked Windows so far, but I thought that there was no problem because I solved it by adding it to the manifest.json of Device (m5stack_core2), but the problem did not occur even if I added it to the manifest.json of "nodes/audioout". It was not fixed.
why. I'm not sure, so I'll reopen.

NW-Lab, 2024-01-05:
Even on Mac, modifying manifest.json under "nodes/audioout" will not fix it.

thanks you

ralphwetzel, 2024-01-05:
> 5 Jan 09:44:47 - [info] Moddable SDK Version: v4.0.0-0-gc692d33

You're running - at least on the Mac - a SDK version that is quite _old_.
I've no evidence that this contributes to the problem - but I know that a lot has changed since then, which still might have an influence.

phoddie, 2024-01-05:
@ralphwetzel has a good point. I'm running with the latest Moddable SDK (v4.3.8). v4 is from August. (It doesn't look like the URL module changed since then, so maybe it is not part of the issue)

To try to isolate the problem, I made a trivial flow that just has the audioout node. The flow is not useful, but if the "url" module is missing, it will generate an exception during the preload phase of the build.

```json
[
    {
        "id": "989a916e7507f80b",
        "type": "tab",
        "label": "Flow 7",
        "disabled": false,
        "info": "",
        "env": [],
        "_mcu": {
            "mcu": false
        }
    },
    {
        "id": "f0966d7fa02f0f4f",
        "type": "audioout",
        "z": "989a916e7507f80b",
        "name": "",
        "volume": 1,
        "x": 420,
        "y": 220,
        "wires": []
    }
]
```

This builds for macOS and ESP32 without an error (with the updated manifest.json for the audioout node). What happens when you try this simple flow?

NW-Lab, 2024-01-05:
sorry. It was a misunderstanding.

It went well.

Since I was using a Plugin, I had to update the Plugin's Node-RED MCU.
I mistakenly thought that updating nodes in Node-RED was enough.
Manifest.json is included at build time, not at export time. got it.

I used xs-dev to update Moddable on my Mac to the latest version, but it didn't work. I don't know much about xs-dev, so I may be using it incorrectly.

thank you very much.

closed 2024-01-05 by NW-Lab

phoddie, 2024-01-05:
No problem. I hope it stays fixed. ;) If not, please reopen this issue.

> I don't know much about xs-dev, so I may be using it incorrectly.

xs-dev is designed to be very simple to use. If something isn't clear, please ask @HipsterBrown on the [xs-dev repository](https://github.com/HipsterBrown/xs-dev). Maybe there is something to be done improve it for everyone.