Moddable-OpenSource/moddable #1287: Build error in esp32/audio.c : "sample rate too low"

stc1988, 2024-01-06:
**Build environment:** macOS,
**Moddable SDK version:** 4.3.8(ESP-IDF v5.1.2  )
**Target device:** M5StackCore2, M5Stick-c

**Steps to Reproduce**
Build app `examples/pins/audioin/wavserver` using this build command: `mcconfig -d -m -p esp32/m5stick_c ssid="MY_SSID" password="MY_PASS"`

```
# cc digital.c.o
# cc audioin.c.o
# cc xs6wifi.c.o
/Users/satoshi/Projects/moddable/modules/pins/audioin/esp32/audioin.c:66:18: error: #error "sample rate too low"
   66 |                 #error "sample rate too low"
      |                  ^~~~~
make: *** [/Users/Satoshi/Projects/moddable/build/tmp/esp32/m5stick_c/debug/wavserver/audioin.c.o] Error 1
make: *** Waiting for unfinished jobs....
```

phoddie, 2024-01-07:
Under ESP-IDF 5.x, the minimum sample rate on the original ESP32 is now 20000.  Here's the relevant defines from `$IDF_PATH/components/soc/esp32/include/soc/soc_caps.h`:

```
#define SOC_ADC_SAMPLE_FREQ_THRES_HIGH          (2*1000*1000)
#define SOC_ADC_SAMPLE_FREQ_THRES_LOW           (20*1000)
```

This example uses 8000 in its manifest. This was fine with IDF v4. If you change the sample rate to 20000, does it work for you?

N.B. For other members of the ESP32 family, the minimum sample rate appears to be 611 so this is only an issue on the original ESP32.

phoddie, 2024-01-07:
Actually, the two devices you cite use PDM, not the ADC, for audio input. So, the check for the sample rate being in bounds for the ADC is unnecessary. Try replacing this....

https://github.com/Moddable-OpenSource/moddable/blob/ddde79a1d9757e803be744f841e83f387b0d8898/modules/pins/audioin/esp32/audioin.c#L65-L69

....with this...

```c
	#if MODDEF_AUDIOIN_I2S_ADC
		#if MODDEF_AUDIOIN_SAMPLERATE < SOC_ADC_SAMPLE_FREQ_THRES_LOW
			#warning "SOC reqires higher audioin sample rate. automatically increasing."
			#undef MODDEF_AUDIOIN_SAMPLERATE
			#define MODDEF_AUDIOIN_SAMPLERATE SOC_ADC_SAMPLE_FREQ_THRES_LOW
		#elif MODDEF_AUDIOIN_SAMPLERATE > SOC_ADC_SAMPLE_FREQ_THRES_HIGH
			#warning "SOC requires lower audioin sample rate. automatically decreasing."
			#undef MODDEF_AUDIOIN_SAMPLERATE
			#define MODDEF_AUDIOIN_SAMPLERATE SOC_ADC_SAMPLE_FREQ_THRES_HIGH
		#endif
	#endif
```

This does two things:

1. Doesn't check the ADC sample rates for non-ADC devices
2. If the requested sample rate is out of bounds for the target device, it is pinned to the closest available value

stc1988, 2024-01-07:
Thanks for confirming.
This replacing  make build succeed and `wavsever` app has no problem.

phoddie, 2024-01-07:
Cool. Thank you. I'll commit the change. It seems like an improvement.

phoddie, 2024-01-10:
Commit now available. Closing.

closed 2024-01-10 by phoddie