Moddable-OpenSource/moddable #1294: Unhandled exception error

louisvangeldrop, 2024-01-17:
**Build environment:** Windows
**Moddable SDK version:** 4.3.8-17-g595da4b7a
**Target device:** ESP32/m5atom_matrix

**Description**
I use a Try... Catch construction, but still get an XS abort: unhandled exception error

**Code**
```
import { wifi, deepSleep, mqttPublishAsync } from 'util'
import config from "mc/config";
import Timer from 'timer'

const ssid = config.ssid ?? "HaakakkerN"
const password = config.password 
const delay = config.delay ?? 5000  // 5 seconde
const mqttHost = config.host ?? "192.168.0.138"
const topic = config.topic ?? "moddable/mqtt/example"

const testAsync = async () => {
    trace('Rebooted.\n')
    try {
        let msg = await wifi({ ssid, password })
        trace(`Wi-Fi: ${msg}\n`);
        await mqttPublishAsync({ server: mqttHost, topic: `${topic}/boodschap`, message: msg })
        Timer.delay(1000)
    } catch (e) {
        trace(`${e}\n`)
    }
    finally {
        deepSleep(delay)
    }
}
testAsync()
```

**Error**

Wi-Fi: 192.168.0.130
Wi-Fi connected to "HaakakkerN"
IP address 192.168.0.130
Rebooted.
Wi-Fi: 192.168.0.130
Wi-Fi connected to "HaakakkerN"
IP address 192.168.0.130
Rebooted.
Wi-Fi: 192.168.0.130
Wi-Fi connected to "HaakakkerN"
IP address 192.168.0.130
Rebooted.
Wi-Fi: 192.168.0.130
Wi-Fi connected to "HaakakkerN"
IP address 192.168.0.130
Rebooted.
Wi-Fi: 192.168.0.130
C:\Users\louis\projecten\moddable\modules\network\wifi\esp32\xs6wifi.c (209) # Exception: connect: esp_wifi_connect failed!
XS abort: unhandled exception

phoddie, 2024-01-17:
I constructed a simple standalone test to try to reproduce what you show.. The test handled the exception as expected:

```js
async function wifi()
{
    throw new Error("boom");
}

const testAsync = async () => {
    trace('Rebooted.\n')
    try {
        await wifi()
    } catch (e) {
        trace(`exception: ${e}\n`)
    }
    finally {
        trace("goto sleep\n");
    }
}
testAsync()
```

Is it possible that you have `config.ssid` set in your build? If so, the Wi-Fi setup module will attempt to connect to Wi-Fi before launching your main.js. If there is an exception there, it would be unhandled. That might look like what you show. If not, then I think you'll need to provide a way to reproduce this for us to be able to investigate further. Thank you.

tve, 2024-01-17:
```
C:\Users\louis\projecten\moddable\modules\network\wifi\esp32\xs6wifi.c (209) # Exception: connect: esp_wifi_connect failed!
XS abort: unhandled exception
```
This does seem to indicate that the exception happens during the initial wifi connection attempt. Shouldn't the standard start-up code handle this more gracefully?

louisvangeldrop, 2024-01-17:
I have config.ssid set to "HaakakkerN". I will remove it from the manifest.json file.

phoddie, 2024-01-17:
@louisvangeldrop, you can remove the default network setup code from your project. That will avoid it triggering unexpectedly. That's what we do for commercial projects that want to manage Wi-Fi themselves. At the start of the `modules` section of your manifest.json add this. 
```
		"~": [
			"$(BUILD)/devices/esp32/setup/network"
		],
```

The `"~"` indicates that the modules listed in the array should be excluded from the build.

louisvangeldrop, 2024-01-17:
I have changed manifest.json according to your suggestion and it seems to work now. I will run the test for a couple of hours.

phoddie, 2024-01-17:
Excellent. Continued good luck.

phoddie, 2024-03-14:
This has been dormant for a while. Closing as it looks likely to have been resolved. If not, please re-open and provide additional information. Thank you.

closed 2024-03-14 by phoddie