Moddable-OpenSource/moddable #1298: Request body is undefined in listen.js

stc1988, 2024-01-27:
**Build environment:** macOS,
**Moddable SDK version:** 4.3.10-16-gb5dc9fe31
**Target device:** mac

**Steps to Reproduce**
1. Build and install the  [example/io/listener/listen](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/io/listener/listen) app using this build command: `mcconfig -d -m -p mac`
2. POST reqeust to listen server.

```
curl -X POST -H "Content-Type: application/json" -d "{"key":"value"}" localhost/json
```

[The log of body](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/io/listener/listen/main.js#L26) is undefined. 

```
POST
http://localhost/json
host: localhost
user-agent: curl/8.4.0
accept: */*
content-type: application/json
content-length: 11
undefined
```

**Other information**

I think options in Request is mismatch between [body](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/io/listener/listen/listen.js#L36) and [requestPromise](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/io/listener/listen/listen.js#L157
)

phoddie, 2024-01-28:
Thanks for the report. I'm able to reproduce the behavior you describe We'll investigate.

phoddie, 2024-01-28:
Your analysis was correct! Change line 157 to:

```js
const request = new Request(new URL(path, base), { method, path, headers, body:requestPromise });
```

We will include this in the next update. Thank you.

closed 2024-02-02 by stc1988