Moddable-OpenSource/moddable #1299: ESP32S3 can no longer be debugged on Windows.

NW-Lab, 2024-01-29:
hello

The program and xsbug will no longer be able to communicate on Windows.
It was probably used until around November.
The esp32s3 devices I have confirmed are esp32/m5atom_s3 and esp32/m5dial.
example/helloworld doesn't work.
The piu/balls screen is displayed, so the program seems to be running.
It is unclear whether it was when ESP-IDF was changed to v5.1.2 or when Moddable was updated.
It works fine with ESP32 (esp32/m5atom_matrix).

Console seems correct.
```
c:\pjt\moddable\examples\helloworld>mcconfig -d -m -p esp32/m5atom_s3
情報: 与えられたパターンのファイルが見つかりませんでした。
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\
Python 3.11.2
Using Git in C:\Espressif\tools\idf-git\2.39.2\cmd\
git version 2.39.2.windows.1
Checking Python compatibility
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v5.1.2

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.27.2.
Not using an unsupported version of tool ninja found in PATH: 1.11.0.
    C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin
    C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin
    C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin
    C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin
    C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
    C:\Espressif\tools\cmake\3.24.0\bin
    C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin
    C:\Espressif\tools\ninja\1.10.2\
    C:\Espressif\tools\idf-exe\1.0.3\
    C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64
    C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64
    C:\Espressif\frameworks\esp-idf-v5.1.2\tools

Checking if Python packages are up to date...
Constraint file: C:\Espressif\espidf.constraints.v5.1.txt
Requirement files:
 - C:\Espressif\frameworks\esp-idf-v5.1.2\tools\requirements\requirements.core.txt
Python being checked: C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Python requirements are satisfied.

Detected installed tools that are not currently used by active ESP-IDF version.
For removing old versions of idf-driver, idf-python-wheels use command 'python.exe C:\Espressif\frameworks\esp-idf-v5.1.2\tools\idf_tools.py uninstall'
For free up even more space, remove installation packages of those tools. Use option 'python.exe C:\Espressif\frameworks\esp-idf-v5.1.2\tools\idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Using recommended ESP-IDF v5.1.2
        1 個のファイルをコピーしました。
        1 個のファイルをコピーしました。
        1 個のファイルをコピーしました。
Reconfiguring ESP-IDF...
Executing action: reconfigure
Running cmake in directory C:\pjt\moddable\build\tmp\esp32\m5atom_s3\debug\helloworld\xsProj-esp32s3\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DSDKCONFIG_DEFAULTS=c:/pjt/moddable/build/tmp/esp32/m5atom_s3/debug/helloworld/xsProj-esp32s3/sdkconfig.mc -DSDKCONFIG_HEADER=c:\pjt\moddable\build\tmp\esp32\m5atom_s3\debug\helloworld\xsProj-esp32s3\build\config\sdkconfig.h -DCMAKE_MESSAGE_LOG_LEVEL=ERROR -DDEBUGGER_SPEED=115200 -DIDF_TARGET=esp32s3 -DESP32_SUBCLASS=esp32s3 -DSDKCONFIG_DEFAULTS=c:\pjt\moddable\build\tmp\esp32\m5atom_s3\debug\helloworld\xsProj-esp32s3\sdkconfig.mc -DCCACHE_ENABLE=1 C:\pjt\moddable\build\tmp\esp32\m5atom_s3\debug\helloworld\xsProj-esp32s3"...
Loading defaults file C:/pjt/moddable/build/tmp/esp32/m5atom_s3/debug/helloworld/xsProj-esp32s3/sdkconfig.mc...
C:\Users\tauchi\AppData\Local\Temp\kconfgen_tmpvpfjb7w0:504 line was updated to CONFIG_ESP_SYSTEM_PANIC_GDBSTUB=n
C:\Users\tauchi\AppData\Local\Temp\kconfgen_tmpvpfjb7w0:585 line was updated to CONFIG_ESP_COREDUMP_ENABLE_TO_UART=n
C:\Users\tauchi\AppData\Local\Temp\kconfgen_tmpvpfjb7w0:590 line was updated to CONFIG_ESP_COREDUMP_ENABLE=n
-- Configuring done
-- Generating done
[... 87 lines cut ...]
Wrote 3072 bytes (83 compressed) at 0x00008000 in 0.1 seconds (effective 226.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
Done
Launching app...
Type Ctrl-C twice after debugging app.
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x28 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40379106
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3818,len:0x88
load:0x403c9700,len:0x4
load:0x403c9704,len:0x988
load:0x403cc700,len:0x2560
entry 0x403c9858
START USB CONSOLE!!!
```
It is currently unknown whether the serial communication string for debugging is not output or there is a communication problem.
I haven't tried it yet on a non-Windows device.

thanks,

phoddie, 2024-01-30:
Thanks for the report. We'll try to reproduce this.

NW-Lab, 2024-01-30:
It may be unrelated (another ISSUE?), but maybe because Tinyusb is no longer installed by default with ESP-IDF v5, I can't build USE_USB=1 (for example, esp32/m5stamp_s3) properly.
It may be because you are using Windows and have downloaded and installed the ESP-IDF installer.

The message will be displayed as shown below.
```C:\pjt\moddable\examples\helloworld>mcconfig -d -m -p esp32/m5stamp_s3
情報: 与えられたパターンのファイルが見つかりませんでした。
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\
Python 3.11.2
Using Git in C:\Espressif\tools\idf-git\2.39.2\cmd\
git version 2.39.2.windows.1
Checking Python compatibility
Setting IDF_PATH: C:\Espressif\frameworks\esp-idf-v5.1.2

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.27.2.
Not using an unsupported version of tool ninja found in PATH: 1.11.0.
    C:\Espressif\tools\xtensa-esp-elf-gdb\12.1_20221002\xtensa-esp-elf-gdb\bin
    C:\Espressif\tools\riscv32-esp-elf-gdb\12.1_20221002\riscv32-esp-elf-gdb\bin
    C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin
    C:\Espressif\tools\xtensa-esp32s2-elf\esp-12.2.0_20230208\xtensa-esp32s2-elf\bin
    C:\Espressif\tools\xtensa-esp32s3-elf\esp-12.2.0_20230208\xtensa-esp32s3-elf\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-12.2.0_20230208\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
    C:\Espressif\tools\cmake\3.24.0\bin
    C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20230921\openocd-esp32\bin
    C:\Espressif\tools\ninja\1.10.2\
    C:\Espressif\tools\idf-exe\1.0.3\
    C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64
    C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64
    C:\Espressif\frameworks\esp-idf-v5.1.2\tools

Checking if Python packages are up to date...
Constraint file: C:\Espressif\espidf.constraints.v5.1.txt
Requirement files:
 - C:\Espressif\frameworks\esp-idf-v5.1.2\tools\requirements\requirements.core.txt
Python being checked: C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe
Python requirements are satisfied.

Detected installed tools that are not currently used by active ESP-IDF version.
For removing old versions of idf-driver, idf-python-wheels, openocd-esp32 use command 'python.exe C:\Espressif\frameworks\esp-idf-v5.1.2\tools\idf_tools.py uninstall'
For free up even more space, remove installation packages of those tools. Use option 'python.exe C:\Espressif\frameworks\esp-idf-v5.1.2\tools\idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Using recommended ESP-IDF v5.1.2
        1 個のファイルをコピーしました。
        1 個のファイルをコピーしました。
        1 個のファイルをコピーしました。
"# Configure tinyusb..."; cd c:\pjt\moddable\build\tmp\esp32\m5stamp_s3\debug\helloworld\xsProj-esp32s3 ; idf.py add-dependency "espressif/esp_tinyusb"
Reconfiguring ESP-IDF...
Executing action: reconfigure
Running cmake in directory C:\pjt\moddable\build\tmp\esp32\m5stamp_s3\debug\helloworld\xsProj-esp32s3\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.1_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DSDKCONFIG_DEFAULTS=c:/pjt/moddable/build/tmp/esp32/m5stamp_s3/debug/helloworld/xsProj-esp32s3/sdkconfig.mc -DSDKCONFIG_HEADER=c:\pjt\moddable\build\tmp\esp32\m5stamp_s3\debug\helloworld\xsProj-esp32s3\build\config\sdkconfig.h -DCMAKE_MESSAGE_LOG_LEVEL=ERROR -DDEBUGGER_SPEED=460800 -DIDF_TARGET=esp32s3 -DESP32_SUBCLASS=esp32s3 -DSDKCONFIG_DEFAULTS=c:\pjt\moddable\build\tmp\esp32\m5stamp_s3\debug\helloworld\xsProj-esp32s3\sdkconfig.mc -DCCACHE_ENABLE=1 C:\pjt\moddable\build\tmp\esp32\m5stamp_s3\debug\helloworld\xsProj-esp32s3"...
Loading defaults file C:/pjt/moddable/build/tmp/esp32/m5stamp_s3/debug/helloworld/xsProj-esp32s3/sdkconfig.mc...
C:/pjt/moddable/build/tmp/esp32/m5stamp_s3/debug/helloworld/xsProj-esp32s3/sdkconfig.mc:1433 CONFIG_CONSOLE_UART_DEFAULT was replaced with CONFIG_ESP_CONSOLE_UART_DEFAULT
C:/pjt/moddable/build/tmp/esp32/m5stamp_s3/debug/helloworld/xsProj-esp32s3/sdkconfig.mc:1434 CONFIG_CONSOLE_UART was replaced with CONFIG_ESP_CONSOLE_UART
C:/pjt/moddable/build/tmp/esp32/m5stamp_s3/debug/helloworld/xsProj-esp32s3/sdkconfig.mc:1435 CONFIG_CONSOLE_UART_BAUDRATE was replaced with CONFIG_ESP_CONSOLE_UART_BAUDRATE
C:\Users\tauchi\AppData\Local\Temp\kconfgen_tmp7fn575qw:504 line was updated to CONFIG_ESP_SYSTEM_PANIC_GDBSTUB=n
C:\Users\tauchi\AppData\Local\Temp\kconfgen_tmp7fn575qw:585 line was updated to CONFIG_ESP_COREDUMP_ENABLE_TO_UART=n
C:\Users\tauchi\AppData\Local\Temp\kconfgen_tmp7fn575qw:590 line was updated to CONFIG_ESP_COREDUMP_ENABLE=n
-- Configuring done
-- Generating done
-- Build files have been written to: C:/pjt/moddable/build/tmp/esp32/m5stamp_s3/debug/helloworld/xsProj-esp32s3/build
        1 個のファイルをコピーしました。
# ld xs_esp32.bin
        1 個のファイルをコピーしました。
[... 16 lines cut ...]
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
-- Adding linker script C:/Espressif/frameworks/esp-idf-v5.1.2/components/esp_rom/esp32s3/ld/esp32s3.rom.newlib.ld
[... cut ...]
ninja failed with exit code 1, output of the command is in the C:\pjt\moddable\build\tmp\esp32\m5stamp_s3\debug\helloworld\xsProj-esp32s3\build\log\idf_py_stderr_output_7604 and C:\pjt\moddable\build\tmp\esp32\m5stamp_s3\debug\helloworld\xsProj-esp32s3\build\log\idf_py_stdout_output_7604
Launching app...
Type Ctrl-C twice after debugging app.
Could not find USB COM Port with VID 0xBEEF and ID 0x1CEE
NMAKE : fatal error U1077: 'echo Launching app... & echo Type Ctrl-C twice after debugging app. && set "XSBUG_PORT=5002" && set "XSBUG_HOST=localhost" && c:\pjt\moddable\build\bin\win\release\serial2xsbug beef:1cee 460800 8N1 -dtr' : リターン コード '0x1'
Stop.
```

thanks,

mkellner, 2024-02-05:
Hello @NW-Lab, can you please try this patch?

https://gist.github.com/mkellner/993ab3febf4f7dd951e3ba9473d88f76

NW-Lab, 2024-02-05:
mkellner san

Thank you for your response.
I checked with example helloworld.
I have confirmed the fix for the original error.(esp32/m5dial:USE_USE=2)
Thank you.

The tinyusb.h not found error is still there.(esp32/m5stamp_s3:USE_USB=1)
There is no tiny_usb in %IDF_PATH%\components, so it's probably an installation problem.

Thank you.

mkellner, 2024-02-05:
Thank you for your quick response.

Please open a new issue for USE_USB=1 as it is indeed a different issue.

NW-Lab, 2024-02-05:
I understand.
thank you.