Moddable-OpenSource/moddable #1302: Virus detected during build of SDK

ethos-jgreenwood, 2024-01-31:
While following the Moddable SDK Getting Started page, on step 6. I had an error that tools.exe was not found. I tried again and window s defender alerted to the regenerated file being a trojan virus. I uploaded to virustotal (see link below). Enough virus scan services are flagging this file as a virus that I don't feel comfortable proceeding.

https://www.virustotal.com/gui/file/5b26746cccc2a8ea1a9fc3c3518261c034d0e3fb42141e40b88847fc53a9dad8/detection

phoddie, 2024-01-31:
Thanks for giving it a try.

Windows Defender sometimes flags the tool as a virus. It is not. @andycarle, do you have any advice here?

andycarle, 2024-01-31:
Hi @ethos-jgreenwood,

Indeed, we have had a recurring issue with Windows Defender and other anti-virus software occasionally incorrectly flagging `tools.exe` as a virus. You can see some history of this in our GitHub issues dating back to [October of 2022](https://github.com/Moddable-OpenSource/moddable/issues/940#issuecomment-1277425854).

In the past, we have submitted the executable to the Windows Defender team for analysis, which has resulted in it being whitelisted for a time. But as the Moddable SDK changes, the build output changes enough to lose that whitelisting.

For users who trust the Moddable SDK build and codebase, the fix is pretty simple:

1) Delete `%MODDABLE%\build\bin` and `%MODDABLE\build\tmp`
2) Exclude the `%MODDABLE%` directory from Windows Security using [these instructions](https://support.microsoft.com/en-us/windows/add-an-exclusion-to-windows-security-811816c0-4dfd-af4a-47e4-c301afe13b26)
3) Rebuild the Moddable SDK and verify that `tools.exe` has been successfully created and is not quarantined

But, of course, you will have to decide for yourself whether or not you are comfortable proceeding. If it provides any peace of mind, `tools.exe` is [built by the Moddable SDK here](https://github.com/Moddable-OpenSource/moddable/blob/b5dc9fe31c5b74b23503a8760b0151235b5ee3ca/build/makefiles/win/tools.mak#L304) entirely from the open source code that is available for perusal in this very GitHub repository.

ethos-jgreenwood, 2024-01-31:
Thank you both @andycarle and @phoddie for the responses.

My main concern was that the executable was being pulled from an external source that had been compromised. Thank you for giving me the information to show that this is an incorrect flagging. I'll dig into the solution provided!

closed 2024-01-31 by ethos-jgreenwood