Moddable-OpenSource/moddable #1314: make: *** [makefile:43: debug] Error 127

tejas2806, 2024-02-23:
> Note: This template is intended to help organize your report. Feel free to modify it as you need. The most important thing is to file a clear, detailed report.

> You may use this template to file a report of a security vulnerability. If you prefer to report the issue privately, please send the report by email to info@moddable.com.

Build environment :- WSL Ubuntu 20.04

**Description**
unable to build moddable tools. something is going wrong in the moddable -> build -> makefiles ->lin -> makefile.

**Steps to Reproduce**
1. cd MODDABLE/build/makefiles/lin
2. make

**Expected behavior**
should build all the moddable tools without throwing errors

**Images**
[image]

**Other information**
Any other details you believe may be helpful in understanding the issue, reproducing the unexpected behavior, or fixing the software

mkellner, 2024-02-23:
Hello @tejas2806, 

It looks like you may have set up your environment some time ago. We changed the tools-build to only build release versions by default.

Do you have `$MODDABLE/build/bin/lin/debug` in your path? That should now be `$MODDABLE/build/bin/lin/release`.

Please give that a try and let us know how it goes.

tejas2806, 2024-02-24:
Hi [mkellner](https://github.com/mkellner)!
I tried out your suggested changes.
[image]

but , i am still getting same error.

mkellner, 2024-02-25:
It is strange that you are getting the same error if you changed the path. The original error was using a path to the debug version of the application.

Please let us know what version of the SDK you are using - that is, type this on the command line and send back the first line: `git log`.
The output from my tree is: `commit 17e61bc92356d0c77db50784566edf67045de381 (HEAD -> public, origin/public, origin/HEAD)`

If you want to use the debug version of the tools, you can build them:
```
cd $MODDABLE/build/makefiles/lin
make debug
```

phoddie, 2024-04-05:
Closing because this has gone quiet. If it is still an issue, please reopen and include additional details. Thank you!