Moddable-OpenSource/moddable #1322: ESP32 io/analog module: adc_oneshot_unit_init_cfg_t structure not initialized

linfan68, 2024-03-10:
This code does not properly init the structure (to all zero), leads to random initialization failure:

https://github.com/Moddable-OpenSource/moddable/blob/dd559841d0fc3ad1e834575fa6c736ed5ed023bd/modules/io/analog/esp32/_analog.c#L256

Error message:
```
E (3505) esp_clk_tree: esp_clk_tree_src_get_freq_hz(21): unknown clk src
E (3509) adc_oneshot: adc_oneshot_new_unit(106): clock source not supported
```

needs to be: 

```
adc_oneshot_unit_init_cfg_t unit_cfg = {};
```

phoddie, 2024-03-11:
Good catch. Thank you. I've applied your fix internally and it will propagate to this public repo in the near future.

phoddie, 2024-03-14:
This [fix is now live](https://github.com/Moddable-OpenSource/moddable/commit/c61adfb7904d04e6c067ca62b9e4996bcd81f807). Thanks again.

closed 2024-03-14 by phoddie