Moddable-OpenSource/moddable #1325: Communication error with xs-bug?

NW-Lab, 2024-03-21:
Hello

When using Node-RED MCU Plugin, I can no longer communicate with Node-RED Editor (PC).
Even if I press his button on the Injct node, it doesn't seem to be transmitted to the MCU.

Environment:
PC:Windows
MCU esp32/m5stack

Environment where connection is not possible
```
21 Mar 09:56:51 - [info] Node-RED version: v3.1.7
21 Mar 09:56:51 - [info] Node.js  version: v20.11.1
21 Mar 09:56:51 - [info] Windows_NT 10.0.19045 x64 LE
21 Mar 09:56:52 - [info] Loading palette nodes
21 Mar 09:56:53 - [info] Node-RED MCU Edition Runtime Version: #6855b56
21 Mar 09:56:53 - [info] Node-RED MCU Edition Plugin  Version: v1.4.1-beta.1
21 Mar 09:56:56 - [info] Moddable SDK Version: v4.5.0-13-ga6a71b2 (x86)
```

The slightly older version of Moddable below works fine.
```
21 Mar 09:54:52 - [info] Node-RED version: v3.1.7
21 Mar 09:54:52 - [info] Node.js  version: v20.11.1
21 Mar 09:54:52 - [info] Windows_NT 10.0.19045 x64 LE
21 Mar 09:54:53 - [info] Loading palette nodes
21 Mar 09:54:54 - [info] Node-RED MCU Edition Runtime Version: #6855b56
21 Mar 09:54:54 - [info] Node-RED MCU Edition Plugin  Version: v1.4.1-beta.1
21 Mar 09:54:57 - [info] Moddable SDK Version: v17e61bc (x86)
```

flow is a simple Inject node and Debug node.
[image]

Thank you.

phoddie, 2024-03-21:
The commit hashes do not exactly match what GitHub shows. I *think* that you mean these two commits: [a6a71b2](https://github.com/Moddable-OpenSource/moddable/commit/a6a71b225bd35201ec76b2c354b7595a1be2ed12) and
[17e61bc](https://github.com/Moddable-OpenSource/moddable/commit/17e61bc92356d0c77db50784566edf67045de381). Would you please confirm? Thank you.

NW-Lab, 2024-03-22:
thank you for your reply.

We confirmed that restoring https://github.com/Moddable-OpenSource/moddable/commit/17e61bc92356d0c77db50784566edf67045de381 (xsString.c) fixed the issue.
I don't know where the problem is.
Thank you.

phoddie, 2024-03-22:
Thank you. It is difficult to see what might have caused a problem here. 

Maybe a change to the ESP32 make file? For example "[don't suppress deploy error](https://github.com/Moddable-OpenSource/moddable/commit/41a2d930c5bc0d44d5c44c53232aac3bfc38719d) "? 

If possible, could you test with some other commits to see if you can narrow down where the problem began?

NW-Lab, 2024-03-23:
@phoddie san

sorry.
For some reason, it now works with the current repository.
I may have misunderstood somewhere.
CLOSE.

closed 2024-03-23 by NW-Lab

phoddie, 2024-03-23:
No problem. If the problem comes back, please reopen this issue with any new information.