Moddable-OpenSource/moddable #1328: Link errors when building with "-p lin" (audioout and servo)

ronron-gh, 2024-04-01:
**Build environment:** Linux
**Moddable SDK version:** 4.5.0-23-gfc5c2b41c
**Target device:** desktop simulator

The following two modules caused link errors when built for Linux desktop simulator.
I have found out the causes of each problem as below, so it would be helpful if you could take countermeasures.

・pins/i2s/audioout
When building with "-p lin", queueCallback() is undefined in audioout.c, resulting in a link error.
When I defined an empty queueCallback(), the link was successful and I was able to move on to execution with xsbug.
（I would be happy if in the future it would be possible to play sound on Linux as well.）

・pins/servo
A link error will occur because there is no description for Linux in _pins/servo/manifest.json_.
The issue was resolved by adding the following description:

```
"platforms":{
    "lin": {
        "modules": {
            "pins/servo": "$(MODULES)/pins/servo/servo",
            "*": "$(MODULES)/pins/servo/sim/*"
        }
    }
}
```

mkellner, 2024-04-02:
Thank you. I've committed a fix for these. They will be in the next release.

ronron-gh, 2024-04-04:
Thank you for your commit. I've confirmed that it works, so I'll close it.

closed 2024-04-04 by ronron-gh