Moddable-OpenSource/moddable #1334: mcconfig: Error opening network: -1 with esp32s3_usb or cdc

sanielfishawy, 2024-04-11:
**Build environment:** macos Ventura 13.1
**Moddable SDK version:** 4.6.0
**Chip.** esp32-s3-wroom1-1
**Board.** ESP32-S3-DevKitC-1-N8R2
**Target device:** esp32s3_usb

**Description**
`UPLOAD_PORT=/dev/cu.usbmodem14301 mcconfig -d -m -p esp32/esp32s3_usb` 

Produces compilation warnings. 

Pressing reset button at the final step produces the following error: 

```
Error opening network: -1.
make: *** [all] Error 1
```

After the error a port check on the os shows the device connected at the strange port of: `/dev/cu.usbmodem1234561`. 

Setting the device to boot loader mode causes the macos to again show the device at `/dev/cu.usbmodem14301`.

**Expected behavior**
xsbug should open. The console in xsbug should show something like:
`Device flashed. Ready to install apps.`

**Steps to Reproduce**
Hold boot and power cycle to put the chip in boot-loader mode. 

Run the following commands: 

```
esptool.py --port /dev/cu.usbmodem14301 erase_flash  //success
mcconfig -d -m -t clean -p esp32/esp32s3_usb  //#clean project
UPLOAD_PORT=/dev/cu.usbmodem14301 mcconfig -d -m -p esp32/esp32s3_usb
```

Notice:

```
      1) Several warnings:

/Users/sanielfishawy/dev/Moddable/moddable/xs/sources/xsBigInt.c: In function 'fxBigInt_uadd_prim':
/Users/sanielfishawy/dev/Moddable/moddable/xs/sources/xsBigInt.c:1369:51: warning: passing argument 3 of '__builtin_uadd_overflow' from incompatible pointer type [-Wincompatible-pointer-types]
 1369 |         if (__builtin_uadd_overflow(ap[i], bp[i], &r)) {
      |                                                   ^~
      |                                                   |
      |                                                   txU4 * {aka long unsigned int *}
/Users/sanielfishawy/dev/Moddable/moddable/xs/sources/xsBigInt.c:1369:51: note: expected 'unsigned int *' but argument is of type 'txU4 *' {aka 'long unsigned int *'}
/Users/sanielfishawy/dev/Moddable/moddable/xs/sources/xsBigInt.c:1374:51: warning: passing argument 3 of '__builtin_uadd_overflow' from incompatible pointer type [-Wincompatible-pointer-types]
 1374 |                 c = __builtin_uadd_overflow(r, c, &rp[i]);
      |                                                   ^~~~~~
      |                                                   |
      |                                                   txU4 * {aka long unsigned int *}
/Users/sanielfishawy/dev/Moddable/moddable/xs/sources/xsBigInt.c:1374:51: note: expected 'unsigned int *' but argument is of type 'txU4 *' {aka 'long unsigned int *'}
/Users/sanielfishawy/dev/Moddable/moddable/xs/sources/xsBigInt.c:1380:55: warning: passing argument 3 of '__builtin_uadd_overflow' from incompatible pointer type [-Wincompatible-pointer-types]
 1380 |                 c = __builtin_uadd_overflow(1, bp[i], &rp[i]);
      |                                                       ^~~~~~
      |                                                       |
      |                                                       txU4 * {aka long unsigned int *}
/Users/sanielfishawy/dev/Moddable/moddable/xs/sources/xsBigInt.c:1380:55: note: expected 'unsigned int *' but argument is of type 'txU4 *' {aka 'long unsigned int *'}

[800/971] Building C object esp-idf/console/CMakeFiles/__idf_console.dir/esp_console_repl.c.obj/Users/sanielfishawy/dev/esp32/esp-idf-v5.1.2/components/console/esp_console_repl.c:479:13: warning: 'esp_console_repl_task' defined but not used [-Wunused-function]
  479 | static void esp_console_repl_task(void *args)
      |             ^~~~~~~~~~~~~~~~~~~~~
/Users/sanielfishawy/dev/esp32/esp-idf-v5.1.2/components/console/esp_console_repl.c:375:18: warning: 'esp_console_common_init' defined but not used [-Wunused-function]
  375 | static esp_err_t esp_console_common_init(size_t max_cmdline_length, esp_console_repl_com_t *repl_com)
      |                  ^~~~~~~~~~~~~~~~~~~~~~~
/Users/sanielfishawy/dev/esp32/esp-idf-v5.1.2/components/console/esp_console_repl.c:354:18: warning: 'esp_console_setup_history' defined but not used [-Wunused-function]
  354 | static esp_err_t esp_console_setup_history(const char *history_path, uint32_t max_history_len, esp_console_repl_com_t *repl_com)
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/sanielfishawy/dev/esp32/esp-idf-v5.1.2/components/console/esp_console_repl.c:329:18: warning: 'esp_console_setup_prompt' defined but not used [-Wunused-function]
  329 | static esp_err_t esp_console_setup_prompt(const char *prompt, esp_console_repl_com_t *repl_com)
      |                  ^~~~~~~~~~~~~~~~~~~~~~~~
[970/971] Generating binary image from built executableesptool.py v4.7.0
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.

    2) Arrives here: 

      Leaving...
      Hard resetting via RTS pin...
      Done

      ### Press and release the RESET button.
      Waiting for serial beef:1cee...

    3) Before pressing the reset button:
        `ls /dev/cu*  # reports: /dev/cu.usbmodem14301`

    4) I press the reset button:

     ### Press Ctrl-C to quit.
     Error opening network: -1.
     make: *** [all] Error 1

    5) After pressing the reset button. 
        `ls /dev/cu*  # reports: /dev/cu.usbmodem1234561`

    6) In this state with strange usb port on the mac.
        `esptool.py --port /dev/cu.usbmodem1234561 erase_flash`

        Gives the following error:
             "I", fcntl.ioctl(self.port.fileno(), TIOCMGET, struct.pack("I", 0))
            OSError: [Errno 6] Device not configured
    
    7) Setting the device to boot loader mode causes the macos to set the port for the device back to /dev/cu.usbmodem14301
```

**Other information**

- Device target esp32s3_cdc produces the same error.
- `UPLOAD_PORT=/dev/cu.usbmodem14301 mcconfig -dl -m -t esp32/esp32s3_cdc` produces the same error
- `mcconfig -dl -m -t esp32/esp32s3_cdc` produces the same error
-  `UPLOAD_PORT=/dev/cu.usbmodem14301 mcrun -d -m -p esp32/esp32s3_usb` on the helloworld project opens xsbug as expected on the mod.
- `UPLOAD_PORT=/dev/cu.usbmodem14301 mcrun -dl -m -p esp32/esp32s3_usb` on the helloworld project unexpectedly opens xsbug on the mod rather than xsbug-log. 
- `UPLOAD_PORT=/dev/cu.usbmodem14301 mcconfig -m -p esp32/esp32s3_usb` gives the error below with a similar change in port along the way:

```
Leaving...
Hard resetting via RTS pin...
Done
Executing action: monitor
Running idf_monitor in directory /Users/sanielfishawy/dev/Moddable/moddable/build/tmp/esp32/esp32s3_usb/release/ch1host/xsProj-esp32s3
Executing "/Users/sanielfishawy/.espressif/python_env/idf5.1_py3.9_env/bin/python /Users/sanielfishawy/dev/esp32/esp-idf-v5.1.2/tools/idf_monitor.py -p /dev/cu.usbmodem14301 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --decode-coredumps info --target esp32s3 --revision 0 /Users/sanielfishawy/dev/Moddable/moddable/build/tmp/esp32/esp32s3_usb/release/ch1host/xsProj-esp32s3/build/xs_esp32.elf -m '/Users/sanielfishawy/.espressif/python_env/idf5.1_py3.9_env/bin/python' '/Users/sanielfishawy/dev/esp32/esp-idf-v5.1.2/tools/idf.py' '-p' '/dev/cu.usbmodem14301' '-n'"...
--- esp-idf-monitor 1.4.0 on /dev/cu.usbmodem14301 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Connection to /dev/cu.usbmodem14301 failed. Available ports:
/dev/cu.BLTH
/dev/cu.Bluetooth-Incoming-Port
/dev/cu.usbmodem1234561

[Errno 2] could not open port /dev/cu.usbmodem14301: [Errno 2] No such file or directory: '/dev/cu.usbmodem14301'
```

- `UPLOAD_PORT=/dev/cu.usbmodem14301 mcconfig -m -p esp32/esp32s3_cdc` gives the error below. Pressing Ctrl+T for example does nothing.

```
Leaving...
Hard resetting via RTS pin...
Done
Executing action: monitor
Running idf_monitor in directory /Users/sanielfishawy/dev/Moddable/moddable/build/tmp/esp32/esp32s3_cdc/release/ch1host/xsProj-esp32s3
Executing "/Users/sanielfishawy/.espressif/python_env/idf5.1_py3.9_env/bin/python /Users/sanielfishawy/dev/esp32/esp-idf-v5.1.2/tools/idf_monitor.py -p /dev/cu.usbmodem14301 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 --revision 0 /Users/sanielfishawy/dev/Moddable/moddable/build/tmp/esp32/esp32s3_cdc/release/ch1host/xsProj-esp32s3/build/xs_esp32.elf -m '/Users/sanielfishawy/.espressif/python_env/idf5.1_py3.9_env/bin/python' '/Users/sanielfishawy/dev/esp32/esp-idf-v5.1.2/tools/idf.py' '-p' '/dev/cu.usbmodem14301' '-n'"...
--- esp-idf-monitor 1.4.0 on /dev/cu.usbmodem14301 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x0 (DOWNLOAD(USB/UART0))
Saved PC:0x40041a79
0x40041a79: ets_delay_us in ROM

waiting for download
None
Waiting for the device to reconnect.
```

sanielfishawy, 2024-04-12:
This issue was resolved as follows:

Make sure xsbug is running before running mcconfig.

The change in port to cu.usbmodem1234561  persists but it does not seem to cause any issues.

phoddie, 2024-04-12:
FWIW – we're updating the error message, `Error opening network: -1` to include a hint that the debugger isn't running. This may help other developers.

closed 2024-04-12 by phoddie

mkellner, 2024-04-12:
There is some behavior with the dual-port dev kits that should be explained. This is with respect to macOS devices.

[image]

When plugged into the **UART** port, the device is enumerated as a `/dev/cu.usbserial-#####` device.

When plugged into the **USB** port, the device is enumerated as a `/dev/cu.usbmodem-#####` device.

The Espressif IDF will install to whatever is connected.

However, the debugger connection will only work in the method that the application was built for. That is, `esp32s3` will use the **UART** port to connect to the debugger (using the UPLOAD_PORT path). `esp32s3_cdc` or `esp32s3_usb` will connect over the **USB** port.

----

I would suggest using the `_cdc` instead of the `_usb` variant if possible (not possible on esp32s2). Your code will be smaller, as it will use the built-in Espressif serial-over-usb driver and does not need to include TinyUSB.

Another quirk with the `_usb` variant is that it enumerates to a different `/dev/cu.usbmodem-#####` whether it is in programming mode or run mode. For example, I see `/dev/cu.usbmodem123401` in programming mode. In run-mode, I see `/dev/cu.usbmodem1234561`. So I set `DEBUGGER_PORT=/dev/cu.usbmodem1234561` and `UPLOAD_PORT=/dev/cu.usbmodem123401` to ensure the proper connections are made.

sanielfishawy, 2024-04-12:
That is extremely helpful and explains everything.