Moddable-OpenSource/moddable #1337: Broken links  in documents of examples

stc1988, 2024-04-27:
* https://github.com/Moddable-OpenSource/moddable/tree/public/examples/piu
* https://github.com/Moddable-OpenSource/moddable/tree/public/examples/commodetto

These documents show no images.
Link that starts with `https://moddable.com/assets/` seems to unavailable.

phoddie, 2024-04-28:
Thanks for the report. This may have happened when we upgraded the Moddable web site (it runs much faster now). We'll take a look.

phoddie, 2024-04-29:
The images are displaying again. The main issue was that the files didn't get moved in the server upgrade. There were a couple of image URLs that needed to have a file extension added in the markdown document.

stc1988, 2024-04-30:
Thank you for confirming. This issue seems  to be fixed.

closed 2024-04-30 by stc1988