Moddable-OpenSource/moddable #1341: Error when examples/data/qrcode launch on Simulator

stc1988, 2024-05-05:
**Build environment:** macOS
**Moddable SDK version:** [e97b452](https://github.com/Moddable-OpenSource/moddable/commit/e97b4526671ee09a808af83b53366d131b6da24c)
**Target device:** desktop simulator

**Steps to Reproduce**
1. Build  the `examples/data/qrcode l` app using this build command: `mcconfig -d -m -p mac`
2. See error `# Break: (host): module "piu/MC" not found!`

phoddie, 2024-05-05:
I can't reproduce this. It builds and runs normally. The `examples/data/qrcode` examples doesn't even include Piu. I'm not sure what's going on here. Would you recheck?

There is a qrcode example that uses Piu: `examples/piu/qrcode`. I tried that and it gave the error `# Break: (host): module "commodetto/Poco" not found!`. That's because of some Commodetto reorganization from a few months ago. I'll fix that.

stc1988, 2024-05-05:
Sorry, after I cleaned up my environment , it worked fine.

closed 2024-05-05 by stc1988

phoddie, 2024-05-05:
Thanks for confirming