Moddable-OpenSource/moddable #1354: Makefiles break when the module name or filename has the "#" character

cmidgley, 2024-06-08:
**Build environment:** Windows
**Moddable SDK version:** OS201127-4913-gd487dd776
**Target device:** Any

**Description**
If a module name or filename has the `#` character the generated `makefile` from `mcconfig` breaks because of incorrect escaping (and `#` causes the remaining text on the line to become a comment)

**Steps to Reproduce**
1) In the `helloworld` example, on Windows, rename `main.js` to `#main.js` and adjust `manifest.json` to use `#main` as well
2) Run the build (`mcconfig -m -d -t build`)

A more advanced test would be to use a module name with `#` as the prefix, which has similar results.

```
mcconfig -m -d -t build 
c:\moddable\build\tmp\win\mc\debug\helloworld\makefile(52) : fatal error U1033: syntax error : '$(MODULES_DIR)\mc\config.xsb' unexpected
Stop.
```

**Background**
The problem is that the makefile is generated using `\` characters between file path elements on Windows (typical paths, like `c:\moddable\examples\helloworld`).  However, `\` is also the escape in makefiles for `#`, `$` and `\` (maybe others, not sure).  `mcconfig` does escape the `#` character (though not consistently) which results in a path such as `c:\moddable\examples\helloworld\\#main.js`.  The `\\` becomes just `\` and it's now `c:\moddable\examples\helloworld\#main.js` which means the makefile views `#main.js` as a comment and logically becomes `c:\moddable\examples\helloworld\` and the build breaks.

**Use case justificaton**
I came across this problem because Node's private module specifiers are [recommended](https://nodejs.org/api/packages.html#subpath-imports) to use `#` for private modules.  For example, an external module specifier might be `@moddable/xs` but a private module might be `#service` or `#service/database`.  My codebase is designed to compile under Node and Moddable, and to make this work I require imports to use private module names to adjust what they point to during the build to use the correct platform-specific files.  

I can use a different symbol as this is not a standard and just a recommendation (though `$` is slightly dangerous as it can result in an environment variable expansion), but the best solution would be to be able to use the `#` symbol.

phoddie, 2024-06-10:
We'd like to have compatibility with Node here, to the degree practical. As part of the work on `mcpack` the module resolution in XS hosts was updated to support `#` for Node private module resolution.

You describe the problem with the generated Windows path. I'm not a regular Windows user. What's the solution - what should the generated path be?

FWIW – I tried this quickly on macOS and it does build sources with `#` in the file name.

cmidgley, 2024-06-10:
It's a Windows issue, because it's all about the stupid `\` path separator.  I see that `mcconfig` already only escapes the `#` in certain places, which I think are those where rules are formed (and not executable commands) so that seems like the right place to make a change.  I've read forward slashes work on makefiles on Windows (did not test this) so it could be either changing the `\` to `\\` or to `/`.  Might be easier on the eyes if `/` works (`\\src\\lib\\\#module` is never fun to look at).

For now, I've switched to using `_` as my prefix, and moving forward nicely.  I'd switch back to `#` if this gets fixed, but it's not a blocking problem.

phoddie, 2024-06-10:
I'd rather not travel down the `/` road on Windows. That only leads to pain. The readability of autogenerated makefiles isn't a top priority.

> I see that mcconfig already only escapes the # in certain places, which I think are those where rules are formed (and not executable commands) so that seems like the right place to make a change.

It looks like there are only two places where that happens:

https://github.com/Moddable-OpenSource/moddable/blob/03fc51096a73fbf28dff7c338065ce08e91f18a0/tools/mcmanifest.js#L586

https://github.com/Moddable-OpenSource/moddable/blob/03fc51096a73fbf28dff7c338065ce08e91f18a0/tools/mcconfig.js#L93

Those changes were made as part of the `mcpack` work.

It seems like it could be enough to `call tool.resolveSlash()` before the replace for `#`.  Since you are set up to reproduce the problem and the number of changes is small (2), maybe you could try a quick fix? The results of that would give us clear guidance on how to proceed. Thank you.

cmidgley, 2024-06-11:
Thanks for the links to the source locations - that helped a lot.  I spent most of a day on this, to find several interesting things:

1) You can't use `\` as an escape on `#` for `nmake`.  Instead, the correct escape is `^` .
2) The code in `mcmanifest.js` currently substitutes `target` and uses that in both the rule and the command, but it needs to do the substitute for just the rule and not the command, plus it needs to do substitution on `source` as well.
3) With these changes I can now compile and link, however if I use a module name with `#` the code fails to execute (both Windows and ESP32).  In Windows simulator I get an `XS abort: unhandled exception` with nothing shown in `xsbug` (no modules, stack, etc).  In ESP32 it fails with:

```
# Exception: (host): module "#test" not found!
XS abort: unhandled exception
c:\moddable\xs\platforms\esp\xsPlatform.c (279) # Break: C: xsDebugger!
```
(where I was using a module named `#test` that `main` imports)

If I change the module name from `#test` to `test` in `manifest.json`, it runs fine on both platforms.  My test setup is `HelloWorld` plus a new `test.js` file that exports a function that `main.js` imports, and `manifest.json` is updated accordingly:

```
{
        "include": "$(MODDABLE)/examples/manifest_base.json",
        "modules": {
                "*": [
                        "./main"
                ],
                "#test": [
                        "./test"
                ]
        }
}
```

In looking at the `makefile` the only difference between it working and not working is the name of the file in the `MODULES` variable and the one rule that runs `xsc`.  Here is what they look like when `#` is used (remove the `^#` and it works):

```
MODULES =\
	$(MODULES_DIR)\Resource.xsb\
	$(MODULES_DIR)\screen.xsb\
	$(MODULES_DIR)\time.xsb\
	$(MODULES_DIR)\timer.xsb\
	$(MODULES_DIR)\main.xsb\
	$(MODULES_DIR)\^#test.xsb\
	... (more) ...
```
```
$(MODULES_DIR)\^#test.xsb: C:\moddable\examples\helloworld\test.js
	@echo # xsc #test.xsb
	xsc C:\moddable\examples\helloworld\test.js -d -c -e -o $(@D) -r #test
```

As best I can tell, the makefile issue is resolved and there is something else going on deeper inside `xsc` (I'm guessing there, since it occurs on both Windows and ESP32), but I've been unable to figure it out.

I have my trivial changes on a branch if you want a pull, but they are simple:

mcconfig.js, line 93, old:
```
this.write(result.target.replaceAll('#', '\\#'));
```
new:
```
this.write(result.target.replaceAll('#', '^#'));
```

mcmanifest.js, lines 585-589, old:
```
var target = result.target;
target = target.replaceAll('#', '\\#');
var targetParts = tool.splitPath(target);
this.line("$(MODULES_DIR)", tool.slash, target, ": ", source);
this.echo(tool, "xsc ", target);
```
new:
```
var target = result.target;
var targetParts = tool.splitPath(target);
this.line("$(MODULES_DIR)", tool.slash, target.replaceAll('#', '^#'), ": ", source.replaceAll('#', '^#'));
this.echo(tool, "xsc ", target);
```

Thoughts on what this second-order issue might be?

phoddie, 2024-06-11:
Thanks for putting so much effort into this. I did a test on macOS with a module file name of `#test.js` included in the manifest as:

```json
	"modules": {
		"*": [
			"./main",
			"./#test.js"
		]
	}
```

I confirmed that this generates byte code for a module specifier of "#test" by checking the generated `mc.xs.c` file for the project:

```c
static const txScript gxScripts[mxScriptsCount] = {
	{ NULL, NULL, 0, (txS1*)gxCode1, 1682, NULL, 0, "main", { XS_MAJOR_VERSION, XS_MINOR_VERSION, XS_PATCH_VERSION, 0 } },
	{ NULL, NULL, 0, (txS1*)gxCode2, 59, NULL, 0, "#test", { XS_MAJOR_VERSION, XS_MINOR_VERSION, XS_PATCH_VERSION, 0 } },
	...
}
```

Doing an import fails about as you describe:

```js
import TEST from "#test";
```

A [similar import statement](https://github.com/Moddable-OpenSource/moddable/blob/03fc51096a73fbf28dff7c338065ce08e91f18a0/examples/packages/hello/main.js#L1) works with the mcpack hello example. But the module specifier for `#test` in that case is `@moddable/example-hello/#test`.

Node private modules are, if my understanding is correct, always evaluated relative to the current module. But with a module specifier of `#test` in the generated archive (e.g. `gxScripts` above), it is not relative to anything. Since you are loading it from the `main` module, it needs to be relative to that. A little renaming in the manifest gets that to work:

```json
	"creation": {
		"main": "root/main"
	},
	"modules": {
		"root/*": "./*"
	}
```

It is a little strange to force a path for `main`, but for packages/modules it is normal.

FWIW – the resolution algorithm for Node (which XS tries to emulate as faithfully as practical for private modules) is in [here under "PACKAGE_IMPORTS_RESOLVE"](https://nodejs.org/docs/latest/api/esm.html#resolution-algorithm-specification)

phoddie, 2024-06-12:
This is feeling increasing academic to me. Apologies if I've lost the plot.

> In the above example, I am importing the private scoped module #test into my regular main. My manifest.json for the above might be...

I shared an example of how a private scoped module for main works. That approach works in Node too. A module specifier of "#test" is no scoped to anything.

You seem to want another way. Please remind me why.

phoddie, 2024-06-14:
Thanks for posting the example repo. Things are a bit busy at the moment, but I will give it a look.

cmidgley, 2024-06-14:
No rush, Peter - I'm progressing just fine with a workaround.  Thanks.

phoddie, 2024-06-14:
> On a related topic.... but that results in all files being in the root namespace

In your example project's manifest you have:

```json
		"./*": [
			"./modules/example/do-something"
		]
```

Change "./" to change the module specifier. For example...

```json
		"example/*": [
			"./modules/example/do-something"
		]
```

... changes the module specifier from `do-something` to `example/do-something`.

cmidgley, 2024-06-14:
I've tried as you suggest without success, but let's wait until we figure out private scoped modules so I'm confident I'm on working with a real issue and not something theoretical (as this could all still be me misunderstanding module resolution).  If the problem persists, I'll put together a small example (after also checking if it's only happening on TypeScript).

phoddie, 2024-06-14:
> I've tried as you suggest without success...

It solves the mapping problem you describe. It has nothing to do with the "#" thing. If you keep a "#" in your test, that will still fail, of course.

phoddie, 2024-06-14:
@cmidgley – I was looking at merging the changes [above](https://github.com/Moddable-OpenSource/moddable/issues/1354#issuecomment-2161409912) for `#` in nmake. The code being changed are used on all platforms but I **think** the changes are only for Windows nmake. If that's correct, the changes need to be conditionalized for Windows.

cmidgley, 2024-06-15:
Yes, good catch! I should have caught that, sorry.  See the pull with the proposed changes, which I've only tested on Windows - and there is a change that intentionally affects all platforms (escape only rules and not commands) as I _think_ that's correct for all makefiles.

cmidgley, 2024-06-16:
I started a new discussion with an example repo and details on reproduction for the namespace/import issue (not the private module use of `#`) to #1358.

closed 2024-06-18 by phoddie