Moddable-OpenSource/moddable #1355: xsbug not works with esp32s3 and ESP-IDF v5.2.2

stc1988, 2024-06-09:
**Build environment:** macOS
**Moddable SDK version:** 03fc510
**Target device:** M5stackCoreS3, M5Dial

**Steps to Reproduce**
1. Build and install the `example/piu/balls` app using this build command: `mcconfig -d -m -p esp32/m5stack_cores3`
2.  xsbug don't show the Machine Tab. 

**Other information**

* 4de60a3 + ESP-IDF v5.2.1 works fine.
* This issue  may be only with `esp32s3` subplatform. `esp32/m5tack_core2` works fine with ESP-IDF v5.2.2

phoddie, 2024-06-12:
This is now fixed. Thank you for reporting it. The fix is part of the `4.8.1` tag.

stc1988, 2024-06-12:
I confirmed this issue fixed. Thank you!!

phoddie, 2024-06-12:
Thanks for the quick confirmation. Subtle changes in ESP-IDF broke our CDC support. Fortunately, @mkellner was able to track that down.