Moddable-OpenSource/moddable #1358: Module import issues with Node modules

cmidgley, 2024-06-16:
**Build environment:** Windows
**Moddable SDK version:** Latest within a week, but on a private fork
**Target device:** Windows simulator and ESP32

**Description**
I am encountering several challenges getting modules to work with Moddable in a way that can also work with Node module resolution.  In #1354 we have an ongoing discussion about private scope modules (`#` prefix modules) and a repo demonstrating the issue.  I have isolated two more conditions, briefly discussed on #1354, that I've been unable to figure out and felt deserved a separate issue for discussion.

This [repo](https://github.com/cmidgley/module-resolution) has the test code to work with, and I've included the branches `main`, `root`, and `no-parents` for each test case (and also describe the small changes in each here).  It is a simple TypeScript program with a `main.ts` that consumes a local module `_mymodule`.  The private module has code at the root shared by all platforms, with Moddable-specific (`esp32`) and Node-specific (`node`) in subdirectories.   The tree structure is like this:

* `/`: Build configuration files like `package.json`, `tsconfig.json` and the root `manifest.json`
* `/src`: All source files, including `main.ts`
* `/src/mymodule`: Source for our private module, with it's `manifest.json` and code that is shared on `esp32` and `node`
* `/src/mymodule/esp32`: Code specific to Moddable (works with simulator or any Moddable target), with an `index.ts` file for the module's main exports.
* `/src/mymodule/node`: Code specific to Node, with an `index.ts` file for the module's main exports.

All three branches work with Node (just run `npm run start` on each to see the successful execution).  

### `main` branch: Named modules for source files fail to resolve

After cloning the repo, use `npm install` (`main` branch) and try running it as a Node module using `npm start`.  As you will see, it works fine and uses `console.log` on Node.  Then run it in Moddable (`npm run fs` or just `mcconfig -m -d`) and notice that it can't find the module `./example.js`.  The problem is the use of `_mymodule/*` in the `src/manifest.json` file.  

### `root` branch: Referencing parent directory files fail to resolve

If you change `_mymodule/*` to `./*` it will find and import the `example.js` module (though using `./*` is a problem, as it uses a shared namespace across all modules).  However, running this now results in the second problem. It will fail to load the `../shared.js` module.  This appears to be due to having `shared` one level higher in the tree and using `../shared.js` on the import.  

### `no-parents` branch: Working version

We can fix this by eliminating `..` on imports as follows:

1) Move `src/mymodule/shared.ts` to `src/mymodule/esp32/shared.ts`
2) Change the import in `esp32/example.ts` to import from `./shared.js` instead of `../shared.js`
3) Change the modules line in `src/manifest` to point to the new location (from `./shared` to `./esp32/shared`)

Now it runs without errors and uses `trace` to output to the console (showing it's platform independent).

## Summary

I understand that Moddable does not use the same module resolution system as Node, and these challenges may be inherent to how Moddable performs module resolution.  However, it makes using ESP32 and Node in the same module difficult, causes namespace pollution (and risk of duplicate names), and makes using third-party Node modules more challenging.  I felt it was worth discussing to see if there might be solutions to these challenges

cmidgley, 2024-06-16:
Update: for the `..` issue, a simple solution appears to be to add a new `/src/mymodule/index-esp32.ts` file (or whatever name you want) that has just `export * from "esp32/index.js";`.  The entry file at the root allows the `../shared.ts` import in `example.ts` to stay in the same parent tree relative to `index-esp32`.

```
/mymodule/index-esp32.ts <- "mymodule" is pointing here in manifest.json, and exports the "./esp32/index.js" file
/mymodule/esp32/index.ts <- the "real" index file for Moddable
/mymodule/esp32/example.ts <- uses `import { shared} from "../shared.js"`
/mymodule/shared.ts
```

A working version can be found in the `index-at-root` branch. This will work for my modules, though we may run into issues with Node packages that use non-root index files (but adding a new index-at-root in a fork would likely address it).  Of course, if this could be native to Moddable (supporting `..` above index root) that would be best, but this works and leaves only the module-name issue (`mymodule/*` vs `./*`), which I'm hoping is something I misunderstand or a simple issue to resolve??!?

phoddie, 2024-06-18:
Node has its own module resolution rules. They are complex and reasonably well documented. They may good sense for Node.

There are no plans to support the Node module resolution algorithm directly in XS or Moddable hosts.

To make it straightforward to work with modules that expect Node module resolution (e.g. npm) we created the `mcpack` tool. `mcpack` is a front-end to `mcconfig` and `mcrun` that analyzes the npm packages and creates a manifest that allows the modules provided by those packages to work within the Moddable SDK's module resolution rules. This involved some remapping in the manifest and creation of module aliases (like your "simple solution" above. It also involved some enhancements to the default module resolution algorithm (in `fxFindModule`) in our hosts.

`mcpack` was introduced in [Moddable SDK 4.2](https://github.com/Moddable-OpenSource/moddable/releases/tag/4.2.0#mcpack). Documentation is admittedly light but there is a [suite of examples](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/packages). We have reports of developers using it successfully with substantial npm packages.

The [private-scope-example](https://github.com/cmidgley/private-scope-example) you shared works by simply deleting your manifest.json and building with `mcpack mcconfig -d -m`.

We are considering adding direct support for module aliasing in the future to eliminate the stub alias modules. When that is done, `mcpack` will be updated to take advantage of that with a simple rebuild.

Long story short: take a look at `mcpack`. It is the supported solution for what you are trying to do

cmidgley, 2024-06-20:
Thanks - I've been able to rework the manifest and structure based on learnings from running `mcpack`.  Long story short - I learned that XS needs to witness the namespace path to the module in js code (not just the relative path) to establish the namespace.  `mcpack` does this by generating a `#module.js` file in another directory, but I'm doing this by using a relative import inside the same module directory to keep the files co-located ("import '../myModule/index.js").  

> I can't use `mcpack` because my use case isn't making `package.json` work on XS, but instead to have private scoped modules work across both Moddable (using `manifest.json` for Moddable-specific files) and Node (which uses `package.json` with node-specific files).  Also `mcpack` doesn't work with TypeScript.

So while all this has helped fix part of the problem, it still doesn't work with TypeScript.  The problem is that these namespace-relative paths (such as `root/#myModule`) are needed so XS has a path context, but they do not work with `paths` in `tsconfig.json`:

```json
"compilerOptions": {
    "paths": {
        "root/#example": [ "path/to/file" ]
    }
}
````

This must be `"#example": ["path/to/file"]` for TypeScript to be happy.  But I can't change the manifest to be just `#example` or XS will fail to load the module.

The `typescript` branch of [private-scope-example](https://github.com/cmidgley/private-scope-example/tree/typescript) has a small example of the Typescript issue (the `main` branch has the updated and working JS version). 

> You must merge the latest update to `mcmanifest.js` in #1356 for typescript private modules to compile.

Please note - I'm not trying to prescribe how Moddable/XS should work.  I'm fine changing my code/structure/build system. But I do need to figure out a solution that:

1) Uses TypeScript 
2) Use private-scope (`#`) modules that internally use relative file references within their private scope
3) Use a common relative root for the source tree (`src/module1`, `src/module2`, ... or `src/modules/module1`, etc)
4) Can be compatible with Node/TypeScript

cmidgley, 2024-06-20:
I've been able work around the `tsconfig.json` issue above by writing a wrapper around `tsc` that post-processes `tsconfig.json` (after `mcconfig` generates it and before the `makefile` compiles it) that removes the `root/#` portion from the paths.  

When I run my multi-module program with this I encounter a new module resolution problem.  It appears that the module resolution trick using `root/#xxx` in the manifest doesn't work when referencing a private scope module from other private scope modules, and I've reproduced this without TypeScript.

See the `main` branch of the same [private-scope-example](https://github.com/cmidgley/private-scope-example) repo.  There are two private scope modules, `#example` and `#another`.  The `#another` module is imported and called by both `main.js` and `example/do-something.js`.  With both importing it, you will be told `module "#another" not found!`, but if it is removed from `example/do-something.js` (leaving it in `main.js), it runs correctly showing that the module itself is loading fine when using the root resolution.

phoddie, 2024-06-21:
> Also mcpack doesn't work with TypeScript

Starting at first principles... yes, it does. There's even an [example](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/packages/ts-hello). With [documentation](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/packages/readme.md#ts-hello). 

> ....my use case isn't making package.json work on XS, but instead to have private scoped modules work across both Moddable (using manifest.json for Moddable-specific files) and Node (which uses package.json with node-specific files). 

If you want / need to invent something new, that's your call. But it seems to be paralleling mcpack enough that it is worth seeing if it could work.

It doesn't seem like there's a reason not to use package.json for the parts that are shared between environments, as long as the tooling supports it.

And you can use mcpack with a manifest. The manifest can be embedded in the package.json (see [fetch](https://github.com/Moddable-OpenSource/moddable/blob/561e6aa1d2f3e40ecda5a0ee5d82b0317ece3049/examples/packages/fetch/package.json#L14-L22) and [run-mod](url) examples) or a separate file (see [balls](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/packages/balls/package.json) example).

cmidgley, 2024-06-23:
I believe I finally have module resolution working.   I eventually figured out how Moddable does resolution by using `mcpack` with a bunch of multi-module tests and then hand-editing the generated output to figure out what works and what doesn't.  I don't use `mcpack` (it doesn't like my project), but it was instrumental in figuring this out.

The one thing that I still must do to get it to work is to post-process the `tsconfig.json` file (discussed starting with [this comment](https://github.com/Moddable-OpenSource/moddable/issues/1358#issuecomment-2180824177)). It's not clear to me how to address this problem in a way that can work for others (and not use a post-processor).  The problem is that for the resolution to work, you need to add some artificial module component (such as `root/`) for Moddable but can't use that with `tsconfig.json`.  Currently, the manifest isn't told about why this root is being added so it has no context to make any decisions about removal.  Adding a field in the manifest to specify the root name so it can be removed from `tsconfig.json` would be ... awkward.

phoddie, 2024-06-24:
> I believe I finally have module resolution working. I eventually figured out how Moddable does resolution by using mcpack with a bunch of multi-module tests and then hand-editing the generated [...]

Glad to hear that you're able to infer your way to a solution. The idea is for `mcpack` to do the heavy lifting of dealing with the Node import algorithm and reduce that to something much simpler that can be more easily handled on a microcontroller. That also avoids having the Node resolution algorithm become a dependency on the embedded runtimes.

If you have packages that work with Node but not `mcpack`, we'd like to see those to try to get them working.  We've had really good success with `mcpack` converting package.json, so exceptions are really helpful to extend our compatibility.

> The one thing that I still must do to get it to work is to post-process the tsconfig.json file

tsconfig.json still feels mysterious to me. Many thing are very easy and a few seemingly simple things are impossible. Maybe you could open a separate issue so we can try to sort that out?

phoddie, 2024-07-03:
@cmidgley – When you have a chance, it would be helpful if you could share one (or more) of the npm packages that you aren't able to get working with `mcpack`. Thanks.

cmidgley, 2024-07-04:
> tsconfig.json still feels mysterious to me. Many thing are very easy and a few seemingly simple things are impossible. Maybe you could open a separate issue so we can try to sort that out?

Will do.

> When you have a chance, it would be helpful if you could share one (or more) of the npm packages that you aren't able to get working with `mcpack`

If I come across any, I'll let you know for sure.  However, I don't currently use npm packages this way and therefore do not use `mcpack`.  My code is an NPM-style (private) package, but not typical of other packages.  For example, I run native TypeScript (not pre-compiled JavaScript) by running with `tsx`.  My build system creates many files (`tsconfig.json`, `manifest.json`, `package.json`, and '#someModule.ts` files) to handle the cross-platform libraries for Node, XS/simulator or XS/esp32.  

In short, my use case is not representative of a normal NPM, not easy to tease apart for examples, and unlikely to be worth the effort to support with `mcpack`!

closed 2024-07-04 by cmidgley