Moddable-OpenSource/moddable #1360: ESP32 _analog.c does not handle calibration correctly

linfan68, 2024-06-21:
I found this chunk of code has a bug:
https://github.com/Moddable-OpenSource/moddable/blob/561e6aa1d2f3e40ecda5a0ee5d82b0317ece3049/modules/io/analog/esp32/_analog.c#L322
```C
void xs_analog_read_(xsMachine *the)
{
	int millivolts = -1;
	int raw;
	Analog analog = xsmcGetHostDataValidate(xsThis, xs_analog_destructor_);

	if (ESP_OK != adc_oneshot_read(analog->handle, analog->channel, &raw))
		modLog("analog onshot_read failed");
	
#if ADC_CALI_SCHEME_CURVE_FITTING_SUPPORTED || ADC_CALI_SCHEME_LINE_FITTING_SUPPORTED
	if (analog->port == 1) {
		adc_cali_raw_to_voltage(gADC1_cali_handle, raw, &millivolts);
	} 
#if ADC_PORTS > (1)
	else if (analog->port == 2) {
		adc_cali_raw_to_voltage(gADC2_cali_handle, raw, &millivolts);
	}
#endif
#endif
	if (-1 == millivolts)
		millivolts = raw;		// uncalibrated

	xsmcSetInteger(xsResult, (millivolts * ((1 << ADC_RESOLUTION) - 1)) / 3300);
}
```

only the **raw** value needs to do the resolution convert (`raw * ((1 << ADC_RESOLUTION) - 1)) / 3300`), but in this code, event the value is coming from adc_cali_raw_to_voltage (which should be in mV already) but still get an extra convert.

phoddie, 2024-06-21:
Thanks for the report. What do you think of changing the last three lines of the function to the following?

```c
	if (-1 == millivolts)
		millivolts = raw * ((1 << ADC_RESOLUTION) - 1)) / 3300;

	xsmcSetInteger(xsResult, millivolts);
```

linfan68, 2024-06-26:
> Thanks for the report. What do you think of changing the last three lines of the function to the following?
> 
> ```c
> 	if (-1 == millivolts)
> 		millivolts = raw * ((1 << ADC_RESOLUTION) - 1)) / 3300;
> 
> 	xsmcSetInteger(xsResult, millivolts);
> ```

Looks good to me. Sorry for the late reply...

closed 2024-06-26 by linfan68

phoddie, 2024-06-26:
Thanks for the review. I'll apply the change.