Moddable-OpenSource/moddable #1367: Serial not works on ESP-IDF5

stc1988, 2024-07-07:
**Build environment:** macOS
**Moddable SDK version:** 3bf5202
**Target device:** M5Stack, M5Stack_core2, Moddable_two

**Steps to Reproduce**
1. Build and install the app`examples/io/serial/echo` using this build command: `mcconfig -d -m  -p esp32/m5stack`
2. Quit xsbug
3. Open [esp-web-tools](https://esphome.github.io/esp-web-tools/) in browser
4. Connect device and see `Logs & Console`
5. Reset Device

Device keeps restarting.

**Other information**

It works in Moddable V4.1 and  ESP-IDF v.4.2.3, so it seem to be affected by updating to ESP-IDF v5

phoddie, 2024-07-16:
This **might** not be a bug.

You are using a debug build. On Moddable Two, that means that pins 1 and 3 are reserved for xsbug communication. Those pins are the default for a new serial connection, so the echo example is trying to use them. The Moddable SDK was [updated](https://github.com/Moddable-OpenSource/moddable/blob/3bf520227129bd7f4445f2422b18335346294db5/build/devices/esp32/xsProj-esp32/main/main.c#L261-L264) some time ago to **reserve** the pins used by xsbug so that a project **cannot interfere** with the xsbug communication.

Please try using an instrumented build (`-I` instead of `-d`) or set the `transmit` and `receive` serial constructor options to unused pins.

phoddie, 2024-08-07:
This issue has been idle for a bit. Is there an update?

stc1988, 2024-08-07:
I’m sorry for the delayed report. The situation is stuck, and using the instrumented build or setting  `transmit` and `receive` did not resolve it.
To add a bit of context, this issue stems from the recent problem of the stack-chan’s serial servo motor not working.

phoddie, 2024-08-08:
I looked into this a bit more. I used Moddable Two, since that was listed in your report.

For a debug build, using the default pins throws an exception as expected. Changing them (I used 16 for transmit and 17 for receive) does not throw an exception (as expected).

For an instrumented build, no exception is thrown with the default pins. But... it does seem to hang - at least the serial port is not responding. Changing the `port` from 2 to 1 allows it to work with the default pins. I don't know why. Changing the transmit and receive as above, does work with port 2.

After some further experimentation, it seems that in IDF v5 port 2 is incompatible with the default pins (transmit 1 and receive 3).

stc1988, 2024-08-11:
Instrumented build + using port 1 is works as expected.
Debug build also does not throw an exception but  it seems that serial communication does not works.

phoddie, 2024-10-11:
I'm not sure what you are testing now. The last time I tried this, it worked as expected. To make progress on this, it would help to restate the problem you are now seeing. Than you.

stc1988, 2024-10-12:
Instrumented build + using port 1 is works as expected on ESP32 device like moddble_two and m5stack.

Now I try to test with ESP32-S3 device `esp32/moddable_six_cdc`.
It doesn’t seem like an error has occurred, but it doesn’t appear that there is an echo response from the device.

phoddie, 2024-10-14:
> Instrumented build + using port 1 is works as expected on ESP32 device like moddble_two and m5stack.

Great.

> Now I try to test with ESP32-S3 device esp32/moddable_six_cdc.

Since you indicate `moddable_six_cdc` instead of `moddable_six`, am I correct to understand that you are connected to the Micro USB port and **not** the serial programmer port?

stc1988, 2024-10-15:
> Since you indicate moddable_six_cdc instead of moddable_six, am I correct to understand that you are connected to the Micro USB port and not the serial programmer port?

That’s right. However, it worked as expected when using the serial programmer port. Thank you.​

> To add a bit of context, this issue stems from the recent problem of the stack-chan’s serial servo motor not working.
This issue persists, but since it seems to be unrelated to the initial case, I will create a separate issue.

closed 2024-10-15 by stc1988

phoddie, 2024-10-15:
OK, good. Understand that the USB port on Moddable Six (and most ESP32-S3 boards) is **not** a serial port from the perspective of the ESP32-S3. It really is USB. USB can run a virtual serial connection (CDC) but it is a completely different API. It is a little confusing because from the computer the connection looks and behaves like a serial connection.

Happy to try to help sort out the stack-chan servo motor issue. That's important. Please let me know.