Moddable-OpenSource/moddable #1368: SSL certification error to OpenAI API

stc1988, 2024-07-09:
**Build environment:** macOS
**Moddable SDK version:** 3bf520227129bd7f4445f2422b18335346294db5
**Target device:** Mac simulator

**Steps to Reproduce**
1. [Set OpenAI API key](https://github.com/Moddable-OpenSource/moddable/blob/3bf520227129bd7f4445f2422b18335346294db5/examples/pins/audioout/openai-stream/main.js#L18). If you don't have api key, remove line 19 and 20.
2. Build and install the app `examples/pins/audioout/openai-stream` using this build command: `mcconfig -d -m  -p mac`
3. Shows error `/Users/satoshi/Projects/moddable/modules/files/resource/Resource.c (44) # Break: Resource: Resource not found: ca236.der!`
4. Fix `manifest_openaistreamer.json` to set `ca236` and re-run app.
5. shows error `/Users/satoshi/Projects/moddable/modules/crypt/etc/x509.js (140) # Break: Error: x509: unsupported curve!`

phoddie, 2024-07-09:
That's unusual. That said, not all curves are supported. You could try disabling use of ECC to fall back to another ciphersuite.

```json
	"config": {
		"tls": {
			"DHE_RSA": false,
			"ECDHE_RSA": false
		}
	}
```

stc1988, 2024-07-09:
I added the config section to `manifest.json`, and see  another error `/Users/satoshi/Projects/moddable/modules/crypt/ssl/ssl_alert.js (83) # Break: Error: alert: 2, 40!`.

phoddie, 2024-07-15:
Thank you for trying. This may be more complex to address. I'll take a look in the coming days.

phoddie, 2024-07-15:
Some good news. Turning off TLS certificate validation allows the connection to succeed. (Of course, you still need to add ca236.der).

While not a long term solution, this allow things to continue working while working on a proper solution. Please give it a try by modifying these [three lines in openaistreamer.js](https://github.com/Moddable-OpenSource/moddable/blob/3bf520227129bd7f4445f2422b18335346294db5/examples/pins/audioout/openai-stream/openaistreamer.js#L46-L48) as shown below.

```js
		const http = {...device.network.https};
		http.socket = {...http.socket, secure: {...http.socket.secure, verify: false}};
		return new streamer({
			...o,
			http,
```

The actual problem appears to occur while parsing the response. In one case, the curve string has`undefined` where a number is expected. I'll look into that.

stc1988, 2024-07-16:
> While not a long term solution, this allow things to continue working while working on a proper solution. Please give it a try by modifying these [three lines in openaistreamer.js](https://github.com/Moddable-OpenSource/moddable/blob/3bf520227129bd7f4445f2422b18335346294db5/examples/pins/audioout/openai-stream/openaistreamer.js#L46-L48) as shown below.
> 
> 		const http = {...device.network.https};
> 		http.socket = {...http.socket, secure: {...http.socket.secure, verify: false}};
> 		return new streamer({
> 			...o,
> 			http,

Thank you for suggestion, I confirmed this code works.

phoddie, 2024-07-16:
I cannot reproduce the `undefined` behavior above. Looking deeper, it is necessary to implement the `secp384r1` curve for this certificate. I have that working, but it needs some more testing.

stc1988, 2024-07-20:
This issue fixed in [4.9.0](https://github.com/Moddable-OpenSource/moddable/releases/tag/4.9.0).

Thank you.

closed 2024-07-20 by stc1988