Moddable-OpenSource/moddable #1372: Worker module import fails with module pathing

cmidgley, 2024-07-18:
**Build environment:** Windows
**Moddable SDK version:** Latest release
**Target device:** Win and ESP32

**Description**
When using pathing in the `manifest.json` that supports private-scope modules, creating a new `Worker` is unable to load the module (using a simple path of `./my-module-name.js`).  Importing that path (with `await import('./my-module-name.js')`) works, showing the module resolution itself is correct.

For example, from the [simple example repo](https://github.com/cmidgley/worker-module-import) in the `example/startWorker.js` file:

```js
import Worker from 'worker'

export async function startWorker() {
    // this will successfully import the module, showing that the path itself is correct
    trace('in startWorker - importing "./worker.js"\n');
    const importedWorker = await import('./worker.js');

    // this will fail to import the module (with or without prior importing of the module, as above)
    trace('import done; starting worker\n');
    const worker = new Worker('./worker.js');

    trace('worker started\n');
}
```

Using a simple module name, such as just 'worker' in 'manifest.json' and starting the worker with `new Worker('worker');` does correctly load and execute the worker module, however, any further imports that it might do fail because the module resolution context would be incorrect.

**Steps to Reproduce**
1. Clone this [repo](https://github.com/cmidgley/worker-module-import)
2. Build on `win` or `esp32` and execute
3. Notice that it successfully loads the module using runtime `import` but it fails when trying to start a `Worker`.  The behavior is slightly different from `win` to `esp32` (IIRC `win` reports the failure in the main thread whereas `esp32` reports the error in a worker thread), but either way it does not load.

phoddie, 2024-07-18:
Interesting. Will take a look. Perhaps a difference in the way import is invoked directly from JavaScript and from XS in C.

cmidgley, 2024-07-19:
Dont know if this helps or not, but I found that this works:

```js
const worker = new Worker('root/example/worker.js');
```

phoddie, 2024-07-19:
Perhaps unsurprising, but macOS seems to fail in the same way as you describe Windows failing.

phoddie, 2024-07-19:
Oh... so maybe the Worker import only expects an absolute specifier. It may not be using the context of the importing module, so it can't resolve the relative specifier. That would explain it.

phoddie, 2024-08-05:
After looking into this, `Worker` only supports absolute specifiers. Recall that the module resolution isn't really using paths, though often the resolver is able to give the illusion that it does. The Worker constructor is one place where that illusion breaks down.

As a workaround, you can resolve the relative path to an absolute specifier before calling the `Worker` constructor using the `URL` class:

```js
let url = new URL( './worker.js', "file:///" + import.meta.uri);
const worker = new Worker(url.toString().slice(8));
```

cmidgley, 2024-08-31:
Thanks, Peter.  This works on simulator (win), but under ESP32 it also needs to have the ".js" extension removed, such as:

```js
const worker = new Worker(url.toString().slice(8).replace('.js', ''));
```

closed 2024-08-31 by cmidgley

phoddie, 2024-09-02:
Glad this works. As much as practical, we want to minimize differences between hosts so I took a quick look into the comment about needing to remove the `.js` extension. 

The embedded module loader strips the `.js` extension when looking up modules. See [here](https://github.com/Moddable-OpenSource/moddable/blob/24e3e54fd3b66379c1f91ad92969c99f941ff4cf/xs/platforms/mc/xsHosts.c#L268-L273). And... in my quick test, that worked on both ESP32 and macOS. Here's the modified `startWorker.js` I used:

```js
import Worker from 'worker'
import URL from "url"

export async function startWorker() {
	let url = new URL( './worker.js', "file:///" + import.meta.uri);
	url = url.toString().slice(8);
	trace(`starting worker ${url}\n`);
	const worker = new Worker(url);
    trace('worker started\n');
}
```