Moddable-OpenSource/moddable #1377: Moddable six migration

mauroForlimpopoli, 2024-07-23:
I'm migrating from Moddable two to six.

I updated all the software for the build and I tried the led-color example in the directory "moddable six" and it works. (but only in the USB programming). So I tried to compile my old program (changing some little things) but I get this error.
How can I solve?
I'm using windows11 
regards

```
C:\Users\mades\projects\moddable\examples\piu\ora>mcconfig -d -m -p esp32/moddable_six_cdc
mcconfig: file storage partition of size 0x10000 created from factory app partition
INFORMAZIONI: impossibile trovare file corrispondenti ai
criteri di ricerca indicati.
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Users\mades\esp32\esp-idf\
"null" non è riconosciuto come comando interno o esterno,
 un programma eseguibile o un file batch.
Using Git in C:\Esp32\tools\idf-git\2.44.0\cmd\
git version 2.44.0.windows.1
Checking Python compatibility
Setting IDF_PATH: C:\Users\mades\esp32\esp-idf

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.28.3.
Not using an unsupported version of tool ninja found in PATH: 1.11.0.
    C:\Esp32\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin
    C:\Esp32\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin
    C:\Esp32\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin
    C:\Esp32\tools\riscv32-esp-elf\esp-13.2.0_20230928\riscv32-esp-elf\bin
    C:\Esp32\tools\esp32ulp-elf\2.35_20220830\esp32ulp-elf\bin
    C:\Esp32\tools\cmake\3.24.0\bin
    C:\Esp32\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin
    C:\Esp32\tools\ninja\1.11.1\
    C:\Esp32\tools\idf-exe\1.0.3\
    C:\Esp32\tools\ccache\4.8\ccache-4.8-windows-x86_64
    C:\Esp32\tools\dfu-util\0.11\dfu-util-0.11-win64
    C:\Esp32\python_env\idf5.2_py3.12_env\Scripts
    C:\Users\mades\esp32\esp-idf\tools

Checking if Python packages are up to date...
Constraint file: C:\Esp32\espidf.constraints.v5.2.txt
Requirement files:
 - C:\Users\mades\esp32\esp-idf\tools\requirements\requirements.core.txt
Python being checked: C:\Esp32\python_env\idf5.2_py3.12_env\Scripts\python.exe
Python requirements are satisfied.

Detected installed tools that are not currently used by active ESP-IDF version.
For removing old versions of amazon-corretto-11-x64-windows-jdk, espressif-ide, idf-driver, idf-python-wheels, ninja, openocd-esp32, riscv32-esp-elf, riscv32-esp-elf-gdb, xtensa-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf use command 'python.exe C:\Users\mades\esp32\esp-idf\tools\idf_tools.py uninstall'
For free up even more space, remove installation packages of those tools. Use option 'python.exe C:\Users\mades\esp32\esp-idf\tools\idf_tools.py uninstall --remove-archives'.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Recommend using ESP-IDF v5.2.1 (found v5.2.2)
  See update instructions at: https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/esp32.md
# bles2gatt bleservices
        1 file copiati.
        1 file copiati.
        1 file copiati.
        1 file copiati.
        1 file copiati.
Reconfiguring ESP-IDF...
Executing action: reconfigure
Running cmake in directory C:\Users\mades\projects\moddable\build\tmp\esp32\moddable_six_cdc\debug\ora\xsProj-esp32s3\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Esp32\python_env\idf5.2_py3.12_env\Scripts\python.exe -DESP_PLATFORM=1 -DSDKCONFIG_DEFAULTS=C:/Users/mades/Projects/moddable/build/tmp/esp32/moddable_six_cdc/debug/ora/xsProj-esp32s3/sdkconfig.mc -DSDKCONFIG_HEADER=C:\Users\mades\Projects\moddable\build\tmp\esp32\moddable_six_cdc\debug\ora\xsProj-esp32s3\build\config\sdkconfig.h -DCMAKE_MESSAGE_LOG_LEVEL=ERROR -DDEBUGGER_SPEED=460800 -DIDF_TARGET=esp32s3 -DESP32_SUBCLASS=esp32s3 -DSDKCONFIG_DEFAULTS=C:\Users\mades\Projects\moddable\build\tmp\esp32\moddable_six_cdc\debug\ora\xsProj-esp32s3\sdkconfig.mc -DCCACHE_ENABLE=1 C:\Users\mades\projects\moddable\build\tmp\esp32\moddable_six_cdc\debug\ora\xsProj-esp32s3"...
C:\Users\mades\AppData\Local\Temp\kconfgen_tmpxlbnoakd:1435: warning: ignoring malformed line 'SPIRAM_USE_CAPS_ALLOC=y'
Loading defaults file C:/Users/mades/projects/moddable/build/tmp/esp32/moddable_six_cdc/debug/ora/xsProj-esp32s3/sdkconfig.mc...
C:/Users/mades/projects/moddable/build/tmp/esp32/moddable_six_cdc/debug/ora/xsProj-esp32s3/sdkconfig.mc:1425 CONFIG_ESP32S3_SPIRAM_SUPPORT was replaced with CONFIG_SPIRAM
C:/Users/mades/projects/moddable/build/tmp/esp32/moddable_six_cdc/debug/ora/xsProj-esp32s3/sdkconfig.mc:1447 CONFIG_CONSOLE_UART_DEFAULT was replaced with CONFIG_ESP_CONSOLE_UART_DEFAULT
[... 166 lines cut ...]
# xsid modGPIO.h.xsi
# xsid led_strip_encoder.h.xsi
# xsid neopixel.h.xsi
# xsid modTimer.h.xsi
# xsid modPreference.h.xsi
# xsid modInstrumentation.h.xsi
# xsid modBLE.h.xsi
# xsid modBLECommon.h.xsi
# xsid modI2C.h.xsi
# xsid owb.h.xsi
# xsid owb_gpio.h.xsi
# xsid owb_rmt.h.xsi
# xsid commodettoBitmap.h.xsi
# xsid commodettoPocoBlit.h.xsi
# xsid piuAll.h.xsi
# xsid piuMC.h.xsi
# xsl modules
ReferenceError: get Image: undefined variable
 at ()
### ReferenceError: get Image: undefined variable!
NMAKE : fatal error U1077: 'xsl args.txt' : codice restituito '0xd'
Stop.

C:\Users\mades\projects\moddable\examples\piu\ora>
```

mauroForlimpopoli, 2024-07-23:
Sorry, maybe my migration is too complicated to affront it in this way. So I tried to inset one piece at a time.
So in this way I have the follow problems:
1)
```
const iconBar = Image.template($ => ({

	path: $,

}));
```
debugger: C:\Users\mades\projects\moddable\examples\piu\six\assets.js (472) # Break: assets: get Image: undefined variable!
2)
These 3 imports
```
import OTA from "esp32/ota";
or 
import OTA from "ota";
```
I followed the instruction of "contributed/modClock/update.js" or "examples/network/http/httpota/main.js" but i get the error "# Break: (host): module "esp32/ota" not found!" or "# Break: (host): module "ota" not found!"
What can I do?
Regards

mkellner, 2024-07-23:
@mauroForlimpopoli, can you confirm that the application works on Moddable Two?

phoddie, 2024-07-23:
@mauroForlimpopoli - FWIW, these two examples work on Moddable Six:

- [HTTP OTA](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/network/http/httpota)
- [Piu Image](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/piu/images)

mauroForlimpopoli, 2024-07-24:
this is the error compiling the example "HTTP OTA"
```
In file included from C:\Users\mades\Projects\moddable\build\devices\esp32\modules\ota\ota.c:23:
C:\Users\mades\esp32\esp-idf\components/app_update/include/esp_ota_ops.h:16:10: fatal error: esp_bootloader_desc.h: No such file or directory
   16 | #include "esp_bootloader_desc.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

phoddie, 2024-07-24:
> this is the error compiling the example "HTTP OTA"

That was resolved by [this commit](https://github.com/Moddable-OpenSource/moddable/commit/3fe65e93a775773ca70b253bd80c536692a450b1) yesterday.

Did the Piu Image example work for you?

mauroForlimpopoli, 2024-07-24:
the "OTA" now works. 
This is the error for "Piu Image"
```
[95/965] Building C object esp-idf/mbedtls/mbedtls/library...edtls/port/crypto_shared_gdma/esp_crypto_shared_gdma.c.obj
FAILED: esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/C_/Users/mades/esp32/esp-idf/components/mbedtls/port/crypto_shared_gdma/esp_crypto_shared_gdma.c.obj
ccache C:\Esp32\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin\xtensa-esp32s3-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"v5.2.2\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -IC:/Users/mades/projects/moddable/build/tmp/esp32/moddable_six_cdc/debug/images/xsProj-esp32s3/build/config -IC:/Users/mades/esp32/esp-idf/components/mbedtls/port/include -IC:/Users/mades/esp32/esp-idf/components/mbedtls/mbedtls/include -IC:/Users/mades/esp32/esp-idf/components/mbedtls/mbedtls/library -IC:/Users/mades/projects/moddable/build/tmp/esp32/moddable_six_cdc/debug/images/xsProj-esp32s3/build/esp-idf/mbedtls/mbedtls/library -IC:/Users/mades/esp32/esp-idf/components/mbedtls/port/sha/dma/include -IC:/Users/mades/esp32/esp-idf/components/mbedtls/port/aes/dma/include -IC:/Users/mades/esp32/esp-idf/components/newlib/platform_include -IC:/Users/mades/esp32/esp-idf/components/freertos/config/include -IC:/Users/mades/esp32/esp-idf/components/freertos/config/include/freertos -IC:/Users/mades/esp32/esp-idf/components/freertos/config/xtensa/include -IC:/Users/mades/esp32/esp-idf/components/freertos/FreeRTOS-Kernel/include -IC:/Users/mades/esp32/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Users/mades/esp32/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -IC:/Users/mades/esp32/esp-idf/components/freertos/esp_additions/include -IC:/Users/mades/esp32/esp-idf/components/esp_hw_support/include -IC:/Users/mades/esp32/esp-idf/components/esp_hw_support/include/soc -IC:/Users/mades/esp32/esp-idf/components/esp_hw_support/include/soc/esp32s3 -IC:/Users/mades/esp32/esp-idf/components/esp_hw_support/port/esp32s3/. -IC:/Users/mades/esp32/esp-idf/components/heap/include -IC:/Users/mades/esp32/esp-idf/components/log/include -IC:/Users/mades/esp32/esp-idf/components/soc/include -IC:/Users/mades/esp32/esp-idf/components/soc/esp32s3 -IC:/Users/mades/esp32/esp-idf/components/soc/esp32s3/include -IC:/Users/mades/esp32/esp-idf/components/hal/platform_port/include -IC:/Users/mades/esp32/esp-idf/components/hal/esp32s3/include -IC:/Users/mades/esp32/esp-idf/components/hal/include -IC:/Users/mades/esp32/esp-idf/components/esp_rom/include -IC:/Users/mades/esp32/esp-idf/components/esp_rom/include/esp32s3 -IC:/Users/mades/esp32/esp-idf/components/esp_rom/esp32s3 -IC:/Users/mades/esp32/esp-idf/components/esp_common/include -IC:/Users/mades/esp32/esp-idf/components/esp_system/include -IC:/Users/mades/esp32/esp-idf/components/esp_system/port/soc -IC:/Users/mades/esp32/esp-idf/components/esp_system/port/include/private -IC:/Users/mades/esp32/esp-idf/components/xtensa/esp32s3/include -IC:/Users/mades/esp32/esp-idf/components/xtensa/include -IC:/Users/mades/esp32/esp-idf/components/xtensa/deprecated_include -IC:/Users/mades/esp32/esp-idf/components/lwip/include -IC:/Users/mades/esp32/esp-idf/components/lwip/include/apps -IC:/Users/mades/esp32/esp-idf/components/lwip/include/apps/sntp -IC:/Users/mades/esp32/esp-idf/components/lwip/lwip/src/include -IC:/Users/mades/esp32/esp-idf/components/lwip/port/include -IC:/Users/mades/esp32/esp-idf/components/lwip/port/freertos/include -IC:/Users/mades/esp32/esp-idf/components/lwip/port/esp32xx/include -IC:/Users/mades/esp32/esp-idf/components/lwip/port/esp32xx/include/arch -IC:/Users/mades/esp32/esp-idf/components/lwip/port/esp32xx/include/sys -IC:/Users/mades/esp32/esp-idf/components/mbedtls/mbedtls/3rdparty/everest/include -IC:/Users/mades/esp32/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m -IC:/Users/mades/esp32/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -mlongcalls  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wall -Wextra -Wwrite-strings -Wformat=2 -Wno-format-nonliteral -Wvla -Wlogical-op -Wshadow -Wformat-signedness -Wformat-overflow=2 -Wformat-truncation -Werror -Wmissing-declarations -Wmissing-prototypes -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Os -freorder-blocks -fmacro-prefix-map=C:/Users/mades/projects/moddable/build/tmp/esp32/moddable_six_cdc/debug/images/xsProj-esp32s3=. -fmacro-prefix-map=C:/Users/mades/esp32/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu17 -Wno-old-style-declaration -MD -MT esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/C_/Users/mades/esp32/esp-idf/components/mbedtls/port/crypto_shared_gdma/esp_crypto_shared_gdma.c.obj -MF esp-idf\mbedtls\mbedtls\library\CMakeFiles\mbedcrypto.dir\C_\Users\mades\esp32\esp-idf\components\mbedtls\port\crypto_shared_gdma\esp_crypto_shared_gdma.c.obj.d -o esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/C_/Users/mades/esp32/esp-idf/components/mbedtls/port/crypto_shared_gdma/esp_crypto_s [... cut ...]
[100/965] Building C object esp-idf/mbedtls/mbedtls/librar...sp32/esp-idf/components/mbedtls/port/aes/dma/esp_aes.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\mades\projects\moddable\build\tmp\esp32\moddable_six_cdc\debug\images\xsProj-esp32s3\build\log\idf_py_stderr_output_11124 and C:\Users\mades\projects\moddable\build\tmp\esp32\moddable_six_cdc\debug\images\xsProj-esp32s3\build\log\idf_py_stdout_output_11124
Launching app...
Type Ctrl-C twice after debugging app.
```

mkellner, 2024-07-24:
@mauroForlimpopoli, it is strange.
For the example `piu/images`, I can build and deploy from a Windows host.

Can you try building clean first? That is: `mcconfig -d -m -p esp32/moddable_six_cdc -t clean` followed by a normal build.

mauroForlimpopoli, 2024-07-25:
mow this is the error
```
[100/965] Building C object esp-idf/mbedtls/mbedtls/librar...edtls/port/crypto_shared_gdma/esp_crypto_shared_gdma.c.obj
FAILED: esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/C_/Users/mades/esp32/esp-idf/components/mbedtls/port/crypto_shared_gdma/esp_crypto_shared_gdma.c.obj
ccache C:\Esp32\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin\xtensa-esp32s3-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"v5.2.2\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -IC:/Users/mades/projects/moddable/build/tmp/esp32/moddable_six_cdc/debug/images/xsProj-esp32s3/build/config -IC:/Users/mades/esp32/esp-idf/components/mbedtls/port/include -IC:/Users/mades/esp32/esp-idf/components/mbedtls/mbedtls/include -IC:/Users/mades/esp32/esp-idf/components/mbedtls/mbedtls/library -IC:/Users/mades/projects/moddable/build/tmp/esp32/moddable_six_cdc/debug/images/xsProj-esp32s3/build/esp-idf/mbedtls/mbedtls/library -IC:/Users/mades/esp32/esp-idf/components/mbedtls/port/sha/dma/include -IC:/Users/mades/esp32/esp-idf/components/mbedtls/port/aes/dma/include -IC:/Users/mades/esp32/esp-idf/components/newlib/platform_include -IC:/Users/mades/esp32/esp-idf/components/freertos/config/include -IC:/Users/mades/esp32/esp-idf/components/freertos/config/include/freertos -IC:/Users/mades/esp32/esp-idf/components/freertos/config/xtensa/include -IC:/Users/mades/esp32/esp-idf/components/freertos/FreeRTOS-Kernel/include -IC:/Users/mades/esp32/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IC:/Users/mades/esp32/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos -IC:/Users/mades/esp32/esp-idf/components/freertos/esp_additions/include -IC:/Users/mades/esp32/esp-idf/components/esp_hw_support/include -IC:/Users/mades/esp32/esp-idf/components/esp_hw_support/include/soc -IC:/Users/mades/esp32/esp-idf/components/esp_hw_support/include/soc/esp32s3 -IC:/Users/mades/esp32/esp-idf/components/esp_hw_support/port/esp32s3/. -IC:/Users/mades/esp32/esp-idf/components/heap/include -IC:/Users/mades/esp32/esp-idf/components/log/include -IC:/Users/mades/esp32/esp-idf/components/soc/include -IC:/Users/mades/esp32/esp-idf/components/soc/esp32s3 -IC:/Users/mades/esp32/esp-idf/components/soc/esp32s3/include -IC:/Users/mades/esp32/esp-idf/components/hal/platform_port/include -IC:/Users/mades/esp32/esp-idf/components/hal/esp32s3/include -IC:/Users/mades/esp32/esp-idf/components/hal/include -IC:/Users/mades/esp32/esp-idf/components/esp_rom/include -IC:/Users/mades/esp32/esp-idf/components/esp_rom/include/esp32s3 -IC:/Users/mades/esp32/esp-idf/components/esp_rom/esp32s3 -IC:/Users/mades/esp32/esp-idf/components/esp_common/include -IC:/Users/mades/esp32/esp-idf/components/esp_system/include -IC:/Users/mades/esp32/esp-idf/components/esp_system/port/soc -IC:/Users/mades/esp32/esp-idf/components/esp_system/port/include/private -IC:/Users/mades/esp32/esp-idf/components/xtensa/esp32s3/include -IC:/Users/mades/esp32/esp-idf/components/xtensa/include -IC:/Users/mades/esp32/esp-idf/components/xtensa/deprecated_include -IC:/Users/mades/esp32/esp-idf/components/lwip/include -IC:/Users/mades/esp32/esp-idf/components/lwip/include/apps -IC:/Users/mades/esp32/esp-idf/components/lwip/include/apps/sntp -IC:/Users/mades/esp32/esp-idf/components/lwip/lwip/src/include -IC:/Users/mades/esp32/esp-idf/components/lwip/port/include -IC:/Users/mades/esp32/esp-idf/components/lwip/port/freertos/include -IC:/Users/mades/esp32/esp-idf/components/lwip/port/esp32xx/include -IC:/Users/mades/esp32/esp-idf/components/lwip/port/esp32xx/include/arch -IC:/Users/mades/esp32/esp-idf/components/lwip/port/esp32xx/include/sys -IC:/Users/mades/esp32/esp-idf/components/mbedtls/mbedtls/3rdparty/everest/include -IC:/Users/mades/esp32/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m -IC:/Users/mades/esp32/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -mlongcalls  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wall -Wextra -Wwrite-strings -Wformat=2 -Wno-format-nonliteral -Wvla -Wlogical-op -Wshadow -Wformat-signedness -Wformat-overflow=2 -Wformat-truncation -Werror -Wmissing-declarations -Wmissing-prototypes -fdiagnostics-color=always -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Os -freorder-blocks -fmacro-prefix-map=C:/Users/mades/projects/moddable/build/tmp/esp32/moddable_six_cdc/debug/images/xsProj-esp32s3=. -fmacro-prefix-map=C:/Users/mades/esp32/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu17 -Wno-old-style-declaration -MD -MT esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/C_/Users/mades/esp32/esp-idf/components/mbedtls/port/crypto_shared_gdma/esp_crypto_shared_gdma.c.obj -MF esp-idf\mbedtls\mbedtls\library\CMakeFiles\mbedcrypto.dir\C_\Users\mades\esp32\esp-idf\components\mbedtls\port\crypto_shared_gdma\esp_crypto_shared_gdma.c.obj.d -o esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/C_/Users/mades/esp32/esp-idf/components/mbedtls/port/crypto_shared_gdma/esp_crypto_shared_gdma.c.obj -c C:/Users/mades [... cut ...]
[105/965] Building C object esp-idf/mbedtls/mbedtls/librar...32/esp-idf/components/mbedtls/port/bignum/bignum_alt.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\mades\projects\moddable\build\tmp\esp32\moddable_six_cdc\debug\images\xsProj-esp32s3\build\log\idf_py_stderr_output_4936 and C:\Users\mades\projects\moddable\build\tmp\esp32\moddable_six_cdc\debug\images\xsProj-esp32s3\build\log\idf_py_stdout_output_4936
Launching app...
Type Ctrl-C twice after debugging app.
```

mkellner, 2024-07-25:
The error looks about the same.
Are there any changes made to your `$MODDABLE` tree? ie. `cd $MODDABLE; git diff`

phoddie, 2024-07-29:
@mauroForlimpopoli – were if you able to get past this issue?

mauroForlimpopoli, 2024-07-30:
Not resolved, At the moment I used new Content rather than new Image......

phoddie, 2024-07-30:
> I used new Content rather than new Image......

That cannot possibly have resolved the compilation error. How did you resolve the compilation error above?

mauroForlimpopoli, 2024-08-06:
I did not more test with new Image.....

Now I have a problem including the Onewire driver. 
By running the demo version I got this error 
```
In file included from C:\Users\mades\Projects\moddable\modules\drivers\onewire\esp\modonewire.c:34:
C:\Users\mades\Projects\moddable\modules\drivers\onewire\esp\owb_rmt.h:34:10: fatal error: driver/rmt.h: No such file or directory
   34 | #include "driver/rmt.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
```
When I run my project this is the error
```
FAILED: xs_esp32.elf
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20230928\xtensa-esp-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S3=0 -Wl,--Map=C:/Users/mades/projects/moddable/build/tmp/esp32/moddable_six_cdc/debug/six/xsProj-esp32s3/build/xs_esp32.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s3.peripherals.ld -T esp32s3.rom.ld -T esp32s3.rom.api.ld -T esp32s3.rom.libgcc.ld -T esp32s3.rom.newlib.ld -T esp32s3.rom.version.ld -T memory.ld -T sections.ld @CMakeFiles\xs_esp32.elf.rsp -o xs_esp32.elf  && cd ."
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: xs_esp32s3.a(mc.xs.o):C:\Users\mades\Projects\moddable\build\tmp\esp32\moddable_six_cdc\debug\six/mc.xs.c:10184:(.rodata.gxBuilders31+0x0): undefined reference to `xs_onewire_destructor'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: xs_esp32s3.a(mc.xs.o):(.rodata.gxBuilders31+0xc): undefined reference to `xs_onewire'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: xs_esp32s3.a(mc.xs.o):(.rodata.gxBuilders31+0x18): undefined reference to `xs_onewire_close'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: xs_esp32s3.a(mc.xs.o):(.rodata.gxBuilders31+0x24): undefined reference to `xs_onewire_read'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: xs_esp32s3.a(mc.xs.o):(.rodata.gxBuilders31+0x30): undefined reference to `xs_onewire_write'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: xs_esp32s3.a(mc.xs.o):(.rodata.gxBuilders31+0x3c): undefined reference to `xs_onewire_select'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: xs_esp32s3.a(mc.xs.o):(.rodata.gxBuilders31+0x48): undefined reference to `xs_onewire_search'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: xs_esp32s3.a(mc.xs.o):(.rodata.gxBuilders31+0x54): undefined reference to `xs_onewire_isPresent'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: xs_esp32s3.a(mc.xs.o):(.rodata.gxBuilders31+0x60): undefined reference to `xs_onewire_reset'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: xs_esp32s3.a(mc.xs.o):(.rodata.gxBuilders31+0x6c): undefined reference to `xs_onewire_crc'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: xs_esp32s3.a(mc.xs.o):(.rodata.gxHeap+0x1fef8): undefined reference to `xs_onewire'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: xs_esp32s3.a(mc.xs.o):(.rodata.gxHeap+0x2229c): undefined reference to `xs_onewire_destructor'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\mades\projects\moddable\build\tmp\esp32\moddable_six_cdc\debug\six\xsProj-esp32s3\build\log\idf_py_stderr_output_14072 and C:\Users\mades\projects\moddable\build\tmp\esp32\moddable_six_cdc\debug\six\xsProj-esp32s3\build\log\idf_py_stdout_output_14072
```
this is my complete manifest

```
{
	"include": [
		"$(MODDABLE)/examples/manifest_base.json",
		"$(MODDABLE)/examples/manifest_piu.json",
		"$(MODULES)/input/keyboard/manifest.json",
		"$(MODDABLE)/modules/files/file/manifest.json",
		"$(MODULES)/pins/analog/manifest.json",
		"$(MODULES)/network/wifi/manifest.json",
		"$(MODDABLE)/examples/manifest_net.json",
		"$(MODDABLE)/modules/network/ble/manifest_server.json",
		"$(MODDABLE)/modules/io/manifest.json",
		"$(MODDABLE)/modules/network/ble/manifest_client.json",
		"$(MODDABLE)/modules/drivers/peripherals/ds1307/manifest.json",
		"$(MODDABLE)/modules/drivers/rtc/manifest.json",
		"$(MODULES)/pins/i2c/manifest.json",
		"$(MODULES)/network/http/manifest.json",
		"$(MODDABLE)/modules/files/preference/manifest.json",
		"$(MODDABLE)/modules/files/flash/manifest.json"
		
		
	],
	"creation":{
		"stack":500
	},
	"modules": {
[... cut ...]
			"$(MODULES)/network/sntp/*",
			"$(MODULES)/drivers/neopixel/*",
[... 114 lines cut ...]
			],
			"modules": {
				"*": [
				"$(MODULES)/files/flash/*",
				"$(MODULES)/files/flash/esp32/*",
				"$(MODULES)/drivers/onewire/esp/*"
				]
			},						
			"build": {
				"PARTITIONS_FILE": "./sdkconfig/partitions.csv"
			},"modules": {
				"ota": "$(BUILD)/devices/esp32/modules/ota/*"
			},
			"preload": "ota" 			
		}
    },
	"ble":{
		"*": [
			"./bleservices/*",
			"./bleservicesCli/*"
		]
	}	
}
```
Can you help me ?

mauroForlimpopoli, 2024-08-06:
... P.S:
Changing "$(MODULES)/drivers/onewire/*"
 in "$(MODULES)/drivers/onewire/sim/*"
the error above desappear, but the xbug give this error # Break: (host): module "onewire" not found!

mauroForlimpopoli, 2024-08-07:
I have another error from the Xsbug.
this :
import DS18X20 from "DS18X20";
give me  this error:
# Break: (host): module "DS18X20" not found!

phoddie, 2024-09-24:
OneWire support, including the DS18X20 temperature sensor, has been re-implemented for ESP-IDF v5. See #1395.

closed 2024-09-24 by phoddie