Moddable-OpenSource/moddable #1381: fetch fails when `content-length` and  "transfer-encoding" are not set.

stc1988, 2024-07-27:
**Build environment:** mac
**Moddable SDK version:** [3fe65e9](https://github.com/Moddable-OpenSource/moddable/commit/3fe65e93a775773ca70b253bd80c536692a450b1)
**Target device:** macOS simulator

**Steps to Reproduce**
1. Build and install the `fetch`  [example app](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/io/tcp/fetch) using this build command: `mcconfig -d -m  -p mac`
2. See error `/Users/satoshi/Projects/moddable/examples/io/tcp/fetch/fetch.js (206) # Break: (anonymous-104): cannot coerce undefined to object!`

**Other information**

This issue seems to occur in the case of [recent commit](https://github.com/Moddable-OpenSource/moddable/commit/764d23c0e78ca0b110190bc1701cd9a256e5098c) and the conditions mentioned in the title.

phoddie, 2024-07-31:
Thank you for the report. Yes, this problem was introduced with the recent change. Fortunately, the fix is simple. Change the line that fails from `if (transferEncoding.toLowerCase() != "chunked")` to `if (transferEncoding?.toLowerCase() != "chunked")`.

stc1988, 2024-07-31:
> Change the line that fails from if (transferEncoding.toLowerCase() != "chunked") to if (transferEncoding?.toLowerCase() != "chunked").

Thank you. This change makes app works.

stc1988, 2024-08-10:
I confirmed this fixed. Thank you.

closed 2024-08-10 by stc1988