Moddable-OpenSource/moddable #1382: examples/packages/balls build error: `multiple manifests`

stc1988, 2024-08-01:
**Build environment:** macOS
**Moddable SDK version:** 3fe65e93a775773ca70b253bd80c536692a450b1
**Target device:** desktop simulator

**Steps to Reproduce**
1. Build and install the `examples/packages/balls` app using this build command: `mcpack mcconfig -d -m`
2. see error.

```
# error: moddable:piu/MC: multiple manifests!
### 1 error(s)
```

phoddie, 2024-08-01:
Unfortunately, we are unable to reproduce this result. We tested on two different systems.

The "multiple manifests" error is coming from `mcpack`. Please check $MODDABLE/build/tmp/mac/mc/mcpack-cache.json. There should be an entry for piu/MC like this:

```json
		{
			"specifier": "moddable:piu/MC",
			"manifests": [
				"/examples/manifest_piu.json"
			]
		}
```

Is there more than one item in the `manifests` array on your system?

Also, you might try deleting `mcpack-cache.json` and re-running `mcpack` (but I'm not optimistic that this will help).

stc1988, 2024-08-01:
>                {
>                         "specifier": "moddable:piu/MC",
>                         "manifests": [
>                                 "/examples/manifest_piu.json",
>                                 "/examples/manifest_piu.json"
>                         ]
>                 },
>                 {
>                         "specifier": "moddable:piu/QRCode",
>                         "manifests": [
>                                 "/modules/piu/MC/qrcode/manifest.json"
>                         ]
>                 },
>                 {
>                         "specifier": "moddable:piu/Timeline",
>                         "manifests": [
>                                 "/examples/manifest_piu.json",
>                                 "/examples/manifest_piu.json"
>                         ]
>                 },

It seems that `manifest_piu.json` is appearing twice overall. Even after deleting  `mcpack-cache.json` and re-running `mcpack`, the result didn’t change.

phoddie, 2024-08-02:
That's really strange. Just curious – are you running a clean copy of the repository or are there modifications (like additional apps)?

stc1988, 2024-08-02:
Three is no modification on my pulic branch, but I will try to clone repository again.

phoddie, 2024-08-02:
`mcpack` expects to process each manifest once. On your computer, it appears to process some manifests (such as manifest_piu.json) twice. The `parseManifest` function is where `mcpack` skips manifests it has already processed. Please add some diagnostics to `mcpack.js` at the top of `parseManifest`:

```js
	parseManifest(path, from) {
		if (this.manifests.already[path])
			return void trace(`already: ${path}\n`);
trace(`parseManifest: ${path}\n`);
```

Then rebuild tools and re-run mcpack. It will trace all the manifests it visits. On my Mac, the output includes:

```
...
parseManifest: /Users/hoddie/Projects/moddable/examples/manifest_piu.json
...
already: /Users/hoddie/Projects/moddable/examples/manifest_piu.json
...
```

I'd like to know what your computer traces for `manifest_piu.json`. Thank you.

stc1988, 2024-08-03:
I found the cause of the problem. I had incorrectly set the environment variables `$MODDABLE`.I set `/Users/Satoshi/Projects/moddable`, but actual path is `/Users/satoshi/Projects/moddable`.

Here is trace log
```
...
parseManifest: /Users/Satoshi/Projects/moddable/examples/manifest_piu.json
...
parseManifest: /Users/Satoshi/Projects/moddable/modules/input/color-picker/manifest.json
parseManifest: /Users/satoshi/Projects/moddable/examples/manifest_piu.json
...
```

phoddie, 2024-08-03:
Yes, that's it!

While it is a misconfiguration, I think it should / should work. You don't usually have trouble with the Moddable SDK tools, despite this misconfiguration. The `resolveFilePath` function in tool.js uses `realpath` to provide the canonical path. `mcpack` uses that in some places, but maybe not everywhere it should. Please try this:

```js
	parseManifest(path, from) {
		path = this.resolveFilePath(path);

		if (this.manifests.already[path])
			return;
```

I don't think that's the best place to put the call to `resolveFilePath`, but if it solves the problem it would be a good hint.

stc1988, 2024-08-04:
> I don't think that's the best place to put the call to resolveFilePath, but if it solves the problem it would be a good hint.

This fix solves the problem.

phoddie, 2024-08-05:
Thanks for confirming. We'll get a proper fix for this integrated so that others won't encounter this problem in the future.

stc1988, 2024-08-10:
I confirmed this fixed.

closed 2024-08-10 by stc1988