Moddable-OpenSource/moddable #1385: Partition Moddable Six

mauroForlimpopoli, 2024-08-07:
I'm trying to change the partition on the Moddable Six.
In the project directory I created a folder "sdkconfig"
In the Manifest I added:
```
	"build": {
		"PARTITIONS_FILE": "./sdkconfig/partitions.csv"
	},
```
and this is the code of partition
```
nvs,		data,	nvs,		0x9000,			0x004000,
otadata,    data,   ota,    	0x00d000,   	0x2000,
phy_init,	data,	phy,		0xf000,			0x001000,
ota_0,    	app, 	ota_0,   	0x010000,		0x260000,
ota_1,    	app,  	ota_1,   	0x270000, 		0x0F0000,
storage,	data,	spiffs,		0x360000,		0x070000,
xs,			0x40,	1,			0x3D0000,		0x010000,
ryc,		data,	spiffs,		0x3E0000,		0x001000,
ryc2,		data,	spiffs,		0x3E1000,		0x001000,
settings,	data,	1,			0x3E2000,		0x00E000
```
When the compiler runs it says:
```
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,8064K,
storage,data,spiffs,0x7f0000,64K,
*******************************************************************************
```
and when the program start there is this error in the xsbug:
C:\Users\mades\Projects\moddable\modules\files\flash\esp32\flash.c (52) # Break: Flash: can't find partition!

This is the complete manifest
```
{
	"include": [
		"$(MODDABLE)/examples/manifest_base.json",
		"$(MODDABLE)/examples/manifest_piu.json",
		"$(MODULES)/input/keyboard/manifest.json",
		"$(MODDABLE)/modules/files/file/manifest.json",
		"$(MODULES)/pins/analog/manifest.json",
		"$(MODULES)/network/wifi/manifest.json",
		"$(MODDABLE)/examples/manifest_net.json",
		"$(MODDABLE)/modules/network/ble/manifest_server.json",
		"$(MODDABLE)/modules/io/manifest.json",
		"$(MODDABLE)/modules/network/ble/manifest_client.json",
		"$(MODDABLE)/modules/drivers/peripherals/ds1307/manifest.json",
		"$(MODDABLE)/modules/drivers/rtc/manifest.json",
		"$(MODULES)/pins/i2c/manifest.json",
		"$(MODULES)/network/http/manifest.json",
		"$(MODDABLE)/modules/files/preference/manifest.json",
		"$(MODDABLE)/modules/files/flash/manifest.json"
		
		
	],
	"creation":{
		"stack":500
	},
	"modules": {
		"pins/smbus": "$(MODULES)/pins/smbus/smbus",
		"*": [
			"./main",
			"./assets",
			"$(MODULES)/input/keyboard/*",
			"$(MODULES)/files/flash/*",			
			"$(MODULES)/network/http/*",
			"$(MODULES)/network/wifi/*",
			"./funcSleep",
			"$(MODULES)/network/sntp/*",
			"$(MODULES)/drivers/neopixel/*",
			"$(MODULES)/drivers/neopixel/esp32/*",
			"$(MODULES)/drivers/rtc/rtc",
			"$(MODULES)/drivers/rtc/ds1307",
			"$(MODULES)/drivers/onewire/sim/*"
		]
	},
	"resources":{
		"*-alpha": [
			"$(MODDABLE)/examples/assets/fonts/OpenSans-Semibold-18",
			"$(MODDABLE)/examples/assets/fonts/OpenSans-Regular-20",
			"$(MODDABLE)/examples/assets/fonts/arial20",
			"$(MODDABLE)/examples/assets/fonts/arial32",
			"$(MODDABLE)/examples/assets/fonts/OpenSans-Semibold-28",
			"$(MODDABLE)/examples/assets/fonts/impact64",
			"$(MODDABLE)/examples/assets/fonts/futura"
		],
		"*-image(50)": [
			"$(MODDABLE)/examples/assets/images/Bluetooth-icon",
			"$(MODDABLE)/examples/assets/images/Wifi-icon",
			"$(MODDABLE)/examples/assets/images/unlock-icon",
			"$(MODDABLE)/examples/assets/images/lock-icon",
			"$(MODDABLE)/examples/assets/images/down-icon",
			"$(MODDABLE)/examples/assets/images/memory-icon",
			"$(MODDABLE)/examples/assets/images/sun-icon",
			"$(MODDABLE)/examples/assets/images/moon-icon",
			"$(MODDABLE)/examples/assets/images/rs232-icon",
			"./assets/logo",
			"./assets/qrcode_google"
		],
		"*-mask": [
			"$(MODDABLE)/examples/assets/images/thermometer-mask",
			"$(MODDABLE)/examples/assets/images/battery-mask",
			"$(MODDABLE)/examples/assets/images/switch-mask"	,
			"$(MODDABLE)/examples/assets/images/option-mask"	,
			
			"./assets/lock-icon-mask",
			"./assets/unlock-icon-mask",
			"./assets/down-icon-mask",
			"./assets/memory-icon-mask",
			"./assets/moon-icon-mask",
			"./assets/rs232-icon-mask",
			"./assets/sun-icon-mask",
			"./assets/battery-icon-mask",
			"./assets/Bluetooth-icon-mask",
			"./assets/Wifi-icon-mask",
			
			"./assets/led_on-disc-mask",
			"./assets/switch_off-mask",
			"./assets/switch_on-glyph-mask",
			"./assets/attendere",
			"./assets/print",
			"./assets/downCloud",
			"./assets/one100",
			"./assets/one100E"
		],			
		"*": [
			"./assets/print-icon",
			"./assets/sms2-icon",
			"./assets/opts-icon",
			"./assets/menuIcon"
		]
	},	
	"preload": [
		"neopixel",
		"http",
		"sntp",
		"file",
		"pins/analog",
		"ble",
		"preference",
		"flash",
		"keyboard",
		"time",
		"timer",
		"funcSleep",
		"wifi",
		"keyboard",
		"assets",
		"ds1307",
		"rtc",
		"smbus",
		"onewire",
		"DS18X20"
	],
	"platforms": {
		"esp": {
			"config": {
				"rotation": 90
			}
		},
		"esp32": {
			"config": {
				"rotation": 90,
				"brightness": "off",
				"file": {
					"root": "/mod/"
				},
				"onewire": {
					"pin": "16"
				}
			},
			"defines": {
				"ls013b4dn04": {
					"dither": 1,
					"updateAll": 1
				},
				"onewire": {
					"driver_rmt": "RMT",
					"rmt_rx_channel": "RMT_CHANNEL_2",
					"rmt_tx_channel": "RMT_CHANNEL_3"
				}
			},
			"include": [
				"$(MODULES)/pins/digital/manifest.json"
			],
			"modules": {
				"*": [
				"$(MODULES)/files/flash/*",
				"$(MODULES)/files/flash/esp32/*",
				"$(MODULES)/drivers/onewire/esp/*"
				]
			},						
			"build": {
				"PARTITIONS_FILE": "./sdkconfig/partitions.csv"
			},
			"modules": {
				"ota": "$(BUILD)/devices/esp32/modules/ota/*"
			},
			"preload": "ota" 			
		}
    },
	"ble":{
		"*": [
			"./bleservices/*",
			"./bleservicesCli/*"
		]
	}	
}
```

How can I do?

the memory map of ESP32-S3 is differet from ESP32
do I need to change the partition offset?
do I need to change the value of "static"
		

regards

phoddie, 2024-08-07:
Thank you for the report. There was an oversight that prevented projects from overriding the partition map of ESP32 devices that have a custom partition map. Here is a patch to fix that. The patch will be included in the next Moddable SDK update.

<details>
<summary>patch.diff</summary>

```
From 14a6d5a3bd42d108d86e435618747d3d88079649 Mon Sep 17 00:00:00 2001
From: Peter Hoddie <peter@moddable.tech>
Date: Wed, 7 Aug 2024 11:18:13 -0700
Subject: [PATCH] allow projects to override device's custom partition map
 #1385

---
 build/devices/esp32/targets/c3_32s_kit_2m/manifest.json        | 2 +-
 build/devices/esp32/targets/esp32h2/manifest.json              | 2 +-
 build/devices/esp32/targets/esp32h2_cdc/manifest.json          | 2 +-
 build/devices/esp32/targets/esp32h2_ili9341/manifest.json      | 2 +-
 build/devices/esp32/targets/lilygo_t_qt/manifest.json          | 2 +-
 build/devices/esp32/targets/lilygo_tdisplay_s3/manifest.json   | 2 +-
 build/devices/esp32/targets/m5atom_s3/manifest.json            | 2 +-
 build/devices/esp32/targets/m5dial/manifest.json               | 2 +-
 build/devices/esp32/targets/m5stack_core2/manifest.json        | 2 +-
 build/devices/esp32/targets/m5stack_cores3/manifest.json       | 2 +-
 build/devices/esp32/targets/m5stack_fire/manifest.json         | 2 +-
 build/devices/esp32/targets/moddable_six/manifest.json         | 2 +-
 build/devices/esp32/targets/moddable_six_cdc/manifest.json     | 2 +-
 build/devices/esp32/targets/qtpys3_cdc/manifest.json           | 2 +-
 build/devices/esp32/targets/qtpys3_ili9341/manifest.json       | 2 +-
 build/devices/esp32/targets/xiao_esp32s3/manifest.json         | 2 +-
 build/devices/esp32/targets/xiao_esp32s3_ili9341/manifest.json | 2 +-
 tools/mcmanifest.js                                            | 2 +-
 18 files changed, 18 insertions(+), 18 deletions(-)

diff --git a/build/devices/esp32/targets/c3_32s_kit_2m/manifest.json b/build/devices/esp32/targets/c3_32s_kit_2m/manifest.json
index 33d688076..a4a4d8ee6 100644
--- a/build/devices/esp32/targets/c3_32s_kit_2m/manifest.json
+++ b/build/devices/esp32/targets/c3_32s_kit_2m/manifest.json
@@ -3,7 +3,7 @@
 		"ESP32_SUBCLASS": "esp32c3",
 		"UPLOAD_SPEED": "460800",
 		"SDKCONFIGPATH": "./sdkconfig",
-		"PARTITIONS_FILE": "./sdkconfig/partitions.csv"
+		"PARTITIONS_FILE_FOR_TARGET": "./sdkconfig/partitions.csv"
 	},
 	"include": [
 		"$(MODULES)/drivers/button/manifest.json",
diff --git a/build/devices/esp32/targets/esp32h2/manifest.json b/build/devices/esp32/targets/esp32h2/manifest.json
index c183973cf..1792b7dce 100644
--- a/build/devices/esp32/targets/esp32h2/manifest.json
+++ b/build/devices/esp32/targets/esp32h2/manifest.json
@@ -2,7 +2,7 @@
 	"build":{
 		"ESP32_SUBCLASS": "esp32h2",
 		"SDKCONFIGPATH": "./sdkconfig",
-		"PARTITIONS_FILE": "./sdkconfig/partitions.csv"
+		"PARTITIONS_FILE_FOR_TARGET": "./sdkconfig/partitions.csv"
 	},
 	"include": [
 		"$(MODULES)/drivers/button/manifest.json",
diff --git a/build/devices/esp32/targets/esp32h2_cdc/manifest.json b/build/devices/esp32/targets/esp32h2_cdc/manifest.json
index 271d39202..024f757e0 100644
--- a/build/devices/esp32/targets/esp32h2_cdc/manifest.json
+++ b/build/devices/esp32/targets/esp32h2_cdc/manifest.json
@@ -3,7 +3,7 @@
 		"ESP32_SUBCLASS": "esp32h2",
 		"USE_USB": "2",
 		"SDKCONFIGPATH": "./sdkconfig",
-		"PARTITIONS_FILE": "./sdkconfig/partitions.csv",
+		"PARTITIONS_FILE_FOR_TARGET": "./sdkconfig/partitions.csv",
 		"PROGRAMMING_MODE_MESSAGE": "INSTRUCTIONS: Hold the BOOT button and press then release the RST button.",
[... 180 lines cut ...]
+		"PARTITIONS_FILE_FOR_TARGET": "./sdkconfig/partitions.csv",
         "PROGRAMMING_MODE_MESSAGE": "INSTRUCTIONS: Hold the B button and press then release the R button.",
         "BEFORE_DEBUGGING_MESSAGE": "Press and release the R button."
 	},
diff --git a/tools/mcmanifest.js b/tools/mcmanifest.js
index 9add26c69..5efe28446 100644
--- a/tools/mcmanifest.js
+++ b/tools/mcmanifest.js
@@ -146,7 +146,7 @@ export class MakeFile extends FILE {
 		const outputConfigDirectory = tool.outputPath + tool.slash + "tmp" + tool.slash + "esp32" + tool.slash + (tool.subplatform ?? "") + tool.slash + (tool.debug ? "debug" : (tool.instrument ? "instrument" : "release")) + tool.slash + tool.environment.NAME + tool.slash + "xsProj-" + ESP32_SUBCLASS;
 		tool.createDirectory(outputConfigDirectory);
 
-		let PARTITIONS_FILE = tool.environment.PARTITIONS_FILE;
+		let PARTITIONS_FILE = tool.environment.PARTITIONS_FILE ?? tool.environment.PARTITIONS_FILE_FOR_TARGET;
 		if (!PARTITIONS_FILE) {
 			const PROJ_DIR_TEMPLATE = `${tool.buildPath}/devices/esp32/xsProj-${tool.environment.ESP32_SUBCLASS}`;
 			PARTITIONS_FILE = `${PROJ_DIR_TEMPLATE}/partitions.csv`
-- 
2.41.0

```

</details>

**Important**: After applying the patch, it is necessary to rebuild Moddable SDK tools and then do a clean build of your project.

mauroForlimpopoli, 2024-08-08:
I followed your indications and now:
```
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,16K,
otadata,data,ota,0xd000,8K,
phy_init,data,phy,0xf000,4K,
ota_0,app,ota_0,0x10000,2432K,
ota_1,app,ota_1,0x270000,960K,
storage,data,spiffs,0x360000,448K,
xs,64,1,0x3d0000,64K,
ryc,data,spiffs,0x3e0000,4K,
ryc2,data,spiffs,0x3e1000,4K,
settings,data,phy,0x3e2000,56K,
*******************************************************************************
```
so I think thats ok.
Thank you.
The ESP32-S3 microcontroller has more memory than ESP32
The previus manifest used only 4MB of memory.
Which limitations I need to respect to use all 8MB of memory

mauroForlimpopoli, 2024-08-08:
PS: the amount of RAM memory available for the program is changed respect EPS32 ?

phoddie, 2024-08-08:
> so I think thats ok.
> Thank you.

Great.

> The ESP32-S3 microcontroller has more memory than ESP32

I believe you are referring to the fact that the module used in Moddable Six has 8 MB of flash whereas the module use in Moddable Two has 4 MB of flash.

> The previus manifest used only 4MB of memory.

The manifest does not define the use of flash memory. That is done by the `partitions.csv` as discussed above.

> Which limitations I need to respect to use all 8MB of memory

You may adjust the size of your partitions in `partitions.csv` as long as the total size of all partitions does not exceed 8 MB.

> PS: the amount of RAM memory available for the program is changed respect EPS32 ?

You may want to take a look at the Moddable Six documentation in the section titled "[8 MB PSRAM](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/devices/moddable-six.md#additional-memory)". 

This matter is separate from the original topic of this issue. Please open a separate issue if you have questions.

mauroForlimpopoli, 2024-08-09:
Ok, thank  you Peter

closed 2024-08-09 by phoddie