Moddable-OpenSource/moddable #1386: Incorrect information in /documentation/files/files.md regarding changing the value of config.file.root

kallistisoft, 2024-08-09:
**Build environment:** All
**Moddable SDK version:** 4.9
**Target device:** Any

**Description**
The documentation for overriding the default `config.file.root` value is incorrect.
Currently in [/documentation/files/files.md](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/files/files.md#host-file-system-configuration) the section "**Host File System Configuration**" has this code fragment:
```javascript
"defines": {
	"file":{
		"root": "#/myroot"
	}
}
```

The correct code fragment is:
```javascript
"config": {
	"file":{
		"root": "/myroot"
	}
}
```

**Errata**
Submitting as an issue to avoid sending a two line PR :)

kallistisoft, 2024-08-09:
Additional question; in `/modules/files/file/manifest.json`

The default file root is `/tmp/` for linux and wasm builds, whereas for windows its './' and on macos it's `/Users/Shared/`

Is there a compelling rational for this discrepancy with linux?
Seems like it should also be './' for consistency?

phoddie, 2024-08-09:
@kallistisoft – thanks for the reports. I'll update the Files document to correct that. The convention is to have the `root` end with a `/` – so you can write `config.file.root + "foo.json"`, so it will be `/myroot/`.

The file root for desktop computers is a little bit arbitrary and, admittedly, inconsistent. But, I'm not sure what would be least surprising; ;) The idea is to have the file system be in some sort of local sandbox, since that better reflects the file system on an embedded microcontroller.

phoddie, 2024-09-14:
Fixed.

closed 2024-09-14 by phoddie