Moddable-OpenSource/moddable #1392: XS reports incorrect byteLength for some TypedArrays

gibson042, 2024-08-16:
**Environment:** XS 15.5.1, slot 32 bytes, ID 4 bytes

**Description**
When a TypedArray has an element size that does not evenly divide its underlying buffer (because of e.g. resize), `byteLength` incorrectly includes the excess bytes.

**Steps to Reproduce**
```js
const buf = new ArrayBuffer(8, { maxByteLength: 8 });
new Uint8Array(buf).set(Array.from({ length: buf.byteLength }, (_, i) => i));
const arr = new Uint32Array(buf);
const log = arr => {
  const bufSize = arr.buffer.byteLength;
  const { byteOffset: arrByteOffset, byteLength: arrByteLength, BYTES_PER_ELEMENT } = arr;
  const arrType = arr.constructor.name;
  const bufSliceDesc = `ArrayBuffer(${bufSize})[${arrByteOffset}:${arrByteOffset + arrByteLength}]`;
  const elRepr = n => "0x" + n.toString(16).padStart(BYTES_PER_ELEMENT * 2, "0");
  print(`${bufSliceDesc} holds ${arrType}(${arr.length}) [${[...arr].map(elRepr).join(" ")}]`);
};
log(arr);
buf.resize(7);
log(arr);
```

**Actual behavior**
```
ArrayBuffer(8)[0:8] holds Uint32Array(2) [0x03020100 0x07060504]
ArrayBuffer(7)[0:7] holds Uint32Array(1) [0x03020100]
```
(little-endian)

**Expected behavior**
```
ArrayBuffer(8)[0:8] holds Uint32Array(2) [0x03020100 0x07060504]
ArrayBuffer(7)[0:4] holds Uint32Array(1) [0x03020100]
```

[`byteLength`](https://tc39.es/ecma262/multipage/indexed-collections.html#sec-get-%typedarray%.prototype.bytelength) is specified to depend upon [TypedArrayByteLength](https://tc39.es/ecma262/multipage/ordinary-and-exotic-objects-behaviours.html#sec-typedarraybytelength), which for a resizable buffer returns elementLength × elementSize (where elementLength comes from [TypedArrayLength](https://tc39.es/ecma262/multipage/ordinary-and-exotic-objects-behaviours.html#sec-typedarraylength) as `floor((byteLength - byteOffset) / elementSize)`, i.e. the integer number of complete TypedArray elements that can fit in the buffer).

phoddie, 2024-08-16:
Thanks for the report and test case. I'm able to reproduce the behavior. This looks straightforward to resolve.

phoddie, 2024-09-14:
Fixed.

closed 2024-09-14 by phoddie