Moddable-OpenSource/moddable #1394: Some `await` expressions break XS parsing

gibson042, 2024-08-20:
**Environment:** XS 15.5.1, slot 32 bytes, ID 4 bytes

**Description**
XS fails to parse a [_ConditionalExpression_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-expressions:~:text=In%2C%20%3FYield%2C%20%3FAwait%5D-,ConditionalExpression,-%5BIn%2C%20Yield%2C%20Await) that starts with an `await`.

**Steps to Reproduce**
```js
async function fn() {
  const notAwaited = {
    then() { throw Error("unexpected `then` invocation"); },
  };
  await null ?? notAwaited; // but at least `||` and `&&` are also affected
}
print("START");
fn().then(() => print("END"));
```

**Actual behavior**
```
SyntaxError: missing ;
```

**Expected behavior**
```
START
END
```

In statement position with an <sub>`Await`</sub> parameter (e.g., inside an `async` function), `await null ?? notAwaited;` should parse as:
* [_ExpressionStatement_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=%3B-,ExpressionStatement,-%5BYield%2C%20Await%5D)<sub>[Yield, Await]</sub> → _Expression_<sub>[+In, ?Yield, ?Await]</sub> **`;`**
* [_Expression_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=%5B%3FYield%2C%20%3FAwait%5D-,Expression,-%5BIn%2C%20Yield%2C%20Await)<sub>[In, Yield, Await]</sub> → _AssignmentExpression_<sub>[?In, ?Yield, ?Await]</sub>
* [_AssignmentExpression_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=In%2C%20%3FYield%2C%20%3FAwait%5D-,AssignmentExpression,-%5BIn%2C%20Yield%2C%20Await)<sub>[In, Yield, Await]</sub> → _ConditionalExpression_<sub>[?In, ?Yield, ?Await]</sub>
* [_ConditionalExpression_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=In%2C%20%3FYield%2C%20%3FAwait%5D-,ConditionalExpression,-%5BIn%2C%20Yield%2C%20Await)<sub>[In, Yield, Await]</sub> → _ShortCircuitExpression_<sub>[?In, ?Yield, ?Await]</sub>
* [_ShortCircuitExpression_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=In%2C%20%3FYield%2C%20%3FAwait%5D-,ShortCircuitExpression,-%5BIn%2C%20Yield%2C%20Await)<sub>[In, Yield, Await]</sub> → _CoalesceExpression_<sub>[?In, ?Yield, ?Await]</sub>
* [_CoalesceExpression_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=In%2C%20%3FYield%2C%20%3FAwait%5D-,CoalesceExpression,-%5BIn%2C%20Yield%2C%20Await)<sub>[In, Yield, Await]</sub> → _CoalesceExpressionHead_<sub>[?In, ?Yield, ?Await]</sub> **`??`** _BitwiseORExpression_<sub>[?In, ?Yield, ?Await]</sub>
  * [_CoalesceExpressionHead_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=In%2C%20%3FYield%2C%20%3FAwait%5D-,CoalesceExpressionHead,-%5BIn%2C%20Yield%2C%20Await)<sub>[In, Yield, Await]</sub> → … → _UnaryExpression_<sub>[Yield, Await]</sub>
    * [_UnaryExpression_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=UnaryExpression%5BYield%2C%20Await,%3A)<sub>[Yield, Await]</sub> → <sub>[+Await]</sub> _AwaitExpression_<sub>[?Yield]</sub>
    * [_AwaitExpression_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-grammar-summary:~:text=%5B~Yield%2C%20%2BAwait%5D-,AwaitExpression,-%5BYield%5D)<sub>[Yield]</sub> → **`await`** _UnaryExpression_<sub>[?Yield, +Await]</sub>
    * [_UnaryExpression_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=UnaryExpression%5BYield%2C%20Await,%3A)<sub>[Yield, Await]</sub> → [_UpdateExpression_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=%5B%3FYield%2C%20%3FAwait%5D-,UpdateExpression,-%5BYield%2C%20Await%5D)<sub>[Yield, Await]</sub> → … → [_PrimaryExpression_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=ReservedWord-,PrimaryExpression,-%5BYield%2C%20Await%5D)<sub>[Yield, Await]</sub> → [_Literal_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=%29-,Literal,-%3A) → [_NullLiteral_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=%7D-,NullLiteral,-%3A%3A) → **`null`**
  * [_BitwiseORExpression_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=In%2C%20%3FYield%2C%20%3FAwait%5D-,BitwiseORExpression,-%5BIn%2C%20Yield%2C%20Await)<sub>[In, Yield, Await]</sub> → … → [_PrimaryExpression_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=ReservedWord-,PrimaryExpression,-%5BYield%2C%20Await%5D)<sub>[Yield, Await]</sub> → [_IdentifierReference_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=A.2%20Expressions-,IdentifierReference,-%5BYield%2C%20Await%5D)<sub>[Yield, Await]</sub> → **[_Identifier_](https://tc39.es/ecma262/multipage/grammar-summary.html#sec-statements:~:text=await-,Identifier,-%3A)** <sub>but not _ReservedWord_</sub>

phoddie, 2024-08-21:
Thanks for reporting this.

The issue is real but more obscure than it seems at first. The problem only happens if the (Stage 3) explicit resource management API support is enabled in the XS build with `mxExplicitResourceManagement`. That is only the case in `xst` -- which is, of course, what you are using in your tests here. The Moddable SDK today defaults to explicit resource management being off.

The problem is caused by `using` not being a keyword and the lookahead required in the parser to handle that. Obviously we need to correct this, but it may take a bit as it is a little tricky.

phoddie, 2024-09-01:
Quick update – we are testing a fix for this. It requires some non-trivial changes to the parser for the additional look-ahead, so we are testing it internally for a bit before pushing it public.

phoddie, 2024-09-14:
Fixed.

closed 2024-09-14 by phoddie