Moddable-OpenSource/moddable #1396: What is `commodetto/static`?

stc1988, 2024-08-31:
I tried running examples/commodetto/static, but I couldn’t find any documentation on it, so I’m not sure what functionality it provides.

The [example description](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/commodetto/readme.md#static) says “efficiently render static,” but in what kind of cases can this functionality be used?

phoddie, 2024-09-01:
The functionality it provides is rendering static efficiently. ;)

It has a few uses:

- Show static when waiting for video to be available
- Demonstrate how to implement and integrate a custom renderer extension for Poco
- Confirm that irregular shaped objects, such as animated GIFs, are not drawing the background pixels

Here's a video of that last one:

https://github.com/user-attachments/assets/8b7018b5-9f1e-49cc-be74-f6bd0a03030c

It is pretty efficient. On an ESP32-S3, it can fill the screen with static at about 96 FPS.

**Note**: The ESP-IDF random number generator (`esp_random`) is too slow for efficient static (it is designed to be a cryptographically secure RNG). `pocostatic.c` should use `rand()` on ESP32 instead. I'll update that in the sources.

meganetaaan, 2024-09-01:
I didn't understand what 'static' meant until I saw this thread. In Japan, we call it 'テレビの砂嵐' (TV's sandstorms), but now I get it. Thank you for explaining!

stc1988, 2024-09-01:
@phoddie 
I understood the meaning of “static” and your explanation. Therefore, I don’t think any revisions to the examples are necessary.​

@meganetaaan 
Thank you for following up.

closed 2024-09-01 by stc1988

phoddie, 2024-09-02:
Thanks for your feedback. A simple solution is to link to Wikipedia and this thread. I made the following update to the documentation:

> The `static` example demonstrates how to use Poco to efficiently render [video static](https://en.wikipedia.org/wiki/Noise_(video)). More information [here](https://github.com/Moddable-OpenSource/moddable/issues/1396).